gepd/Deviot #152: DEV

tablatronix, 2017-08-26:
Before report an issue, please make sure it hasn't been reported before.

### Operating system
osx
### Board used
nodemcu v1
### Description of problem
upload hotkey not wokring
### Steps to Reproduce
press option E

### Actual Results
ascii character outputs
### Expected Results

### Additional info

*You can edit the text above according to your need*

gepd, 2017-08-27:
At this moment I don't have access to OSX, what shortcut could work insted of `Option E`?

You can test it editing this [file/line](https://github.com/gepd/Deviot/blob/develop/shortcuts/Default%20(OSX).sublime-keymap#L58)

tablatronix, 2017-08-27:
Was there a reason you changed them, or did you just want them single key not double?

gepd, 2017-08-27:
Someone opened a issue for a shortcut that was overriden a system shortcut, so I made some changes to have kind the same shortcuts in all platforms, but shortcuts in windows work different from linux and as I don't have access to OSX I haven not tried if all is working correctly there.

Would be something like `cmd+alt+e` better/comfortable?

tablatronix, 2017-08-29:
Not sure i understand, PC is still alt D - E

tablatronix, 2017-08-29:
#132 ?

tablatronix, 2017-08-29:
I would remove most of the shortcuts and only set ones for the absolute needed.
Serial port console stuff
compile upload
library access

gepd, 2017-08-29:
Yes, remove the rest can be a possibility. The main issue with shortcuts is that windows let you use `alt+d, alt+f` but linux don't, so I assume in OSX will have the same behavior.

I've testing in Linux and seems that shotcuts like: `alt+shift+f` could be work. 
That section could have the following shortcuts:

Compile: `alt+shift+f`
Upload: `alt+shift+e`
Override baudrate: `alt+shift+o`
Clean Sketch: `alt+shift+x`
Open pio.ini: `alt+shift+i`

tablatronix, 2017-08-29:
I actually prefer 2 strokes easier to use than double modifiers.
alt D + alt E is esaier than ctrl +shift + x

gepd, 2017-08-29:
alt based modifiers seems not to be available in Linux I suppose it aren't available in OSX too. They are used to open the float menu, what is working is ctrl modifier, but `ctrl+d` is used by the system, so I should use a new letter

tablatronix, 2017-08-29:
oh yeah talk about complicated
you almost need a script to read key maps and show whats available. lol

gepd, 2017-08-29:
Yes, it's a good idea. The problem is the diferences between each platform

John-Titor, 2017-11-26:
The macOS modifiers are all option+?, which are nonstandard (usually option is used for entering other characters); I'd recommend not using option+<anything> without another modifier.