gepd/Deviot #154: Doc request: show creation of new sketch, start to finish

jwhendy, 2017-08-27:
After a break from using Arduino, I used it today and revisited Deviot. I ran into the same issue as I had in #88 and as a sanity check, I started to create a reproducible example. But... I couldn't really do this as I found there doesn't appear to be any instructions around starting a new `.ino` from scratch.

For example, I thought perhaps using subl3 File -> New file might be different than Deviot -> New sketch, which could cause some sort of issues with my `platformio` environment in the sketch dir.

[The wiki](https://github.com/gepd/Deviot/wiki) tells me what it is, but not really how to use it. [Instructables](http://www.instructables.com/id/Deviot-the-IoT-Plugin-for-Sublime-Text/) (written by @gepd also) alludes to me being prompted to enter `platformio` relevant stuff, but that doesn't happen to me. 

Here are some super basics I think would really help to answer. I have maybe 5 years of Arduino coding experience and have managed to use `platformio` manually, and Deviot is far from clear to me.

- does it matter where the file lives? (Arduino's own IDE, for example, will insist on the sketch being in it's own dir, and maybe require it live in the Arduino path)
- is File -> New file any different than Deviot -> New sketch? If so, what does Deviot do?
- For Deviot -> New Sketch, what is one to input? I typed `deviot-test.ino` and then was told to pick a dir; I saw no options to create a new folder, so I went and made `~/Arduino/deviot-test` manually, then came back to subl3 and picked that dir. I ended up with `~/Arduino/deviot-test/deviot-test.ino/deviot-test.ino` (yes, it named the folder with the `.ino` extention)
- am I supposed to know `platformio` and create my `platformio.ini` myself, or should Deviot be doing that (I don't get any prompts as it stands when I build for the first time)
- running `platformio` manually will require the sketch to live in `sketch-dir/src`; is that the case with Deviot? Should it be?
- what does one need to do for libraries? Per #88, I'm back to Deviot not seeing them. It *appears* that they need to live in `.platformio/lib`, but `platformio` itself is happy with them symlinked in `sketch-dir/lib`, so why doesn't this satisfy Deviot? (I'll re-open this separately, but am adding to noob "gotchas" for documenting as well)

If you point me in the right direction, I can go through these steps and take screenshots, contribute to the wiki, or whatnot. At present, I'm too unsure/confused to even make a stab at documenting!

jwhendy, 2017-08-27:
Attempt 2: making progress...

- for creating a new sketch, type the name only (e.g. `sketch-name`), not the extension of the file; the `.ino` will be added in the directory you choose in the next step
- for the directory choice, this is where you want a new directory *created*; the newly created dir will have the same name as your sketch
- all the `platformio` related stuff is put in `/tmp/Deviot/sketch-name`, *sort of*. The `.ino` file is still in `/path/to/sketch-name`, but `src` in `/tmp/Deviot/sketch-name` is empty?

That's where I'm at so far. I don't get the separation of the files. It makes both directories unusable outside of Deviot from what I can see. Why not just do:

```
[~/Arduino/deviot-test]$ tree
.
├── lib
├── platformio.ini
└── src
    └── deviot-test.ino
```

Or perhaps:
```
[~/Arduino/deviot-test]$ tree
.
├── deviot-test.ino -> src/deviot-test.ino
├── lib
├── platformio.ini
└── src
    └── deviot-test.ino
```

Or maybe symlink the actual `.ino` to the `/tmp` dir as well; that way at least there would be a `platformio` compliant directory somewhere on the system. Sorry if I'm vastly misunderstanding how this is working and the motivation. I'm just poking around a bit and thinking through what I see going on in the background.

gepd, 2017-08-27:
> does it matter where the file lives? (Arduino's own IDE, for example, will insist on the sketch being in it's own dir, and maybe require it live in the Arduino path)

Deviot will check some things before work with you project, first; if your project structure is like Arduino (folder/sketch) or like PlatformIO folder/platformio.ini + src, lib folders.

At this moment it's not strict if the sketch is inside of a folder or not... I'll improve this, in some case it may cause a problem if you don't use a folder.

> is File -> New file any different than Deviot -> New sketch? If so, what does Deviot do?
The difference is 

* `File > New` will generate an empty file
* `Deviot > New Sketch` will generate a file with an empty template (`ino` or `cpp` template)

> For Deviot -> New Sketch, what is one to input? I typed deviot-test.ino and then was told to pick a dir; I saw no options to create a new folder, so I went and made ~/Arduino/deviot-test manually, [...]

I've just push a commit to fix the problem with the file extension included in the folder, but I've also added an improvement:

**Before the improvement**

if you wanted to created a new file, you needed to input only the filename without the extension. By default, it was using a `.ino` extension, but you could change that in `Deviot > Options > Use CPP File on New File` (Yes, that option need to be renamed) 

**With the improvement**

If you input `test.ino` you will end with `your_path/test/ino.ino`
If you input `test.cpp` you will end with `your_path/test/ino.cpp` 

So now, first it will look if there is an extension and if it has it, will be use to name the sketch and to select the template, overriding the choise in `Deviot > Option`

If you don't use an extension, it will work as before

> am I supposed to know platformio and create my platformio.ini myself, or should Deviot be doing that (I don't get any prompts as it stands when I build for the first time)

You don't need to create the `platformio.ini` file. Deviot will do for you. If you select one board or multiples boards, they will be added to `platformio.ini` each time you compile or upload an sketch

If you have a PlatformIO project, Deviot will recognize your `platformio.ini`, So `Deviot > Select Environment` will show all the environments already initialized and if you have some extra flag or configuration it will work same as in PlatformIO Core

> what does one need to do for libraries? Per #88, I'm back to Deviot not seeing them. It appears that they need to live in .platformio/lib, but platformio itself is happy with them symlinked in [...]

Here, if a symlinked lib work in `platformio` (You mean PlatformIO IDE?) should work in deviot, I've not test that, but maybe you can upload a test to check that.

To keep adding info: There is not change in the PIO Library Manager
`Find/Install Library` Will do just that, find a library depending of the term you like and select the library selected, this library will be stored in `.platformio/lib` 

`Remove Library` will search you current installed libraries (only in ` .platformio/lib`) and will remove the one you select

There is an option to add an extra folder where you can have multiples libraries. This feature was added thinking in the Arduino IDE library folder. From `Deviot > Extra Library Folder` you can select a folder with all your libraries and it will be added automatically in the `platformio.ini` when you compile the sketch. You'll have to manually add and remove the libraries from that folder. 

> for creating a new sketch, type the name only (e.g. sketch-name), not the extension of the file; the .ino will be added in the directory you choose in the next step

This is already fixed :) check it and tell me if it's working now for you (in the develop branch)

> for the directory choice, this is where you want a new directory created; the newly created dir will have the same name as your sketch

Yes, it will work same as the Arduino IDE, but it will change with an extra option, I'll explain in the next point

> all the platformio related stuff is put in /tmp/Deviot/sketch-name, sort of. The .ino file is still in /path/to/sketch-name, but src in /tmp/Deviot/sketch-name is empty?

Here is where Arduino IDE or PlatformIO matter.

This proyect born as an alternative to Arduino IDE, so should be logic to be have full compatibility with that kind of project, but it shouldn't lost the compatibility with PIO too.

**Arduino IDE**

If you come from the Arduino IDE, could be a little confusing had a new folder structure, new files to compile your sketch. So by default, Deviot works like the Arduino IDE, and will let your sketch clean without extra files. The rest of the files will be added in a tmp folder. To do this Deviot use the `src_dir` option to everride the `src` folder that is where you put your sketch (in the `platformio.ini` file).

**PlatformIO**

If you open a PlatformIO project, it will work as it is expected, the file/folder structure wont be modified.

**New Sketch**

By default Deviot will generate your project like the Arduino IDE, but there is an option in `Deviot > Options > Use PlatformIO Structure` That will create your project like PlatformIO does, in the same path where you put your sketch. This structure change will be made when you compile or upload the sketch.

* It will first initialize the environment (Generate the `platformio.ini`, `lib` and `src` folder)
* Finally Deviot will automatically move your sketch to the `src` folder and will start to compile the sketch

I hope with those explanation you can understand a little more about Deviot. But any extra doubt, you can ask again, I don't mind to explain. I know I haven't added them enough documentation to the wiki

jwhendy, 2017-09-06:
Wow, that was an amazing response! Sorry for the delay in getting back to you. Most points sound like awesome improvements... to try and get to just the questions/loose ends:

>Here, if a symlinked lib work in platformio (You mean PlatformIO IDE?) should work in deviot, I've not test that, but maybe you can upload a test to check that.

What I mean is that what works when using `platformio` directly doesn't work for `deviot`. I'll use the project I was working on when I wrote this, which was to use LED strips in my son's Lite Brite :) 

```
[~/Arduino/lite-brite]$ tree
.
├── lib
│   └── FastLED -> ../../libraries/FastLED/
├── lite-brite.ino -> src/lite-brite.ino
├── platformio.ini
└── src
    └── lite-brite.ino

[~/Arduino/lite-brite]$ platformio run
### [ output removed ] ###
[SUCCESS] Took 0.86 seconds
```

From Deviot, *without* the extra library option:
```
~/Arduino/lite-brite/lite-brite.ino:1:21: fatal error: FastLED.h: No such file or directory
#include "FastLED.h"
```

That's all I meant: `platformio` understands the directory structure, but `deviot` does not.

This goes back to having things in `/tmp` for `deviot` and in the directory itself for `platformio`. Since this is a wrapper, my preference is to just keep it in the folder. I didn't know about the option you mentioned to `Use PlatformIO Structure` -- that looks great! Good point on not confusing users. I didn't think of that at all, so now I understand the `/tmp` chioce. 

I'll try the development version... I must have the "normal" one as these changes aren't in my version.

Thanks again!

gepd, 2017-09-07:
I've just check the symlinked library problem, and in ubuntu it has compiled without any problem.
You can try updating the PlatformIO version from `Deviot > Options > Upgrade PlatfromIO` There is a posibility that you had an old version where that feature is not working, if that is not the case, can you tell me what is the O.S you have tried it?

jwhendy, 2017-09-07:
I just updated the original post above to show answered questions; let me start a new issue with a reproducible issue just focused on resolving these libraries. Hope that's alright, as this one got pretty muddy with so many issue (totally my fault!). Thanks for your patience and assistance!

closed 2017-09-07 by jwhendy

jwhendy, 2017-09-07:
See #162; made reproducible example to help get to the root of the libraries issue.