gepd/Deviot #155: DEV cannot uplload when network is unavailable

tablatronix, 2017-08-27:
Before report an issue, please make sure it hasn't been reported before.

### Operating system
osx
### Board used
nodemcu v1
### Description of problem
upload fails no error
### Steps to Reproduce
connect to bad network 
### Actual Results
```
Traceback (most recent call last):
  File "/Applications/Sublime Text.app/Contents/MacOS/sublime_plugin.py", line 797, in run_
    return self.run()
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot/commands/deviot_select_port.py", line 6, in run
    QuickMenu().quick_serial_ports()
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot/libraries/quick_menu.py", line 220, in quick_serial_ports
    self.quick_list = self.quick_serial_list()
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot/libraries/quick_menu.py", line 257, in quick_serial_list
    ports_list = self.get_ports_list()
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot/libraries/preferences_bridge.py", line 127, in get_ports_list
    services = self.get_mdns_services()
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot/libraries/preferences_bridge.py", line 356, in get_mdns_services
    MDNS.start()
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot/libraries/mdns/mdns.py", line 27, in start
    self._zeroconf = Zeroconf()
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot/libraries/mdns/zeroconf.py", line 1610, in __init__
    interfaces = normalize_interface_choice()
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot/libraries/mdns/zeroconf.py", line 1551, in normalize_interface_choice
    sock.connect(("8.8.8.8", 80))
OSError: [Errno 51] Network is unreachable
```
### Expected Results

### Additional info

*You can edit the text above according to your need*

gepd, 2017-08-27:
I'm trying to reproduce this problem without success until now. 

It's shown after upload an sketch or after select a port?
I'm only disconnecting my laptop from internet, are you using a different method?

tablatronix, 2017-08-27:
I was joined to a wifi ap with no internet.

tablatronix, 2017-08-27:
sublime wont start now, ill look into it later

gepd, 2017-08-27:
Sublime crashes?

I've push some changes, but nothing related with this error

tablatronix, 2017-08-28:
That was bizarre, somehow my prefs got corrupted when I upgraded packages.

Still cannot reproduce, no idea why it was checking network during upload, or why it would fail.

Edge case, if it happens again I will repost.

tablatronix, 2017-08-28:
shit happened again, osx disabled wifi

clicked uplaod

tablatronix, 2017-08-28:
Ok reproducible now, when my wifi is off, i cannot upload or select serial ports

tablatronix, 2017-08-28:
ref
https://github.com/jstasiak/python-zeroconf/issues/25

gepd, 2017-08-28:
Can you check if the last commit solved the problem?

tablatronix, 2017-08-28:
What does this mean ?

```Exception in thread Thread-2:
Traceback (most recent call last):
  File "./python3.3/threading.py", line 901, in _bootstrap_inner
  File "./python3.3/threading.py", line 858, in run
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot/beginning/pio_install.py", line 193, in install
    save_env_paths(env_path)
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot/beginning/pio_install.py", line 369, in save_env_paths
    save_sysetting('env_path', paths)
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot/libraries/tools.py", line 155, in save_sysetting
    config.read(sys_path)
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot/libraries/configparser/__init__.py", line 705, in read
    self._read(fp, filename)
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot/libraries/configparser/__init__.py", line 1101, in _read
    raise MissingSectionHeaderError(fpname, lineno, line)
Deviot.libraries.configparser.MissingSectionHeaderError: File contains no section headers.
file: '/Users/shawn/Library/Application Support/Sublime Text 3/Packages/User/Deviot/deviot.ini', line: 1
```

tablatronix, 2017-08-28:
I can confirm this fix works.

although that new error after upgrading happens alot.

gepd, 2017-08-29:
great! I'll close this

closed 2017-08-29 by gepd