gepd/Deviot #157: dev corruption after upgrading w package manager

tablatronix, 2017-08-28:
Before report an issue, please make sure it hasn't been reported before.

### Operating system
osx
### Board used

### Description of problem
package manager upgrade package
### Steps to Reproduce

### Actual Results
deviot nor working
Traceback (most recent call last):
```
  File "./python3.3/threading.py", line 901, in _bootstrap_inner
  File "./python3.3/threading.py", line 858, in run
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot/beginning/pio_install.py", line 193, in install
    save_env_paths(env_path)
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot/beginning/pio_install.py", line 369, in save_env_paths
    save_sysetting('env_path', paths)
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot/libraries/tools.py", line 155, in save_sysetting
    config.read(sys_path)
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot/libraries/configparser/__init__.py", line 705, in read
    self._read(fp, filename)
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot/libraries/configparser/__init__.py", line 1101, in _read
    raise MissingSectionHeaderError(fpname, lineno, line)
Deviot.libraries.configparser.MissingSectionHeaderError: File contains no section headers.
file: '/Users/shawn/Library/Application Support/Sublime Text 3/Packages/User/Deviot/deviot.ini', line: 1
```
### Expected Results

### Additional info
blow away user folder to fix

*You can edit the text above according to your need*

gepd, 2017-08-29:
To solve that error, please remove the file `deviot.ini` located in `Packages/User/Deviot`

I've recently changed a library and it has compatibility problems with the old `deviot.ini` file
I'm currently working to solve it

gepd, 2017-09-11:
This was fixed with https://github.com/gepd/Deviot/commit/ef012b106773689536287d7b3db338ccdf42d5ae

Re open if you find any problem related with this

closed 2017-09-11 by gepd