gepd/Deviot #159: Installing on Windows with Python 2 and 3

Suyash458, 2017-09-02:
### Operating System
Windows 10

### Description of problem
I have Python 2 installed, but it's not in the system path because Python 3 is the default interpreter. Is there any way Deviot can detect the Python 2 installation?

gepd, 2017-09-02:
Where is located your Python2? 
It's usually installed in `C:\Python27` it's your case?

Suyash458, 2017-09-03:
I've installed Python 2 on a different drive, the path looks like `D:\Software\Python27`

gepd, 2017-09-03:
Please try this:

Open the file: `pio_install.py` locate in  and replace the line

Suyash458, 2017-09-03:
It works now, I get a message saying PlatformIO is installed. 
Can I use Deviot now? Or do I have to change directories elsewhere too?

Thank You!

gepd, 2017-09-03:
You should be able to use Deviot now, but if you find any problem post it here to fix it!

Suyash458, 2017-09-03:
Great, thanks again! I'll close this for now.

closed 2017-09-03 by Suyash458