gepd/Deviot #160: Update to develop instructions on wiki

jwhendy, 2017-09-06:
I just followed the [instructions here](https://github.com/gepd/Deviot/wiki/Setup) to use the `develop` branch of Deviot. The Package Control settings, at least on my system, already contained some lines. To adjust the instructions so that someone less familiar with dictionary structures can do this, I made a slight tweak to the wiki page.

Instead of this: 

```
{
"repositories": ["https://github.com/gepd/Deviot/tree/develop"]
}
```

I changed it to this:
```
{
    "bootstrapped": true,
    "in_process_packages":
    [
    ],

    "installed_packages":
    [
	"Package Control"
    ],

    "repositories": ["https://github.com/gepd/Deviot/tree/develop"]
}
```

I *think* this would show adding the `repositories` section to ST3 Package Control default user settings; to my recollection I've never touched them so this is what the user would be greeted with. This way they don't accidentally try to add a nested dictionary.

In googling around I found that github doesn't do wiki pull requests, so I followed the instructions [here](https://stackoverflow.com/questions/10642928/how-to-pull-request-a-wiki-page-on-github) and manually "forked" the wiki. You can see my updated page [here](https://github.com/jwhendy/deviot-wiki/blob/master/Setup.md).

gepd, 2017-09-07:
Thanks @jwhendy that is a good point, I've edited the wiki as you suggested and I've changed the permissions of the wiki to public so you can add/change anything from there 👍

jwhendy, 2017-09-07:
Awesome -- thanks!

closed 2017-09-07 by jwhendy