gepd/Deviot #162: Library path requirements between deviot and platformio

jwhendy, 2017-09-07:
This is related to #88 and some discussion in #154. 

My understanding is that deviot should now tolerate a typical platformio setup. Here is a way I went about setting up a reproducible example. It requires at least some external library; if you don't have one installed, you can use [this one](https://github.com/bogde/HX711), a pretty small library for reading a wheatstone bridge HX711 board.

Steps:
- setup the [developer repo](https://github.com/gepd/Deviot/wiki/Setup#developer)
- subl3: `Preferences -> Package Control -> Install -> Deviot 2017.09.07.03.02.41`. Output:
```
[ Deviot 2.1.1 ] Setup
10:15:34 Symlink has been detected
10:15:35 PlatformIO is installed in your system. You can enjoy Deviot now!
```
- ensure platformio is up to date: `Deviot -> Options -> Upgrade PlatformIO`
```
[ Deviot 2.1.1 ] Starting...
10:44:13 Searching for PlatformIO Updates...

You're up-to-date!
PlatformIO 3.4.1 is currently the newest version available.
```
- `Deviot -> New sketch -> deviot-lib-test`; then navigate to `~/Arduino` for the directory choice
- `Deviot -> Select board`; I choose Uno
- add library to use (HX711 or one of your choice):
```
//deviot-lib-test.ino
#include "HX711.h"

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}
```
- `Deviot -> Compile`: get error; this is also needed for Deviot to create `platformio.ini`
```
~/Arduino/deviot-lib-test/lib/HX711/HX711.cpp:2:19: fatal error: HX711.h: No such file or directory
#include <HX711.h>
^
compilation terminated.
```

- go to directory to make platformio compliant:
```
$ cd ~/Arduino/deviot-lib-test
$ mkdir src
$ mv deviot-lib-test.ino src/
$ ln -s src/deviot-lib-test.ino 
$ mkdir lib
$ ln -s ~/Arduino/libraries/HX711 ## or library of your choice
$ cp /tmp/Deviot/deviot-lib-test/platformio.ini ./
```
- show that platformio compiles:
```
$ platformio run
[ ... ]
[SUCCESS] Took 0.75 seconds
```
- re-try in Deviot now that we have a `platformio` compliant structure. Get same error:

[image]

So, as it is, it doesn't appear that Deviot will take accept the structure that works for `platformio`.

---

I did a second test, which *did* work:

- new sketch called `pio-struc`, located in `~/Arduino`
- `Deviot -> Use PlatformIO Structure`
- `Deviot -> Select board`: Arduino Uno
- `Deviot -> Compile` on blank template (seems required to trigger creation of `src`, `lib`, and `platformio.ino`
- Add `#include "HX711.h"` as above, save file
- link library:
```
$ cd ~/Arduino/pio-struct
$ ln -s ~/Arduino/libraries/HX711 lib/
```
- `Deviot -> Compile`, success.
```
[ Deviot 2.1.1 ] Starting...
[Thu Sep  7 10:51:04 2017] Processing uno (platform: atmelavr; board: uno; framework: arduino)
[ ... ]
[SUCCESS] Took 0.82 seconds
```
---

Based on conversations in #154 , I'm not sure if you meant that the first example *should* work, or only if one chooses the `platformio` structure option. I can see a few ways to approach...

- **bug:** if it works with `platformio`, it should work with Deviot
- **not a bug:** Deviot by default expects a simple Arduino like structure, and if the user wishes to use the other structure, he/she should specify it
- **not a bug:** from further reading, it appears that `platformio` doesn't really intend for `~/Arduino/libraries` to be linked in this folder (it's for truly custom libraries created for the project itself)... instead, I think the intended way is to use [`lib_extra_dirs`](http://docs.platformio.org/en/latest/projectconf/section_env_library.html#projectconf-lib-extra-dirs) which I verified is what Deviot does with `Deviot -> Library options -> Extra library folder`.
- **bug:** since Arduino doesn't require special behavior to load from `~/Arduino/libraries`, neither should Deviot (add this directory by default to `lib_extra_dirs` during install)

---

As a side note for compatibility, if you have this structure, compiling works with Deviot, `platformio` and Arduino simultaneously (omitted `lib` and `platformio.ini` for brevity).
```
[deviot-lib-test]$ tree
.
├── deviot-lib-test.ino -> src/deviot-lib-test.ino
└── src
    └── deviot-lib-test.ino
```

Interestingly, if you switch the real file (have it top level) and symlink (put in `src`), Arduino fails!
```
stat /home/jwhendy/Arduino/deviot-lib-test/src/deviot-lib-test.ino: too many levels of symbolic links
Error compiling for board Arduino/Genuino Uno.
```

Alright... that's what I've got. Hope that's helpful!

gepd, 2017-09-08:
Look how it works in my ubuntu VM

[image]

I'm thinking the problem can be other, what is your O.S?

jwhendy, 2017-09-08:
Hmm. That's interesting for sure. To be clear, you are *not* using the `Use PlatformIO Structure` option and it still works after linking? As I said above:

>...I'm not sure if you meant that the first example *should* work, or only if one chooses the platformio structure option.

I'm on arch linux.
```
$ uname -a
Linux whammy 4.12.10-1-ARCH #1 SMP PREEMPT Wed Aug 30 12:18:42 CEST 2017 x86_64 GNU/Linux
```

gepd, 2017-09-08:
I've check now with `Use PlatformIO Structure` activated, and it work as well. But I think I know what is the step you miss.

When you activate `Use PlatformIO Structure` The folder structure will change, if you have you sketch in:

`~/Desktop/sketch/deviot-lib-test.ino` Deviot will move the sketch to

jwhendy, 2017-09-08:
Ah, I know what the issue is. I got used to symlinking `dir/file.ino` to `dir/src/file.ino` due to previous issues. I thought Deviot wanted `dir/file.ino`, but it would fail on me due to libraries and then I'd give up and switch to using `platformio` directly, which would complain that there was nothing in `src`. So, I adopted this symlink pattern.

If I edit `dir/src/file.ino`, it works. To reproduce, with your structure as-is try:
```
$ cd test
$ ln -s src/test.ino
```

Now open `test/test.ino` and try to compile (without `Use PlatformIO Structure` option).

Not sure if you consider this a bug... but technically the file is there in `src`, I'm guessing that Deviot's path just isn't looking there if you're editing from one dir higher?

gepd, 2017-09-08:
If you don't have checked the `Use PlatformIO Structure` and there is no `platformio.ini` file in `dir/`, deviot will add the `src_dir` flag in the `platformio.ini` located in `tmp/Deviot/test/` with the path of your folder sketch. So will not matter if you have `dir/src/test.ino` or `dir/test.ino` the `dir/` folder will be use as the `src` folder without use the `src_dir` flag

let me know if you understand that, I think that explanation was a little confusing

jwhendy, 2017-09-08:
I don't totally get it...

- I don't have `Use PlatformIO Structure` checked
- there *is* a `platformio.ini` file present in `dir/`

I take it that deviot will, then, *not* add the `src_dir` flag. Indeed:
```
$ cat /tmp/Deviot/pio-struct-test/platformio.ini
[env:nanoatmega328]
platform = atmelavr
framework = arduino
board = nanoatmega328
```

Based on this, is it odd that I get the error? Without this `src_dir` flag, where does deviot look for `lib`? In my mind, I wondered if it would look in `../lib` if it expects that I'm editing `dir/src/file.ino`. In fact, I'm editing `dir/file.ino` so it should be looking in `./lib`.

gepd, 2017-09-08:
If you have:

```
├── lib
│   └── HX711 -> /home/jwhendy/Arduino/libraries/HX711
├── platformio.ini
└── src
    └── pio-struct-test.ino
```
Deviot will considerate this a `PlatformIO` project and will compile without problem

with:
```
├── lib
│   └── HX711 -> /home/jwhendy/Arduino/libraries/HX711
├── pio-struct-test.ino
├── platformio.ini
└── src
    └── (symlink) pio-struct-test.ino
```
This will work in `PlatformIO CLI` but deviot will move the sketch to the `src` folder, this is a bug as I said in https://github.com/gepd/Deviot/issues/165

In both cases, Deviot will not use the `tmp/Deviot` folder (even if it's already there) because the `platformio.ini` file is in the project folder and PlatformIO structure will always have priority, also it will not add the `src_folder`

But if you have:

```
├── lib
│   └── HX711 -> /home/jwhendy/Arduino/libraries/HX711
├── pio-struct-test.ino
└── src
    └── (symlink) pio-struct-test.ino
```
It won't be a PlatformIO project, so `platformio.ini` and the rest of the folders will be created (or use in case it already are there) in `tmp/Deviot/...`

That means you should use `tmp/Deviot/pio-struct-test/lib` to store your libraries (make a symlink)

But be aware, the `src_dir` flag will overwrite the `src` folder in your project, not the `lib` folder