gepd/Deviot #165: Having Deviot be more platformio compliant/tolerant

jwhendy, 2017-09-07:
In tinkering around with the `Use PlatformIO Structure` option, I was surprised that it overwrites things that are already there. In my mind, it would be awesome if Deviot were compatible with `platformio` and the Arduino IDE (based on the option). I think this could be similar to #137 in that things that are already fine as they are get rearranged a bit.

---

**Example 1**
Let's say one has been working with `platformio` and wants to try Deviot. They have an existing project structure like so:
```
[pio-struct-test]$ tree
├── lib
├── platformio.ini
└── src
    └── pio-struct-test.ino  ## just an empty setup{} and loop{}
```

Inside `platformio.ini`:
```
[env:nano]
platform = atmelavr
framework = arduino
board = nanoatmega328
```

Start ST3 and do `Deviot -> Compile`:
```
[ Deviot 2.1.1 ] Starting...
The current working directory ~/Arduino/pio-struct-test will be used for project.
[ ... ]
Error: Unknown environment names 'nanoatmega328'. Valid names are 'nano'
```

With `platformio` natively:
```
[pio-struct-test]$ platformio run
[Thu Sep  7 18:25:58 2017] Processing nano (platform: atmelavr; board: nanoatmega328; framework: arduino)
[ ... ]
[SUCCESS] Took 0.49 seconds
```

Oddly, creating a second project from scratch using `Deviot -> New sketch` did not prompt to ask for a board and populated the `platform.ini` automatically:
```
[env:nanoatmega328]
platform = atmelavr
board = nanoatmega328
framework = arduino
```

The `platformio` docs suggest that `env` [can be anything](http://docs.platformio.org/en/latest/projectconf/section_env.html), but Deviot seems to need `env` to match what it wants.

---

**Example 2**

If one has a top level `file.ino` file symlinked from `src/file.ino`, Deviot moves it anyway. The use case here is that the directory remains Arduino IDE compatible; if you try and open `src/file.ino`, Arduino will prompt to put it in a directory with the same name. Structure before:
```
[pio-struct-test]$ tree
├── lib
├── pio-struct-test.ino -> src/pio-struct-test.ino
├── platformio.ini
└── src
    └── pio-struct-test.ino
```

Start ST3, open the *top level* symlink, compile and the screen goes blank!
```
[pio-struct-test]$ tree
├── lib
├── platformio.ini
└── src
    └── pio-struct-test.ino -> src/pio-struct-test.ino
```

Deviot replaces the "real file" with a symlink pointing to itself. In my opinion, if the directory is already `platformio` compliant, nothing should be done.

---

Some ideas which I'd be happy to try and help on (with guidance/suggestions):

- [ ] have Deviot accept Arduino and `platformio` compatible directories (perhaps undesirable, but doing the symlink thing has been working for me to have a 'hybrid' structure)
- [ ] if the directory is already `platformio` compliant, don't touch it, or perhaps cache the contents and only add/update options if the user selects something new/different?
- [ ] don't overwrite existing `platformio.ini` files, or at least work with them if they are compliant
- [ ] make `Use PlatformIO Structure` a per-file setting. It appears to be universal at the moment; if the user *doesn't* want it made into this structure, they have to toggle the option on a per-file option
- [ ] alternatively to *any* changes... just add more documentation about what Deviot can and can't work with, what users can expect with the options, etc.

Let me know if there's anything of merit here. I'm learning `python` and have switched to using this for all of my micro-controller tinkering... I'm happy to try and contribute as I've been looking for `python` projects to work on anyway for better/faster hands-on learning :) Thanks for tolerating all my activity!

gepd, 2017-09-08:
The **Example 1** is a bug, it shouldn't be showing the error. It's caused because deviot will store the last environment you use in other project, so in that case is trying to compile with a environment not initialized, that is why the error is displayed. But what is weird to me, it should initialized the environment. but I'll check and fix this.

The **Example 2** is a case I didn't think of when I started the project, because I usually don't work with symlinks, however is an improvement I'll defenetelly add

gepd, 2017-09-08:
> have Deviot accept Arduino and platformio compatible directories (perhaps undesirable, but doing the symlink thing has been working for me to have a 'hybrid' structure)

Improving the **Example 2** will let to work with both cases

> if the directory is already platformio compliant, don't touch it, or perhaps cache the contents and only add/update options if the user selects something new/different?

I'll check this, but I'm almost sure if you open a `platformio` structure project, will not touch anything but the `platformio.ini`

> don't overwrite existing platformio.ini files, or at least work with them if they are compliant

How is that? platformio work with that file to compile/upload the sketch, or you mean do not add any extra flag?

> make Use PlatformIO Structure a per-file setting. It appears to be universal at the moment; if the user doesn't want it made into this structure, they have to toggle the option on a per-file option

Yes, at this moment it's universal... added to the to-do list

> alternatively to any changes... just add more documentation about what Deviot can and can't work with, what users can expect with the options, etc.

Yes, there is a lot things to do with that, I'm working in bug that has been took some weeks, and after that I'll try to add more attention to the documentation. 

I started this project to improve my python skill too, so don't worry. 

Any contribution is welcome, you can check the [sublime api](https://www.sublimetext.com/docs/3/api_reference.html), it may help you to understand the deviot code, and maybe to help me improve deviot

jwhendy, 2017-09-08:
> But what is weird to me, it should initialized the environment. but I'll check and fix this.

An observation that is possibly related:

- `Deviot -> Options -> Use PlatformIO Structure`
- `Deviot -> New sketch` and name/store it as whatever
- `Deviot -> Compile` will hang the first time:
```
[ Deviot 2.1.1 ] Starting...
The current working directory ~/Arduino/deviot-test-1 will be used for project.
[ ... ]
`platformio run --target clean` - clean project (remove compiled files)
`platformio run --help` - additional information
```
- `Deviot -> Compile` a second time immediately after, succeeds. Does it have to do with creating `dir/file.ino` and then compiling with the `PlatformIO Structure` option triggering the move to `src/`?

jwhendy, 2017-09-08:
>I'll check this, but I'm almost sure if you open a platformio structure project, will not touch anything but the platformio.ini

Unless `dir/file.ino` also exists. Then it will move to `dir/src/file.ino`, I believe.

How is that? platformio work with that file to compile/upload the sketch, or you mean do not add any extra flag?

>Per #137, it seemed like maybe `platformio.ini` was being edited undesirably, but my main point was `[env:foo]` shouldn't affect the function of the file. My file was compliant (worked with `platformio` directly), but Deviot wouldn't accept it (because it wanted `[env:nanoatmega328]`, not `[env:nano]`.

>Yes, there is a lot things to do with that, I'm working in bug that has been took some weeks, and after that I'll try to add more attention to the documentation.

I can help with that if you'd like. I haven't contributed heavily to many projects, so it can feel "weird" documenting someone else's creation, but if this is helpful count me in!

> I started this project to improve my python skill too, so don't worry. Any contribution is welcome, you can check the sublime api, it may help you to understand the deviot code, and maybe to help me improve deviot

Cool! I'm coming from mostly `R`, but haven't done much with objects at this point. There seems to be this huge wall between the tutorials I go through and `python` in the real world. I was looking through `Package Control` today and just couldn't follow along! Same here... it makes me wonder how you figure out how to even structure these projects :) I start seeing `__init__` and tons of `self` and just fall apart...

gepd, 2017-09-08:
> Unless dir/file.ino also exists. Then it will move it to dir/src/file.ino, I believe.

At this moment it will, but I'll improve that to do it only when not symlink is present

> Per #137, it seemed like maybe platformio.ini was being edited undesirably, but my main point was [env:foo] shouldn't affect the function of the file. My file was compliant (worked with platformio directly), but Deviot wouldn't accept it (because it wanted [env:nanoatmega328], not [env:nano].

Yes, that issue is fixed, and it was because the `src_dir` flag, but now deviot will add flag only while is compiling or uploading the firmware, when it finished, it will remove it automatically.

The issue with `[env:foo]` is a bug that I'll fix soon, meanwhile you can select the environment manually from `Deviot > Select Environment`, it will avoid that error

> I can help with that if you'd like. I haven't contributed heavily to many projects, so it can feel "weird" documenting someone else's creation, but if this is helpful count me in!

You can start with what's most comfortable for you, there's no pressure with that, I've screwed up many times, I've rewritten deviot about 3 times, I won't bother if someone else does it too, it's the only way to learn. 😄

gepd, 2017-09-10:
The problem with the symplink should be fixed now, please test it

gepd, 2018-02-17:
Most of the problem are fixed and will be included in v2.2, the new features and improvements will be added in v2.3 (I'll open a issue to track this)

closed 2018-02-17 by gepd