gepd/Deviot #166: Checkbox for Use PlatformIO Structure menu option

jwhendy, 2017-09-08:
In [this post](https://github.com/gepd/Deviot/issues/137#issuecomment-318223622) of #137 the `Use PlatformIO Structure` option shows a checkbox. I didn't have this on my system... I did find that changing `presets/main_menu.json` adds it:
```
                               {
                                    "caption":"menu_always_native",
                                    "checkbox":true,
                                    "command":"deviot_pio_structure"
                                },
```

 Any reason not to have one? Since the behavior is quite different (build dir, where the `.ino` file lives), I think knowing which setting is active would be helpful. I can submit a PR since I already did it; you can decline it if I don't understand why you don' currently have this.

gepd, 2017-09-08:
I should have missed that, do the checkbox work after you added `"checkbox":true`?

jwhendy, 2017-09-08:
Indeed!

As is:
[image]

With `"checkbox":true`, unchecked:
[image]

Checked:
[image]

Quick tests suggest the behavior works (checked, uses pio structure, unchecked doesn't) and it also stays consistent between shutdown/restart (persists).

gepd, 2017-09-08:
I haven't noted this because on windows it has a diffent behavior, I've accepted the PR

Thanks!

jwhendy, 2017-09-08:
I think the screenshot I saw was on Windows actually... it looked like a great thing to have and so I just had to see how it was working with other options that use it. Happy to help!

closed 2017-09-08 by jwhendy