gepd/Deviot #169: folder issue

clarkill78, 2017-09-09:
Hi,

I was using deviot in sublime text 3 perfectly and today it does not work anymore :( I have updated platformio core to 4.3.1.

After installed deviot in sublime text 3, it looks to 'deviot' folder and now I have 'deviot (arduino ide)' folder instead. If I rename this folder to 'deviot', close ST3 and restrt, the menu is back in ST3, until it renames the folder back with (arduino ide).
I have the following error: error loading syntax file "packages/deviot/console/tmlLanguage". It is due to the wrong folder name.

How can I fix it?
Thx in advance

gepd, 2017-09-09:
Thanks for the report @clarkill78 
I'm working now in that issue, I'll let you know when I push the fix(es)

gepd, 2017-09-10:
Not it should be fixed :)

clarkill78, 2017-09-10:
Hi,

thanks for your reactivity!

I restarted from a fresh installation, just to be sure. 
Launched a fresh install of ST3, install deviot 2.1.1, apply the changes you have made on the three python files. 
I do not have the menu 'deviot' in ST3. I can compile and download code (normal) but impossible to have the menu. I tried to rename 'deviot (arduino ide)' to 'deviot' but it does not change the behavior.

gepd, 2017-09-10:
I'm sorry, I forgot to metion it's only available in the develop branch now. 

Follow [this](https://github.com/gepd/Deviot/wiki/Setup#developer) steps to install it.

Note: At this moment there is an issue so, after add the develop url into `repositories`, remove the plugin, and when you search the plugin again (`Package Control > Install Package`), you will have two versions; `Deviot (Arduino IDE)` and `Deviot`. Install the last one.

gepd, 2017-09-12:
I've launched a new release with all fixes, please reopen this issue if you keep having problems

closed 2017-09-12 by gepd