gepd/Deviot #170: always compile project not focused asset

tablatronix, 2017-09-16:
Is there a way to have a "project", so that when you compile it compiles a specific file, not the current focused file ?

Is there say anyway I could have all assets open, be editing a included library, and when i compile it compiles the primary assert via some flag, or ini setting ?

tablatronix, 2017-09-16:
Or even easier to implement, optionally open a dialog select to pick which open file to compile, and remember the last one autoselected

gepd, 2017-09-16:
PlatformIO is project based, you can open any file inside of the source folder and compile it.

There was an issue when I launched v2.0 but it was fixed, have you tried to compile with other file?

tablatronix, 2017-09-16:
I guess i need to look into the pio way of doing this, these are not in the same source folder or even project, I am working on dependancies external to the project but need to compile an arduino sketch for example to test it, or some external unit tests,   so I have to go back to that file  ( the includee) and compile there, was looking for a way to hit compile anywhere and it could always compile that file instead.

gepd, 2017-09-16:
What is your O.S?

tablatronix, 2017-09-16:
I use pc and mac

gepd, 2017-09-16:
In mac you can use symplink to the external files

But could be easier and cross-platform to have all your external sources in the same folder, (like libraries) and then use the `lib_dir` flag.

In deviot you can assign a folder from `Deviot Menu > Library Options > Extra Library Folder`, this option will add the `lib_dir` flag on each environment when the project is compiled, but you can also edit the `platformio.ini` file (`Devito Menu > Open Build Config File`) and add
```
[common]
lib_dir =
pathtoexternalsource1
pathtoexternalsource2
...
```
That way it will add the external source on each environment

tablatronix, 2017-09-16:
Yeah that is how i use libraries, but how does thst address making compile global so it does not compile the cpp or h file but the main cpp or ino

tablatronix, 2017-09-16:
Added example of what I am specifically talking about, in case it is still not clear.

gepd, 2017-09-17:
I see, what do you think of an option to "freeze" deviot on a specific project? so it wouldn't matter what file you are working on, it will always compile the same project.

The steps could be this:

1. Open/Select the file to "Freese"
2. Select the option to freeze

tablatronix, 2017-09-17:
Would this affect multiple instances of sublime text , that would be bad.

I think that would be acceptable, i mean you would only turn it on and off when you needed something like that, say you are working on (compiling/uploading) a library example file, and debugging the cpp file for a bug in the actual library, this would help with the back and forth, or focused window, also i cant tell you how many times I "compile" the serial terminal, or a .h file on accident.

gepd, 2017-09-17:
The popup dialog is also possible, it should list only the files previously compiled or all projects currently open? This should be enable/disable from an option too

what do you think?

tablatronix, 2017-09-17:
Maybe previous compiled and currently open in current session/instance only

tablatronix, 2017-09-17:
And yes optional in configmor menu you dont want this always on

tablatronix, 2017-09-17:
Oh and why does deviot not have a standard package ini file like most plugins?

gepd, 2017-09-17:
> Oh and why does deviot not have a standard package ini file like most plugins?

Which one do you mean?

tablatronix, 2017-09-17:
The package control settings  config files that other plugins use to control more advanced preferences

gepd, 2017-09-18:
Mainly because Deviot use a own menu, most of the plugins uses the config file because there is not other way to access to the options, but in this case you can edit each parameter of the plugin from the menu. Most the options are updated automatically.

Which option would you include in the configuration file?

tablatronix, 2017-09-18:
Most plugins use the prefs menu to allow advanced configs, and their key binding files. Users know how to back these up and transfer them. I don't think you need to put every option in the menu as this gets more advanced and users get picky, would be a good place to allow users to modify baud rate lists, verbose modes, and other things users want hacked in etc.

gepd, 2017-09-20:
At this moment Deviot haven't the menu option to edit the key bindings file, but you can always edit them from `Packages/Deviot (Arduino IDE)/Shortcuts`. 

In the case of the config file, I've to create the template with all the available options, but technically you can edit the file in `Packages/User/deviot.sublime-settings` as you want. I haven't done it because time reason. I've given priority to GUI options because Deviot born as an alternative to the Arduino IDE, and the idea is to keep it easy.

I've pushed a new commit with the freeze feature. (Deviot 2.2.0-dev.3)
It works as I wrote before.

1. Select the tab with the sketch to "freeze"
2. Go to `Deviot Menu > Freeze Sketch`
3. Now Deviot will compile the same sketch from any tab.

Note: You can unfreeze the sketch from any tab

tablatronix, 2017-09-21:
Ok i was curious maybe throw it on a milestone or a roadmap ?

gepd, 2017-09-21:
Definitelly, I'll first finish to write the documentation in the wiki and I'll start to solve this kind of issues.

Let me know if the new feature is working well

tablatronix, 2017-09-21:
Yeah works pretty great!

gepd, 2017-09-21:
So I'll close this issue, re-open it if you find any problem

closed 2017-09-21 by gepd