gepd/Deviot #171: development

tablatronix, 2017-09-16:
Before report an issue, please make sure it hasn't been reported before.

### Operating system

### Board used

### Description of problem
serial monitor stops on compile error
I noticed the serial monitor stops running on a compile error.

### Error(s) in ST Console? (View > Show Console)

### Steps to Reproduce

### Actual Results

### Expected Results

### Additional info

*You can edit the text above according to your need*

tablatronix, 2017-09-16:
i will reproduce and post info, Not sure if its always, but often enough I have noticed

tablatronix, 2017-11-12:
Any idea where I should look to fix this. Its driving me insane.

gepd, 2017-11-13:
Do you have now, the steps to reproduce this?

tablatronix, 2017-11-13:
Yeah it happens if serial is running, and there is a compile or upload failure.

gepd, 2017-11-16:
Try removing this line: https://github.com/gepd/Deviot/blob/develop/platformio/upload.py#L82

I'm waiting for free time to push all new changes I already have, I hope to do it soon, I'll definitely have a final solution for this

tablatronix, 2018-02-02:
is this a platformio bug ?

gepd, 2018-02-12:
fixed in `dev-14`

closed 2018-02-12 by gepd