gepd/Deviot #175: feature re-upload / upload without compile

tablatronix, 2017-09-22:
Is it possible to retry uploads, or add an upload only menu option ?

gepd, 2017-09-25:
I'm CHecking this and I realize that Deviot have a bug that makes re-compile the code each time even if there is not change in the code.

In PIO CLI it hasn't the same behavior, so I'll try to find where is the problem. There is not option to only upload without compile in PIO, but solving this problem should be the same.

tablatronix, 2017-09-25:
Gotcha I was wondering why it always compiles, thought it was just the way it was written, great

gepd, 2017-09-26:
This problem took longer than I expected, I had to make deeper modifications to solve this.

There is a new option in `Deviot Menu> Compile / Upload Options> platformio.ini untouch` (disabled by default) That will remove any option / flag added by Deviot before compile or upload.

What does it matter in this issue? PlatformIO will re-compile the project, if the source files or `platformio.ini` are edited, this will happen even if the file has the same information. I suppose it generate the hash based on the last datetime modification of the project files.

So if you have `platformio.ini untouch` enabled, it will almost always re-compile the code.

Without this option actived, this should not happen because Deviot will not remove the flags after compile / upload, like the extra libraries folder flag, or overwrite the upload baudrate, etc.

closed 2018-02-12 by gepd