gepd/Deviot #176: Incompatible with python 2.7.4

tarzan115, 2017-09-25:
### Operating system
`Windows 10`
`Sublime Text 3 build 3143`
### Board used
`ESP 8266`
### Description of problem
I used python version 2.7.3 is work fine. but when I upgrade to version 2.7.4 I got this error.
### Error(s) in ST Console? (View > Show Console)
```
Traceback (most recent call last):
  File "C:\Python27\Lib\runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "C:\Python27\Lib\runpy.py", line 72, in _run_code
    exec code in run_globals
  File "C:\Users\doanh\AppData\Roaming\Sublime Text 3\Packages\User\Deviot\penv\Scripts\platformio.exe\__main__.py", line 5, in <module>
  File "c:\users\doanh\appdata\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\platformio\__main__.py", line 21, in <module>
    import click
  File "c:\users\doanh\appdata\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\click\__init__.py", line 18, in <module>
    from .core import Context, BaseCommand, Command, MultiCommand, Group, \
  File "c:\users\doanh\appdata\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\click\core.py", line 8, in <module>
    from .types import convert_type, IntRange, BOOL
  File "c:\users\doanh\appdata\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\click\types.py", line 5, in <module>
    from ._compat import open_stream, text_type, filename_to_ui, \
  File "c:\users\doanh\appdata\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\click\_compat.py", line 6, in <module>
    from weakref import WeakKeyDictionary
  File "C:\Python27\Lib\weakref.py", line 14, in <module>
    from _weakref import (
ImportError: cannot import name _remove_dead_weakref
```

gepd, 2017-09-25:
Do you mean `2.7.13` and `2.7.14` ?

tarzan115, 2017-09-25:
I already installed `Python 2.7.13` but after installing `Python 2.7.14` I got that error.

gepd, 2017-09-25:
Can you please try installing platformio (in Python `2.7.14`) from the console with:

`pip install -U platformio`

after it finished, run:

`pio --version` 

If everything is ok you will see:

`PlatformIO, version 3.4.1`

enderman08, 2017-12-19:
same, linux debian

gepd, 2017-12-20:
@enderman08 with the same python version? (`2.7.14`)

enderman08, 2017-12-22:
yes

enderman08, 2017-12-27:
here what it says
`pio --help 
Traceback (most recent call last):
  File "/home/ender/.config/sublime-text-3/Packages/User/Deviot/penv/bin/platformio", line 7, in <module>
    from platformio.__main__ import main
  File "/home/ender/.config/sublime-text-3/Packages/User/Deviot/penv/local/lib/python2.7/site-packages/platformio/__main__.py", line 21, in <module>
    import click
  File "/home/ender/.config/sublime-text-3/Packages/User/Deviot/penv/local/lib/python2.7/site-packages/click/__init__.py", line 18, in <module>
    from .core import Context, BaseCommand, Command, MultiCommand, Group, \
  File "/home/ender/.config/sublime-text-3/Packages/User/Deviot/penv/local/lib/python2.7/site-packages/click/core.py", line 8, in <module>
    from .types import convert_type, IntRange, BOOL
  File "/home/ender/.config/sublime-text-3/Packages/User/Deviot/penv/local/lib/python2.7/site-packages/click/types.py", line 5, in <module>
    from ._compat import open_stream, text_type, filename_to_ui, \
  File "/home/ender/.config/sublime-text-3/Packages/User/Deviot/penv/local/lib/python2.7/site-packages/click/_compat.py", line 6, in <module>
    from weakref import WeakKeyDictionary
  File "/usr/lib/python2.7/weakref.py", line 14, in <module>
    from _weakref import (
ImportError: cannot import name _remove_dead_weakref`

gepd, 2017-12-29:
can you run this command in your console: `/home/ender/.config/sublime-text-3/Packages/User/Deviot/penv/local/lib/python2.7/site-packages/platformio --help` and let me know if you see the same error?

enderman08, 2017-12-29:
but platformio is a directory
I also reinstalled deviot in a fit of rage pio updated itself and there is no error, strange

gepd, 2017-12-29:
It may be an incompatibility with platformio fixed in a later version, I'll wait until @tarzan115 or someone else confirm this

gepd, 2018-02-11:
Re-open if you have new info about this issue

closed 2018-02-11 by gepd

Suyash458, 2018-06-09:
Facing the same issue while trying to upload to a teensy 3.2 on Win10 with Python 2 v2.7.14 and PlatformIO v3.5.3. Any leads?