gepd/Deviot #177: pio install errors during deviot install

rkinnett, 2017-09-25:
During deviot installation on two separate Windows machines (7 and 10), installation failed at multiple points during dependency installation, with "Error 5" Access Denied errors.  Encountered the same installation errors whether installing via package manager, the latest release (2.1.5), or current master.  Verified python root and scripts are in path.

### Potential Solution:
Installation completed successfully after selecting "Use Development PIO Version" in Deviot->Options menu.  I'm now happily flashing OTA.  Although the problem seems to have been with a dependency and not deviot directly, I wanted to leave a note here in  case anyone else encounters the same issue.  

By the way, I tried uninstalling deviot to reproduce the error for this writeup but could no longer reproduce the errors after deviot had previously successfully installed.  I imagine there must be some configuration or platformio files leftover after deviot uninstallation that helps with reinstallation.  Leftover files aside, I'm mentioning this because it may be hard to reproduce on any PC on which deviot has already been installed(/uninstalled).

Thanks @gepd for the plugin and your hard work keeping this going!

### First case:  Windows 10, Python 2.7.13
(Note: I didn't save the log verbatim but recall clearly the last 2 lines after hours of searching online for similar errors)
```[ Deviot 2.1.5 ] Setup
22:32:02 PlatformIO is not installed
Preparing to install it. Please be patient it; may take a while.
Downloading necessary files...
Extracting files...
Installing PlatformIO...
There was an error
New python executable in C:\Users\Ryan\AppData\Roaming\Sublime Text 3\Packages\User\Deviot\penv\Scripts\python.exe
ERROR: The executable "C:\Users\Ryan\AppData\Roaming\Sublime Text 3\Packages\User\Deviot\penv\Scripts\python.exe" could not be run: [Error 5] Access Denied
```

### Second Case:  Windows 7, Python 2.7.14
```
[ Deviot 2.1.5 ] Setup
22:32:02 PlatformIO is not installed
Preparing to install it. Please be patient it; may take a while.
22:32:02 Downloading necessary files...
22:32:04 Extracting files...
22:32:04 Installing PlatformIO...
22:32:15 There was an error
22:32:15 New python executable in C:\Users\Ryan\AppData\Roaming\SUBLIM~1\Packages\User\Deviot\penv\Scripts\python.exe
Installing setuptools, pip, wheel...
  Complete output from command C:\Users\Ryan\AppDat...v\Scripts\python.exe -c "import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip wheel:
  Collecting setuptools
  Downloading setuptools-36.5.0-py2.py3-none-any.whl (478kB)
Collecting pip
  Downloading pip-9.0.1-py2.py3-none-any.whl (1.3MB)
Collecting wheel
  Downloading wheel-0.30.0-py2.py3-none-any.whl (49kB)
Installing collected packages: setuptools, pip, wheel
Exception:
Traceback (most recent call last):
  File "C:\Users\Ryan\AppData\Roaming\Sublime Text 3\Packages\User\Deviot\penv\virtualenv\virtualenv_support\pip-8.0.2-py2.py3-none-any.whl\pip\basecommand.py", line 209, in main
    status = self.run(options, args)
  File "C:\Users\Ryan\AppData\Roaming\Sublime Text 3\Packages\User\Deviot\penv\virtualenv\virtualenv_support\pip-8.0.2-py2.py3-none-any.whl\pip\commands\install.py", line 317, in run
    prefix=options.prefix_path,
  File "C:\Users\Ryan\AppData\Roaming\Sublime Text 3\Packages\User\Deviot\penv\virtualenv\virtualenv_support\pip-8.0.2-py2.py3-none-any.whl\pip\req\req_set.py", line 731, in install
    **kwargs
  File "C:\Users\Ryan\AppData\Roaming\Sublime Text 3\Packages\User\Deviot\penv\virtualenv\virtualenv_support\pip-8.0.2-py2.py3-none-any.whl\pip\req\req_install.py", line 841, in install
    self.move_wheel_files(self.source_dir, root=root, prefix=prefix)
  File "C:\Users\Ryan\AppData\Roaming\Sublime Text 3\Packages\User\Deviot\penv\virtualenv\virtualenv_support\pip-8.0.2-py2.py3-none-any.whl\pip\req\req_install.py", line 1040, in move_wheel_files
    isolated=self.isolated,
  File "C:\Users\Ryan\AppData\Roaming\Sublime Text 3\Packages\User\Deviot\penv\virtualenv\virtualenv_support\pip-8.0.2-py2.py3-none-any.whl\pip\wheel.py", line 525, in move_wheel_files
    shutil.move(temp_record, record)
  File "C:\Python27\Lib\shutil.py", line 317, in move
    os.unlink(src)
WindowsError: [Error 5] Access is denied: 'C:\\Users\\Ryan\\AppData\\Roaming\\SUBLIM~1\\Packages\\User\\Deviot\\penv\\Lib\\site-packages\\setuptools-36.5.0.dist-info\\RECORD.pip'
----------------------------------------
...Installing setuptools, pip, wheel...done.
Traceback (most recent call last):
  File "virtualenv.py", line 2304, in <module>
    main()
  File "virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "virtualenv.py", line 924, in create_environment
    download=download,
  File "virtualenv.py", line 877, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env)
  File "virtualenv.py", line 783, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command C:\Users\Ryan\AppDat...v\Scripts\python.exe -c "import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip wheel failed with error code 2
```

gepd, 2017-09-25:
Thanks for your repport @rkinnett 

In https://github.com/gepd/Deviot/issues/9 someone repported the same problem but as you said, it's very difficult to reproduce in a machine with deviot previously successfully installed.

I suspect It can be the location of the dependencies, I'll try moving it to `~/.platformio` to see if this solves the problem

fiveangle, 2017-10-07:
I encountered the same.  Gave up and now use only on my Mac :)