gepd/Deviot #179: DEV installing staging platforms issue?

tablatronix, 2017-09-26:
platformio says to use staging for esp8266 you can use this in ini

`platform = https://github.com/platformio/platform-espressif8266.git#feature/stage`

http://docs.platformio.org/en/latest/platforms/espressif8266.html#using-arduino-framework-with-staging-version

```
[ Deviot 2.2.0-dev.3 ] Starting...
[09/26/17 10:06:42] Processing huzzah (build_flags: -DDEBUG_ESP_PORT=Serial -DDEBUG_ESP_CORE -DDEBUG_ESP_WIFI; platform: https://github.com/platformio/platform-espressif8266.git#feature/stage; upload_speed: 115200; board: huzzah; framework: arduino; lib_extra_dirs: C:\Users\alverson\Documents\projects\arduino\sketches\libraries)
PlatformManager: Installing platform-espressif8266
espressif8266_stage @ 1569289 is already installed
The platform 'https://github.com/platformio/platform-espressif8266.git#feature/stage' has been successfully installed!
The rest of packages will be installed automatically depending on your build environment.
Error: Unknown development platform 'https://github.com/platformio/platform-espressif8266.git#feature/stage'
```

Do you know if there is a way to make this work, or install this along side stable ?

tablatronix, 2017-09-26:
It looks like it creates a new platform folder, I set the platform to the folder name instead
`espressif8266_stage` and it seems to be working

gepd, 2017-09-26:
Seems that it could be installing this way too:

`pio platform install https://github.com/platformio/platform-espressif8266.git#feature/stage `

I guess there is a missing step in the platformio docs, as you said after use `espressif8266_stage` it works without problem

```
[env:nodemcuv2]
platform = espressif8266_stage
board = nodemcuv2
framework = arduino
```

tablatronix, 2017-09-26:
Well they said they updated the docs to this new easier method, so it makes me think they are handling this automatically now, or installing to a tmp base, I also have a folder called
`espressif8266@src-eb7495f88eb0afa18fedff98bfb5e40f`

Not sure what I did, Id have to wipe them out and retest to see.

tablatronix, 2018-01-25:
After updating platformio this worked flawlessly.

eg.
http://docs.platformio.org/en/latest/platforms/espressif32.html#using-arduino-framework-with-staging-version

tablatronix, 2018-01-28:
i cannot seem to get tags to work

`platform = git+https://github.com/platformio/platform-espressif8266#2.4.0`

PlatformManager: Installing platform-espressif8266
git version 2.14.3 (Apple Git-98)
Cloning into '/Users/shawn/.platformio/platforms/_tmp_installing-xe4Z8J-package'...
warning: Could not find remote branch 2.4.0 to clone.
fatal: Remote branch 2.4.0 not found in upstream origin
Error: VCS: Could not process command ['git', 'clone', '--recursive', '--depth', '1', '--branch', '2.4.0', 'https://github.com/platformio/platform-espressif8266', '/Users/shawn/.platformio/platforms/_tmp_installing-xe4Z8J-package']

tablatronix, 2018-01-28:
ugh it looks like you have to use platrformio platforms

https://github.com/platformio/platform-espressif8266/releases

which are not tagged the same for some stupid reason

gepd, 2018-01-28:
I've noted you have also missed the `v` in the tag, it should be:

`platform = git+https://github.com/platformio/platform-espressif8266#v2.4.0`

tablatronix, 2018-01-28:
yeah I think i got it figured out, it is quite the pita.

I will keep this open, i am going to try to post some useful info for anyone else having the same issues.

It looks like the easiest way to deal with this, is always use staging, then manually change the versions or manually checkout specific stuff.

Another problem I was having that was causing me confusion is that core_version does not always work ( it uses a bash command in platform.json to generate the global )
`ARDUINO_ESP8266_GIT_VER`