gepd/Deviot #182: DEV Serial options greyed out

tablatronix, 2017-09-28:
Serial options

Clean Monitor View
and 
Send 
are greyed out for me.

gepd, 2017-09-28:
That options are only available when the monitor serial is running.
In your case it's greyed out while the monitor is running?

tablatronix, 2017-09-28:
It was running, 

its working now, It must have been stuck in some condition where it thought it was not.

I will see if it happens again, might have been a fluke related to the other error uploading and how it stops the serial monitor

gepd, 2017-09-28:
It could be, when you are running the Monitor Serial and your upload fails, it wont start again automatically

tablatronix, 2017-09-28:
yeah i know same with compile fail
#171

gepd, 2017-09-28:
It shouldn't happen at compile, because the compilation doesn't use the serial port. At least with the upload, it's programed to work that way. The other could be a bug

closed 2017-09-29 by gepd