gepd/Deviot #183: DEV cant upload

tablatronix, 2017-09-28:
```
[ Deviot 2.2.0-dev.7 ] Starting...
[09/28/17 11:23:40] Processing huzzah (upload_speed: 576000, 115200; platform: espressif8266_stage; build_flags: -DDEBUG_ESP_PORT=Serial -DDEBUG_ESP_CORE -DDEBUG_ESP_WIFI; board: huzzah; framework: arduino)
Verbose mode can be enabled via `-v, --verbose` option
Converting espsmartconfig.ino
Collected 67 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <ESP8266WiFi> v1.0
Compiling .pioenvs\huzzah\src\espsmartconfig.ino.o
Looking for upload port...
Use manually specified: COM4
Uploading .pioenvs\huzzah\firmware.bin
*** [upload] Error -1
 [ERROR] Took 2.12 seconds 
```

??

tablatronix, 2017-09-28:
ini file is wierd

` upload_speed = 576000, 115200`

tablatronix, 2017-09-28:
Looks like overwrite upload speed, appends to upload_speed, which is invalid build option, I guess this is related to the new untouch fearture

gepd, 2017-09-28:
Do you have selected a diferent `upload_speed`?

tablatronix, 2017-09-28:
yes if you manually enter an upload_speed, and use overwrite with a different baud , it causes this string

gepd, 2017-09-28:
Try the new changes in `dev.9`

tablatronix, 2017-09-29:
Seems ok, but what is the desired behavior, ignore overwrite when set in ini, or always override ( this seems complicated unless you can do it inline, else you gotta overwrite the file, and comment out old or something crazy)

closed 2018-02-17 by gepd