gepd/Deviot #186: DEV Serial newlines skipping

tablatronix, 2017-10-06:
super minor issue, but posting for posterity.

This might be a speed issue or unavoidable, but I noticed if I am outputting to serial console at 115200 every say 50ms

Maybe some room for optimization as it does not miss data, but newlines only from what I can see.

It misses some newlines

```
void setup(){
	Serial.begin(115200);
}

void loop(){
	Serial.println("newline");
	delay(50);
}
```

```
newline
newlinenewline
newline
newline
newline
newline
newline
newline
newline
newline
newline
newline
newline
newline
newline
newline
newline
newline
newline
newline
newline
newline
newline
newlinenewline
newline
newline
```

gepd, 2017-10-06:
I've also noticed this but haven't had time to fix it

gepd, 2018-02-11:
I've released `dev13` and this shouldn't keep happening, let me know if you note the change

tablatronix, 2018-02-11:
Will do.
I saw you starred a fork for wifimanager esp32,i am the lead dev atm for the main repo, check out https://github.com/tzapu/WiFiManager/issues/513

gepd, 2018-02-11:
Thanks!

I'll check that. I'm currently testing the zhouhan branch, but comparing the code, seems like tzapu branch is better, I'll report any problem there...

closed 2018-02-17 by gepd