gepd/Deviot #188: Deviot doesn't upload

texjoachim, 2017-10-08:
### Operating system
openSUSE Tumbleweed

### Board used
Arduino Uno (Chinese clone)

### Description of problem
Deviot compiles the sketch, but doesn't upload it.

### Error(s) in ST Console? (View > Show Console)
Only:
---
[ Deviot 2.1.6 ] Starting...
---

### Steps to Reproduce
1. Write sketch.
2. Compile sketch.
3. Try to upload sketch.

### Actual Results
Nothing gets uploaded.

### Expected Results
Sketch gets uploaded to board.

### Additional info
Uploading works via Atom (Platformio) and VSCode (Platformio).

texjoachim, 2017-10-08:
I can, btw. compile via Deviot - Compile and then upload it to the Arduino via commandline just fine:

```
joachim@gogmazios:~/Dokumente/Arduino/TestDeviot|
⇒  platformio run --target upload
[Sun Oct  8 16:52:56 2017] Processing uno (platform: atmelavr; board: uno; framework: arduino)
------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Converting TestDeviot.ino
Collected 24 compatible libraries
Looking for dependencies...
No dependencies
Compiling .pioenvs/uno/src/TestDeviot.ino.o
Archiving .pioenvs/uno/libFrameworkArduinoVariant.a
Indexing .pioenvs/uno/libFrameworkArduinoVariant.a
Compiling .pioenvs/uno/FrameworkArduino/CDC.o
Compiling .pioenvs/uno/FrameworkArduino/HardwareSerial.o
Compiling .pioenvs/uno/FrameworkArduino/HardwareSerial0.o
Compiling .pioenvs/uno/FrameworkArduino/HardwareSerial1.o
Compiling .pioenvs/uno/FrameworkArduino/HardwareSerial2.o
Compiling .pioenvs/uno/FrameworkArduino/HardwareSerial3.o
Compiling .pioenvs/uno/FrameworkArduino/IPAddress.o
Compiling .pioenvs/uno/FrameworkArduino/PluggableUSB.o
Compiling .pioenvs/uno/FrameworkArduino/Print.o
Compiling .pioenvs/uno/FrameworkArduino/Stream.o
Compiling .pioenvs/uno/FrameworkArduino/Tone.o
Compiling .pioenvs/uno/FrameworkArduino/USBCore.o
Compiling .pioenvs/uno/FrameworkArduino/WInterrupts.o
Compiling .pioenvs/uno/FrameworkArduino/WMath.o
Compiling .pioenvs/uno/FrameworkArduino/WString.o
Compiling .pioenvs/uno/FrameworkArduino/_wiring_pulse.o
Compiling .pioenvs/uno/FrameworkArduino/abi.o
Compiling .pioenvs/uno/FrameworkArduino/hooks.o
Compiling .pioenvs/uno/FrameworkArduino/main.o
Compiling .pioenvs/uno/FrameworkArduino/new.o
Compiling .pioenvs/uno/FrameworkArduino/wiring.o
Compiling .pioenvs/uno/FrameworkArduino/wiring_analog.o
Compiling .pioenvs/uno/FrameworkArduino/wiring_digital.o
Compiling .pioenvs/uno/FrameworkArduino/wiring_pulse.o
Compiling .pioenvs/uno/FrameworkArduino/wiring_shift.o
Archiving .pioenvs/uno/libFrameworkArduino.a
Indexing .pioenvs/uno/libFrameworkArduino.a
Linking .pioenvs/uno/firmware.elf
Checking program size
text	   data	    bss	    dec	    hex	filename
948	      0	      9	    957	    3bd	.pioenvs/uno/firmware.elf
Building .pioenvs/uno/firmware.hex
Looking for upload port...
Auto-detected: /dev/ttyUSB0
Uploading .pioenvs/uno/firmware.hex

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: reading input file ".pioenvs/uno/firmware.hex"
avrdude: writing flash (948 bytes):

Writing | ################################################## | 100% 0.16s

avrdude: 948 bytes of flash written
avrdude: verifying flash memory against .pioenvs/uno/firmware.hex:
avrdude: load data flash data from input file .pioenvs/uno/firmware.hex:
avrdude: input file .pioenvs/uno/firmware.hex contains 948 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.13s

avrdude: verifying ...
avrdude: 948 bytes of flash verified

avrdude: safemode: Fuses OK (E:00, H:00, L:00)

avrdude done.  Thank you.

============================================= [SUCCESS] Took 2.43 seconds =============================================
joachim@gogmazios:~/Dokumente/Arduino/TestDeviot|
```

gepd, 2017-10-08:
After try to upload, can you please copy here the output of the sublime console `ST Menu > View > Show Console` (It's different from Deviot console)

texjoachim, 2017-10-09:
This is what I get:

```
DPI scale: 1
startup, version: 3143 linux x64 channel: stable
executable: /opt/sublime_text/sublime_text
working dir: /
packages path: /home/joachim/.config/sublime-text-3/Packages
state path: /home/joachim/.config/sublime-text-3/Local
zip path: /opt/sublime_text/Packages
zip path: /home/joachim/.config/sublime-text-3/Installed Packages
ignored_packages: ["Vintage"]
pre session restore time: 0.136408
startup time: 0.70705
first paint time: 0.708977
reloading plugin Default.auto_indent_tag
reloading plugin Default.block
reloading plugin Default.comment
reloading plugin Default.convert_syntax
reloading plugin Default.copy_path
reloading plugin Default.delete_word
reloading plugin Default.detect_indentation
reloading plugin Default.duplicate_line
reloading plugin Default.echo
reloading plugin Default.exec
reloading plugin Default.fold
reloading plugin Default.font
reloading plugin Default.goto_line
reloading plugin Default.history_list
reloading plugin Default.indentation
reloading plugin Default.install_package_control
reloading plugin Default.kill_ring
reloading plugin Default.mark
reloading plugin Default.new_templates
reloading plugin Default.open_context_url
reloading plugin Default.open_in_browser
reloading plugin Default.pane
reloading plugin Default.paragraph
reloading plugin Default.paste_from_history
reloading plugin Default.profile
reloading plugin Default.quick_panel
reloading plugin Default.run_syntax_tests
reloading plugin Default.save_on_focus_lost
reloading plugin Default.scroll
reloading plugin Default.set_unsaved_view_name
reloading plugin Default.settings
reloading plugin Default.show_scope_name
reloading plugin Default.side_bar
reloading plugin Default.sort
reloading plugin Default.swap_line
reloading plugin Default.switch_file
reloading plugin Default.symbol
reloading plugin Default.transform
reloading plugin Default.transpose
reloading plugin Default.trim_trailing_white_space
reloading plugin Default.ui
reloading plugin CSS.css_completions
reloading plugin Diff.diff
reloading plugin HTML.encode_html_entities
reloading plugin HTML.html_completions
reloading plugin 0_package_control_loader.00-package_control
reloading plugin 0_package_control_loader.01-pygments
reloading plugin 0_package_control_loader.50-markupsafe
reloading plugin 0_package_control_loader.50-python-markdown
reloading plugin 0_package_control_loader.50-pyyaml
reloading plugin 0_package_control_loader.51-python-jinja2
reloading plugin 0_package_control_loader.55-mdpopups
reloading plugin A File Icon.A File Icon
reloading plugin Boxy Theme.BT
reloading plugin Boxy Theme.Icons
[... 114 lines cut ...]

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./python3.3/threading.py", line 901, in _bootstrap_inner
  File "./python3.3/threading.py", line 858, in run
  File "/home/joachim/.config/sublime-text-3/Packages/Deviot (Arduino IDE)/platformio/upload.py", line 45, in start_upload
    self.check_port_selected()
  File "/home/joachim/.config/sublime-text-3/Packages/Deviot (Arduino IDE)/libraries/project_check.py", line 230, in check_port_selected
    ports_list = self.get_ports_list()
  File "/home/joachim/.config/sublime-text-3/Packages/Deviot (Arduino IDE)/libraries/preferences_bridge.py", line 127, in get_ports_list
    ports_list = serial_port_list()
  File "/home/joachim/.config/sublime-text-3/Packages/Deviot (Arduino IDE)/libraries/serial.py", line 29, in serial_port_list
    ports = list(list_ports.comports())
  File "/home/joachim/.config/sublime-text-3/Packages/Deviot (Arduino IDE)/libraries/pyserial/tools/list_ports_posix.py", line 123, in comports
    return [(d, describe(d), hwinfo(d)) for d in devices]
  File "/home/joachim/.config/sublime-text-3/Packages/Deviot (Arduino IDE)/libraries/pyserial/tools/list_ports_posix.py", line 123, in <listcomp>
    return [(d, describe(d), hwinfo(d)) for d in devices]
  File "/home/joachim/.config/sublime-text-3/Packages/Deviot (Arduino IDE)/libraries/pyserial/tools/list_ports_posix.py", line 94, in describe
    return usb_lsusb_string(sys_usb)
  File "/home/joachim/.config/sublime-text-3/Packages/Deviot (Arduino IDE)/libraries/pyserial/tools/list_ports_posix.py", line 81, in usb_lsusb_string
    return base
NameError: global name 'base' is not defined
```
I hope this helps.

texjoachim, 2017-10-09:
In case it helps: these are the Python versions on my system:

```
joachim@gogmazios:~|⇒  python --version
Python 2.7.13
joachim@gogmazios:~|⇒  python3 --version
Python 3.6.2
joachim@gogmazios:~|⇒  
```

texjoachim, 2017-10-09:
lsusb -v -s 3:6 returns nothing when my Arduino Uno is plugged in.

lsusb -v -s 3:4 returns:

```
joachim@gogmazios:~|⇒  lsusb -v -s 3:4

Bus 003 Device 004: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1a86 QinHeng Electronics
  idProduct          0x7523 HL-340 USB-Serial adapter
  bcdDevice            2.54
  iManufacturer           0 
  iProduct                2 USB2.0-Serial
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           39
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower               96mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      1 
      bInterfaceProtocol      2 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
```

However dmesg says:

```
[ 1415.778427] usb 3-6: new full-speed USB device number 4 using xhci_hcd
[ 1415.919297] usb 3-6: New USB device found, idVendor=1a86, idProduct=7523
[ 1415.919299] usb 3-6: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[ 1415.919300] usb 3-6: Product: USB2.0-Serial
[ 1416.448448] usbcore: registered new interface driver usbserial
[ 1416.448463] usbcore: registered new interface driver usbserial_generic
[ 1416.448475] usbserial: USB Serial support registered for generic
[ 1416.449122] usbcore: registered new interface driver ch341
[ 1416.449135] usbserial: USB Serial support registered for ch341-uart
[ 1416.449149] ch341 3-6:1.0: ch341-uart converter detected
[ 1416.449489] usb 3-6: ch341-uart converter now attached to ttyUSB0
```

What is going on?

texjoachim, 2017-10-10:
I tried it and there was no change. Here is the output on the console:

```
DPI scale: 1
startup, version: 3143 linux x64 channel: stable
executable: /opt/sublime_text/sublime_text
working dir: /
packages path: /home/joachim/.config/sublime-text-3/Packages
state path: /home/joachim/.config/sublime-text-3/Local
zip path: /opt/sublime_text/Packages
zip path: /home/joachim/.config/sublime-text-3/Installed Packages
ignored_packages: ["Vintage"]
pre session restore time: 0.150703
startup time: 0.739573
first paint time: 0.741764
reloading plugin Default.auto_indent_tag
reloading plugin Default.block
reloading plugin Default.comment
reloading plugin Default.convert_syntax
reloading plugin Default.copy_path
reloading plugin Default.delete_word
reloading plugin Default.detect_indentation
reloading plugin Default.duplicate_line
reloading plugin Default.echo
reloading plugin Default.exec
reloading plugin Default.fold
reloading plugin Default.font
reloading plugin Default.goto_line
reloading plugin Default.history_list
reloading plugin Default.indentation
reloading plugin Default.install_package_control
reloading plugin Default.kill_ring
reloading plugin Default.mark
reloading plugin Default.new_templates
reloading plugin Default.open_context_url
reloading plugin Default.open_in_browser
reloading plugin Default.pane
reloading plugin Default.paragraph
reloading plugin Default.paste_from_history
reloading plugin Default.profile
reloading plugin Default.quick_panel
reloading plugin Default.run_syntax_tests
reloading plugin Default.save_on_focus_lost
reloading plugin Default.scroll
reloading plugin Default.set_unsaved_view_name
reloading plugin Default.settings
reloading plugin Default.show_scope_name
reloading plugin Default.side_bar
reloading plugin Default.sort
reloading plugin Default.swap_line
reloading plugin Default.switch_file
reloading plugin Default.symbol
reloading plugin Default.transform
reloading plugin Default.transpose
reloading plugin Default.trim_trailing_white_space
reloading plugin Default.ui
reloading plugin CSS.css_completions
reloading plugin Diff.diff
reloading plugin HTML.encode_html_entities
reloading plugin HTML.html_completions
reloading plugin 0_package_control_loader.00-package_control
reloading plugin 0_package_control_loader.01-pygments
reloading plugin 0_package_control_loader.50-markupsafe
reloading plugin 0_package_control_loader.50-python-markdown
reloading plugin 0_package_control_loader.50-pyyaml
reloading plugin 0_package_control_loader.51-python-jinja2
reloading plugin 0_package_control_loader.55-mdpopups
reloading plugin A File Icon.A File Icon
reloading plugin Boxy Theme.BT
reloading plugin Boxy Theme.Icons
[... 127 lines cut ...]
OSError: lsusb failed

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./python3.3/threading.py", line 901, in _bootstrap_inner
  File "./python3.3/threading.py", line 858, in run
  File "/home/joachim/.config/sublime-text-3/Packages/Deviot (Arduino IDE)/platformio/upload.py", line 45, in start_upload
    self.check_port_selected()
  File "/home/joachim/.config/sublime-text-3/Packages/Deviot (Arduino IDE)/libraries/project_check.py", line 230, in check_port_selected
    ports_list = self.get_ports_list()
  File "/home/joachim/.config/sublime-text-3/Packages/Deviot (Arduino IDE)/libraries/preferences_bridge.py", line 127, in get_ports_list
    ports_list = serial_port_list()
  File "/home/joachim/.config/sublime-text-3/Packages/Deviot (Arduino IDE)/libraries/serial.py", line 29, in serial_port_list
    ports = list(list_ports.comports())
  File "/home/joachim/.config/sublime-text-3/Packages/Deviot (Arduino IDE)/libraries/pyserial/tools/list_ports_posix.py", line 123, in comports
    return [(d, describe(d), hwinfo(d)) for d in devices]
  File "/home/joachim/.config/sublime-text-3/Packages/Deviot (Arduino IDE)/libraries/pyserial/tools/list_ports_posix.py", line 123, in <listcomp>
    return [(d, describe(d), hwinfo(d)) for d in devices]
  File "/home/joachim/.config/sublime-text-3/Packages/Deviot (Arduino IDE)/libraries/pyserial/tools/list_ports_posix.py", line 94, in describe
    return usb_lsusb_string(sys_usb)
  File "/home/joachim/.config/sublime-text-3/Packages/Deviot (Arduino IDE)/libraries/pyserial/tools/list_ports_posix.py", line 81, in usb_lsusb_string
    return base
NameError: global name 'base' is not defined
```

gepd, 2017-10-10:
Seems like the files weren't overwritten, the new branch hasn't `return base` in the line 81

You can see in the new branch the line 81 is different: https://github.com/gepd/Deviot/blob/hotfix/linux_serial/libraries/pyserial/tools/list_ports_posix.py#L81

texjoachim, 2017-10-11:
Completely removed Deviot und installed the archive. Now it works:

```
[ Deviot 2.1.6 ] Starting...
[Wed Oct 11 18:13:56 2017] Processing uno (platform: atmelavr; board: uno; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Converting TestDeviot.ino
Collected 24 compatible libraries
Looking for dependencies...
No dependencies
Compiling .pioenvs/uno/src/TestDeviot.ino.o
Archiving .pioenvs/uno/libFrameworkArduinoVariant.a
Indexing .pioenvs/uno/libFrameworkArduinoVariant.a
Compiling .pioenvs/uno/FrameworkArduino/CDC.o
Compiling .pioenvs/uno/FrameworkArduino/HardwareSerial.o
Compiling .pioenvs/uno/FrameworkArduino/HardwareSerial0.o
Compiling .pioenvs/uno/FrameworkArduino/HardwareSerial1.o
Compiling .pioenvs/uno/FrameworkArduino/HardwareSerial2.o
Compiling .pioenvs/uno/FrameworkArduino/HardwareSerial3.o
Compiling .pioenvs/uno/FrameworkArduino/IPAddress.o
Compiling .pioenvs/uno/FrameworkArduino/PluggableUSB.o
Compiling .pioenvs/uno/FrameworkArduino/Print.o
Compiling .pioenvs/uno/FrameworkArduino/Stream.o
Compiling .pioenvs/uno/FrameworkArduino/Tone.o
Compiling .pioenvs/uno/FrameworkArduino/USBCore.o
Compiling .pioenvs/uno/FrameworkArduino/WInterrupts.o
Compiling .pioenvs/uno/FrameworkArduino/WMath.o
Compiling .pioenvs/uno/FrameworkArduino/WString.o
Compiling .pioenvs/uno/FrameworkArduino/_wiring_pulse.o
Compiling .pioenvs/uno/FrameworkArduino/abi.o
Compiling .pioenvs/uno/FrameworkArduino/hooks.o
Compiling .pioenvs/uno/FrameworkArduino/main.o
Compiling .pioenvs/uno/FrameworkArduino/new.o
Compiling .pioenvs/uno/FrameworkArduino/wiring.o
Compiling .pioenvs/uno/FrameworkArduino/wiring_analog.o
Compiling .pioenvs/uno/FrameworkArduino/wiring_digital.o
Compiling .pioenvs/uno/FrameworkArduino/wiring_pulse.o
Compiling .pioenvs/uno/FrameworkArduino/wiring_shift.o
Archiving .pioenvs/uno/libFrameworkArduino.a
Indexing .pioenvs/uno/libFrameworkArduino.a
Linking .pioenvs/uno/firmware.elf
Checking program size
text	   data	    bss	    dec	    hex	filename
926	      0	      9	    935	    3a7	.pioenvs/uno/firmware.elf
Building .pioenvs/uno/firmware.hex
Looking for upload port...
Use manually specified: /dev/ttyUSB0
Uploading .pioenvs/uno/firmware.hex
avrdude: AVR device initialized and ready to accept instructions
Reading | ################################################## | 100% 0.00s
avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: reading input file ".pioenvs/uno/firmware.hex"
avrdude: writing flash (926 bytes):
Writing | ################################################## | 100% 0.16s
avrdude: 926 bytes of flash written
avrdude: verifying flash memory against .pioenvs/uno/firmware.hex:
avrdude: load data flash data from input file .pioenvs/uno/firmware.hex:
avrdude: input file .pioenvs/uno/firmware.hex contains 926 bytes
avrdude: reading on-chip flash data:
Reading | ################################################## | 100% 0.13s
avrdude: verifying ...
avrdude: 926 bytes of flash verified
avrdude done.  Thank you.
========================= [SUCCESS] Took 2.40 seconds =========================
```

Thank you!
(How do I get back to the "real" version now? Will it update automagically?)

gepd, 2017-10-28:
I'm sorry for the delay @texjoachim I'm preparing a new release with all this changes, I'll let you know the steps to back to the stable release again

texjoachim, 2017-10-28:
Don't worry. Thanks for all the good work!

gepd, 2018-02-17:
Hi @texjoachim I've released the `v2.2.0` you should receive the new version in a few hours, to force update run the option `Package Control: Upgrade/Override All Packages` (`Prefences > Package Control`)

Let me know if you have any problem

closed 2018-02-21 by gepd