gepd/Deviot #190: dev serial baud rate problem

tablatronix, 2017-10-11:
osx

```
Traceback (most recent call last):
  File "/Applications/Sublime Text.app/Contents/MacOS/sublime_plugin.py", line 797, in run_
    return self.run()
  File "/Users/user/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/commands/deviot_choose_baudrate.py", line 14, in run
    items = Quick.serial_baudrate_list()
  File "/Users/user/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/quick_menu.py", line 408, in serial_baudrate_list
    self.index = items.index(current)
ValueError: 115200 is not in list
```

gepd, 2017-10-11:
Can you tell me the steps to reproduce this issue?

tablatronix, 2017-10-12:
I am just trying to click serial baud rate

Booli, 2017-10-12:
Same problem here: 

```Traceback (most recent call last):
  File "/Applications/Sublime Text.app/Contents/MacOS/sublime_plugin.py", line 797, in run_
    return self.run()
  File "/Users/pim/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/commands/deviot_choose_baudrate.py", line 14, in run
    items = Quick.serial_baudrate_list()
  File "/Users/pim/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/quick_menu.py", line 408, in serial_baudrate_list
    self.index = items.index(current)
ValueError: 19200 is not in list
```
I noticed that there is no arrow also any more, so I can't select different Baudrates.

gepd, 2017-10-13:
I've tried to replicate this issue but I couldn't. For me it's working in OSX, Linux and Windows

Try this:
`Deviot Menu > Select Language` and select the same language you already are using, it will make update the menu file.

@Booli There not arrow now because baudrate options now are displying in a quick panel

gepd, 2017-10-14:
I think I've found the problem, I've fixed and updated to `dev.10`

gepd, 2017-10-20:
reopen this issue if you keep getting this error

closed 2017-10-20 by gepd