gepd/Deviot #191: Deviot doesn't upload - Error 1

Scartoug, 2017-10-16:
Deviot doesn't upload my sketch  causing error 1

`[ Deviot 2.2.0-dev.10 ] Starting...
[10/15/17 23:07:32] Processing uno (upload_protocol: stk500v2; platform: atmelavr; upload_flags: -Pusb; board: uno; framework: arduino)
Verbose mode can be enabled via `-v, --verbose` option
Converting Bli2nk.ino
Collected 24 compatible libraries
Looking for dependencies...
Project does not have dependencies
Compiling .pioenvs\uno\src\Bli2nk.ino.o
Archiving .pioenvs\uno\libFrameworkArduinoVariant.a
Compiling .pioenvs\uno\FrameworkArduino\CDC.o
Compiling .pioenvs\uno\FrameworkArduino\HardwareSerial.o
Compiling .pioenvs\uno\FrameworkArduino\HardwareSerial0.o
Compiling .pioenvs\uno\FrameworkArduino\HardwareSerial1.o
Compiling .pioenvs\uno\FrameworkArduino\HardwareSerial2.o
Compiling .pioenvs\uno\FrameworkArduino\HardwareSerial3.o
Compiling .pioenvs\uno\FrameworkArduino\IPAddress.o
Compiling .pioenvs\uno\FrameworkArduino\PluggableUSB.o
Compiling .pioenvs\uno\FrameworkArduino\Print.o
Compiling .pioenvs\uno\FrameworkArduino\Stream.o
Compiling .pioenvs\uno\FrameworkArduino\Tone.o
Compiling .pioenvs\uno\FrameworkArduino\USBCore.o
Compiling .pioenvs\uno\FrameworkArduino\WInterrupts.o
Compiling .pioenvs\uno\FrameworkArduino\WMath.o
Compiling .pioenvs\uno\FrameworkArduino\WString.o
Compiling .pioenvs\uno\FrameworkArduino\_wiring_pulse.o
Compiling .pioenvs\uno\FrameworkArduino\abi.o
Compiling .pioenvs\uno\FrameworkArduino\hooks.o
Compiling .pioenvs\uno\FrameworkArduino\main.o
Compiling .pioenvs\uno\FrameworkArduino\new.o
Compiling .pioenvs\uno\FrameworkArduino\wiring.o
Compiling .pioenvs\uno\FrameworkArduino\wiring_analog.o
Compiling .pioenvs\uno\FrameworkArduino\wiring_digital.o
Compiling .pioenvs\uno\FrameworkArduino\wiring_pulse.o
Compiling .pioenvs\uno\FrameworkArduino\wiring_shift.o
Archiving .pioenvs\uno\libFrameworkArduino.a
Linking .pioenvs\uno\firmware.elf
Checking program size
text	   data	    bss	    dec	    hex	filename
2224	     22	    168	   2414	    96e	.pioenvs\uno\firmware.elf
Building .pioenvs\uno\firmware.hex
Looking for upload port...
Programming .pioenvs\uno\firmware.hex
avrdude done.  Thank you.
*** [program] Error 1
 [ERROR] Took 3.74 seconds 
`

Scartoug, 2017-10-16:
> [ Deviot 2.2.0-dev.10 ] Starting...
[10/15/17 23:07:32] Processing uno (upload_protocol: stk500v2; platform: atmelavr; upload_flags: -Pusb; board: uno; framework: arduino)
Verbose mode can be enabled via `-v, --verbose` option
Converting Bli2nk.ino
Collected 24 compatible libraries
Looking for dependencies...
Project does not have dependencies
Compiling .pioenvs\uno\src\Bli2nk.ino.o
Archiving .pioenvs\uno\libFrameworkArduinoVariant.a
Compiling .pioenvs\uno\FrameworkArduino\CDC.o
Compiling .pioenvs\uno\FrameworkArduino\HardwareSerial.o
Compiling .pioenvs\uno\FrameworkArduino\HardwareSerial0.o
Compiling .pioenvs\uno\FrameworkArduino\HardwareSerial1.o
Compiling .pioenvs\uno\FrameworkArduino\HardwareSerial2.o
Compiling .pioenvs\uno\FrameworkArduino\HardwareSerial3.o
Compiling .pioenvs\uno\FrameworkArduino\IPAddress.o
Compiling .pioenvs\uno\FrameworkArduino\PluggableUSB.o
Compiling .pioenvs\uno\FrameworkArduino\Print.o
Compiling .pioenvs\uno\FrameworkArduino\Stream.o
Compiling .pioenvs\uno\FrameworkArduino\Tone.o
Compiling .pioenvs\uno\FrameworkArduino\USBCore.o
Compiling .pioenvs\uno\FrameworkArduino\WInterrupts.o
Compiling .pioenvs\uno\FrameworkArduino\WMath.o
Compiling .pioenvs\uno\FrameworkArduino\WString.o
Compiling .pioenvs\uno\FrameworkArduino\_wiring_pulse.o
Compiling .pioenvs\uno\FrameworkArduino\abi.o
Compiling .pioenvs\uno\FrameworkArduino\hooks.o
Compiling .pioenvs\uno\FrameworkArduino\main.o
Compiling .pioenvs\uno\FrameworkArduino\new.o
Compiling .pioenvs\uno\FrameworkArduino\wiring.o
Compiling .pioenvs\uno\FrameworkArduino\wiring_analog.o
Compiling .pioenvs\uno\FrameworkArduino\wiring_digital.o
Compiling .pioenvs\uno\FrameworkArduino\wiring_pulse.o
Compiling .pioenvs\uno\FrameworkArduino\wiring_shift.o
Archiving .pioenvs\uno\libFrameworkArduino.a
Linking .pioenvs\uno\firmware.elf
Checking program size
text	   data	    bss	    dec	    hex	filename
2224	     22	    168	   2414	    96e	.pioenvs\uno\firmware.elf
Building .pioenvs\uno\firmware.hex
Looking for upload port...
Programming .pioenvs\uno\firmware.hex
avrdude done.  Thank you.
*** [program] Error 1
 [ERROR] Took 3.74 seconds 

>

gepd, 2017-10-16:
Are you using an arduino uno?
Have you tried to upload the same sketch with the Arduino IDE?

Scartoug, 2017-10-19:
I'm using an Arduino UNO and my sketch can be uploaded through Arduino IDE, working

gepd, 2017-10-19:
Please follow [this](https://github.com/gepd/Deviot/wiki/Setup#developer) steps to install the develop version in the status bar you should see `Deviot 2.2.0-dev.10`

Then activate the verbose output from `Deviot Menu > Options > Full Verbose Output When Building`

After this, try to upload your sketch again. This time the output should have more information, paste that info here.

Ben-Lichtman, 2017-11-21:
I am getting a similar issue with error 1

Here is the verbose output of upload:

[ Deviot 2.2.0-dev.11 ] Starting...
[Tue Nov 21 21:38:48 2017] Processing megaatmega2560 (upload_protocol: stk500v1; framework: arduino; upload_flags: -P$UPLOAD_PORT; platform: atmelavr; board: megaatmega2560; upload_port: /dev/tty.usbmodem1411)
--------------------------------------------------------------------------------
avr-g++ -o "/Users/ben/Desktop/Sensor/project.ino.cpp" -x c++ -fpreprocessed -dD -E "/var/folders/h_/ck8vd1ln5jj2r5y8wyfs7vc40000gn/T/tmp0uA9rv"
Collected 41 compatible libraries
Looking for dependencies...
Project does not have dependencies
avr-g++ -o .pioenvs/megaatmega2560/src/project.ino.o -c -fno-exceptions -fno-threadsafe-statics -fpermissive -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega2560 -DF_CPU=16000000L -DPLATFORMIO=30401 -DARDUINO_ARCH_AVR -DARDUINO_AVR_MEGA2560 -DARDUINO=10619 -I/Users/ben/.platformio/packages/framework-arduinoavr/cores/arduino -I/Users/ben/.platformio/packages/framework-arduinoavr/variants/mega -I/Users/ben/Desktop/Sensor /Users/ben/Desktop/Sensor/project.ino.cpp
avr-gcc-ar rc .pioenvs/megaatmega2560/libFrameworkArduinoVariant.a
avr-gcc-ranlib .pioenvs/megaatmega2560/libFrameworkArduinoVariant.a
avr-g++ -o .pioenvs/megaatmega2560/FrameworkArduino/CDC.o -c -fno-exceptions -fno-threadsafe-statics -fpermissive -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega2560 -DF_CPU=16000000L -DPLATFORMIO=30401 -DARDUINO_ARCH_AVR -DARDUINO_AVR_MEGA2560 -DARDUINO=10619 -I/Users/ben/.platformio/packages/framework-arduinoavr/cores/arduino -I/Users/ben/.platformio/packages/framework-arduinoavr/variants/mega /Users/ben/.platformio/packages/framework-arduinoavr/cores/arduino/CDC.cpp
avr-g++ -o .pioenvs/megaatmega2560/FrameworkArduino/HardwareSerial.o -c -fno-exceptions -fno-threadsafe-statics -fpermissive -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega2560 -DF_CPU=16000000L -DPLATFORMIO=30401 -DARDUINO_ARCH_AVR -DARDUINO_AVR_MEGA2560 -DARDUINO=10619 -I/Users/ben/.platformio/packages/framework-arduinoavr/cores/arduino -I/Users/ben/.platformio/packages/framework-arduinoavr/variants/mega /Users/ben/.platformio/packages/framework-arduinoavr/cores/arduino/HardwareSerial.cpp
avr-g++ -o .pioenvs/megaatmega2560/FrameworkArduino/HardwareSerial0.o -c -fno-exceptions -fno-threadsafe-statics -fpermissive -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega2560 -DF_CPU=16000000L -DPLATFORMIO=30401 -DARDUINO_ARCH_AVR -DARDUINO_AVR_MEGA2560 -DARDUINO=10619 -I/Users/ben/.platformio/packages/framework-arduinoavr/cores/arduino -I/Users/ben/.platformio/packages/framework-arduinoavr/variants/mega /Users/ben/.platformio/packages/framework-arduinoavr/cores/arduino/HardwareSerial0.cpp
avr-g++ -o .pioenvs/megaatmega2560/FrameworkArduino/HardwareSerial1.o -c -fno-exceptions -fno-threadsafe-statics -fpermissive -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega2560 -DF_CPU=16000000L -DPLATFORMIO=30401 -DARDUINO_ARCH_AVR -DARDUINO_AVR_MEGA2560 -DARDUINO=10619 -I/Users/ben/.platformio/packages/framework-arduinoavr/cores/arduino -I/Users/ben/.platformio/packages/framework-arduinoavr/variants/mega /Users/ben/.platformio/packages/framework-arduinoavr/cores/arduino/HardwareSerial1.cpp
avr-g++ -o .pioenvs/megaatmega2560/FrameworkArduino/HardwareSerial2.o -c -fno-exceptions -fno-threadsafe-statics -fpermissive -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega2560 -DF_CPU=16000000L -DPLATFORMIO=30401 -DARDUINO_ARCH_AVR -DARDUINO_AVR_MEGA2560 -DARDUINO=10619 -I/Users/ben/.platformio/packages/framework-arduinoavr/cores/arduino -I/Users/ben/.platformio/packages/framework-arduinoavr/variants/mega /Users/ben/.platformio/packages/framework-arduinoavr/cores/arduino/HardwareSerial2.cpp
avr-g++ -o .pioenvs/megaatmega2560/FrameworkArduino/HardwareSerial3.o -c -fno-exceptions -fno-threadsafe-statics -fpermissive -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega2560 -DF_CPU=16000000L -DPLATFORMIO=30401 -DARDUINO_ARCH_AVR -DARDUINO_AVR_MEGA2560 -DARDUINO=10619 -I/Users/ben/.platformio/packages/framework-arduinoavr/cores/arduino -I/Users/ben/.platformio/packages/framework-arduinoavr/variants/mega /Users/ben/.platformio/packages/framework-arduinoavr/cores/arduino/HardwareSerial3.cpp
avr-g++ -o .pioenvs/megaatmega2560/FrameworkArduino/IPAddress.o -c -fno-exceptions -fno-threadsafe-statics -fpermissive -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega2560 -DF_CPU=16000000L -DPLATFORMIO=30401 -DARDUINO_ARCH_AVR -DARDUINO_AVR_MEGA2560 -DARDUINO=10619 -I/Users/ben/.platformio/packages/framework-arduinoavr/cores/arduino -I/Users/ben/.platformio/packages/framework-arduinoavr/variants/mega /Users/ben/.platformio/packages/framework-arduinoavr/cores/arduino/IPAddress.cpp
avr-g++ -o .pioenvs/megaatmega2560/FrameworkArduino/PluggableUSB.o -c -fno-exceptions -fno-threadsafe-statics -fpermissive -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega2560 -DF_CPU=16000000L -DPLATFORMIO=30401 -DARDUINO_ARCH_AVR -DARDUINO_AVR_MEGA2560 -DARDUINO=10619 -I/Users/ben/.platformio/packages/framework-arduinoavr/cores/arduino -I/Users/ben/.platformio/packages/framework-arduinoavr/variants/mega /Users/ben/.platformio/packages/framework-arduinoavr/cores/arduino/PluggableUSB.cpp
[... cut ...]
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x00
avrdude done.  Thank you.
*** [program] Error 1
========================== [ERROR] Took 53.28 seconds ==========================

Build works fine, and I have the correct board selected. I have also tried this on the dev branch, same thing.

gepd, 2018-02-12:
re-open if you keep with this issue

closed 2018-02-12 by gepd