gepd/Deviot #192: DEV feature serial header newline and questions

tablatronix, 2017-10-17:
Can you add a newline before the serial header

```
NEWLINE <---
[ Deviot 2.2.0-dev.10 ] Serial Monitor | COM5
```

Also is this header optional somewhere ( I really like it, but might want it off sometimes )

Also what is clean-view ? in the serial menu ?

gepd, 2017-10-17:
I'll try to make some time this week to add this little change. 
I'm currently developing a plugin to work with micropython so until release the first version I'll give priority to that proyect.

Look [this](https://github.com/gepd/Deviot/wiki/Deviot-Menu#clean-monitor-view) page in the wiki, to know about the clean-view options (and the rest of the options in deviot)

tablatronix, 2017-10-18:
Thanks, I didnt think the wiki was up to date great, although auto clean does not seem to work for me

gepd, 2017-10-18:
I've just test it and is working in my machine, do you have anything in the ST console after run the command?

Note that this command only works when the serial monitor is running

tablatronix, 2017-10-19:
Run what command? I thought it was automatic, I had 30,000 lines

gepd, 2017-10-19:
It's automatic if you had the `auto-clean` option activated or you can manually clean it with the `clean-view` command

tablatronix, 2017-10-19:
Yeah that is what i am saying , auto clean doesnt seem to do anything , i have never seen the window clear itself

tablatronix, 2017-10-19:
Ok maybe it is working, I think it might only work on a continuous session. If it stops and starts it probably breaks, maybe exacerbated by the serial stopping bug

gepd, 2017-10-19:
I've pushed an improvement that should make it always work, let me know if you can note the change

tablatronix, 2017-10-19:
Thanks, Ill check out that micropython plugin when it comes out, I am bout to start messing with it and python in general.

gepd, 2017-10-19:
It's really good to work with micropython, esp8266 is fully supported and the esp32 is still in development. For me it's more comfortable because I've more time working with python

gepd, 2017-10-27:
Look [first alpha release](https://github.com/gepd/uPiotMicroPythonTool)

closed 2017-10-28 by tablatronix