gepd/Deviot #194: Problema sobre el puerto de mi arduino.

anzeer, 2017-10-26:
No me aparece el puerto COM3 del arduino 

[image]

Eso sucede al momento de darle a cargar, con el compilador original de arduino si me lo reconoce, he buscado bastante y no encuentro solución...

gepd, 2017-10-26:
¿Puedes ir a `View > Show Console` y pegar aqui toda la información que aparece?

En el administrador de dispositivos de windows ¿también aparece como conectado?

anzeer, 2017-10-27:
> DPI scale: 1
startup, version: 3143 windows x64 channel: stable
executable: /C/Program Files/Sublime Text 3/sublime_text.exe
working dir: /C/Program Files/Sublime Text 3
packages path: /C/Users/Anz/AppData/Roaming/Sublime Text 3/Packages
state path: /C/Users/Anz/AppData/Roaming/Sublime Text 3/Local
zip path: /C/Program Files/Sublime Text 3/Packages
zip path: /C/Users/Anz/AppData/Roaming/Sublime Text 3/Installed Packages
ignored_packages: ["Vintage"]
pre session restore time: 2.02466
startup time: 2.06966
first paint time: 2.06966
reloading plugin Default.auto_indent_tag
reloading plugin Default.block
reloading plugin Default.comment
reloading plugin Default.convert_syntax
reloading plugin Default.copy_path
reloading plugin Default.delete_word
reloading plugin Default.detect_indentation
reloading plugin Default.duplicate_line
reloading plugin Default.echo
reloading plugin Default.exec
reloading plugin Default.fold
reloading plugin Default.font
reloading plugin Default.goto_line
reloading plugin Default.history_list
reloading plugin Default.indentation
reloading plugin Default.install_package_control
reloading plugin Default.kill_ring
reloading plugin Default.mark
reloading plugin Default.new_templates
reloading plugin Default.open_context_url
reloading plugin Default.open_in_browser
reloading plugin Default.pane
reloading plugin Default.paragraph
reloading plugin Default.paste_from_history
reloading plugin Default.profile
reloading plugin Default.quick_panel
reloading plugin Default.run_syntax_tests
reloading plugin Default.save_on_focus_lost
reloading plugin Default.scroll
reloading plugin Default.set_unsaved_view_name
reloading plugin Default.settings
reloading plugin Default.show_scope_name
reloading plugin Default.side_bar
reloading plugin Default.sort
reloading plugin Default.swap_line
reloading plugin Default.switch_file
reloading plugin Default.symbol
reloading plugin Default.transform
reloading plugin Default.transpose
reloading plugin Default.trim_trailing_white_space
reloading plugin Default.ui
reloading plugin CSS.css_completions
reloading plugin Diff.diff
reloading plugin HTML.encode_html_entities
reloading plugin HTML.html_completions
reloading plugin Package Control.1_reloader
reloading plugin Package Control.2_bootstrap
reloading plugin Package Control.Package Control
reloading plugin Deviot-develop.Deviot
plugins loaded
Traceback (most recent call last):
  File "C:\Program Files\Sublime Text 3\sublime_plugin.py", line 210, in on_api_ready
    m.plugin_loaded()
  File "C:\Users\Anz\AppData\Roaming\Sublime Text 3\Packages\Deviot-develop\Deviot.py", line 47, in plugin_loaded
    from package_control import events
ImportError: No module named 'package_control'
Package Control: Not running package cleanup since bootstrapping is not yet complete
Traceback (most recent call last):
  File "C:\Program Files\Sublime Text 3\sublime_plugin.py", line 210, in on_api_ready
    m.plugin_loaded()
  File "2_bootstrap in C:\Program Files\Sublime Text 3\Packages\Package Control.sublime-package", line 103, in plugin_loaded
  File "package_control.loader in C:\Program Files\Sublime Text 3\Packages\Package Control.sublime-package", line 193, in add
  File "./python3.3/zipfile.py", line 921, in __init__
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Program Files\\Sublime Text 3\\Installed Packages\\0_package_control_loader.sublime-package'
reloading settings Packages/User/deviot.sublime-settings

En el administrador de dispositivos de Windows si me aparece conectado, Gracias por responder.

gepd, 2017-10-27:
Necesitas instalar el control de paquetes, la manera más fácil es ir a `Tools > Install Package Control` la otra es seguir [estas](https://packagecontrol.io/installation) instrucciones

anzeer, 2017-10-27:
Lo instale y sigue sin aparecer:
[image]
Tampoco me salen errores en la consola. Realmente es un muy buen trabajo el que desarrollo, gracias por seguir mi caso.

gepd, 2017-10-27:
Gracias por el apoyo @anzeer !
¿Has reiniciado ST después de instalar el Control de Paquetes? es necesario hacerlo

Puedes copiar nuevamente la salida de la consola para ver si hay alguna otra cosa que pueda estar afectando al plugin

gepd, 2018-02-09:
Vuelve abrir este tópico si continuas con el problema

closed 2018-02-09 by gepd