gepd/Deviot #195: Arduino Uno Serial COM port not listed

jostidetosti, 2017-10-29:
### Operating system
Windows 8.1 x64

### Board used
Arduino Uno

### Description of problem
When trying to upload to Arduino Uno, can't select the Arduino as port. Uploading from Arduino IDE works fine. I can also upload to a Wemos d1 board from Deviot.

### Error(s) in ST Console? (View > Show Console)
Only this, when starting sublime:
> Traceback: error.
> 
> Output:
> 
> 
> Error:
> Invalid Parameter - TrueColorMatte
> 
> 
> Highlighting colors with gutter icons is not supported with current ImageMagick setup. Try configuring the "icon_factory setting"
> reloading settings Packages/User/deviot.sublime-settings
> Package Control: No updated packages
> Unable to open clipboard
> Quick panel unavailable

May be unrelated.

### Steps to Reproduce
Connect Arduino to pc, start sublime text, click Deviot -> upload.

### Actual Results
The uno is not in the list of ports.
[https://i.imgur.com/XL85nL7.png](https://i.imgur.com/XL85nL7.png)

### Expected Results
I can select the uno in the list of ports.

### Additional info
Latest sublime text 3 and Deviot versions. Uploading with Deviot from Kubuntu 17.04 works fine (I have dual boot).
Seems similar to #141

gepd, 2017-10-29:
can you check the ST console and paste the output here after call the list of ports?

jostidetosti, 2017-10-29:
> DPI scale: 1
> startup, version: 3143 windows x64 channel: stable
> executable: /C/Program Files/Sublime Text 3/sublime_text.exe
> working dir: /C/Program Files/Sublime Text 3
> packages path: /C/Users/Borre/AppData/Roaming/Sublime Text 3/Packages
> state path: /C/Users/Borre/AppData/Roaming/Sublime Text 3/Local
> zip path: /C/Program Files/Sublime Text 3/Packages
> zip path: /C/Users/Borre/AppData/Roaming/Sublime Text 3/Installed Packages
> ignored_packages: ["Vintage"]
> pre session restore time: 0.229578
> startup time: 0.270578
> first paint time: 0.271578
> reloading plugin Default.auto_indent_tag
> reloading plugin Default.block
> reloading plugin Default.comment
> reloading plugin Default.convert_syntax
> reloading plugin Default.copy_path
> reloading plugin Default.delete_word
> reloading plugin Default.detect_indentation
> reloading plugin Default.duplicate_line
> reloading plugin Default.echo
> reloading plugin Default.exec
> reloading plugin Default.fold
> reloading plugin Default.font
> reloading plugin Default.goto_line
> reloading plugin Default.history_list
> reloading plugin Default.indentation
> reloading plugin Default.install_package_control
> reloading plugin Default.kill_ring
> reloading plugin Default.mark
> reloading plugin Default.new_templates
> reloading plugin Default.open_context_url
> reloading plugin Default.open_in_browser
> reloading plugin Default.pane
> reloading plugin Default.paragraph
> reloading plugin Default.paste_from_history
> reloading plugin Default.profile
> reloading plugin Default.quick_panel
> reloading plugin Default.run_syntax_tests
> reloading plugin Default.save_on_focus_lost
> reloading plugin Default.scroll
> reloading plugin Default.set_unsaved_view_name
> reloading plugin Default.settings
> reloading plugin Default.show_scope_name
> reloading plugin Default.side_bar
> reloading plugin Default.sort
> reloading plugin Default.swap_line
> reloading plugin Default.switch_file
> reloading plugin Default.symbol
> reloading plugin Default.transform
> reloading plugin Default.transpose
> reloading plugin Default.trim_trailing_white_space
> reloading plugin Default.ui
> reloading plugin CSS.css_completions
> reloading plugin Diff.diff
> reloading plugin HTML.encode_html_entities
> reloading plugin HTML.html_completions
> reloading plugin 0_package_control_loader.00-package_control
> reloading plugin All Autocomplete.all_views_completions
> reloading plugin Color Highlighter.color_converter
> reloading plugin Color Highlighter.color_highlighter
> reloading plugin Color Highlighter.color_hover_listener
> reloading plugin Color Highlighter.color_scheme
> reloading plugin Color Highlighter.color_scheme_color_highlighter
> reloading plugin Color Highlighter.color_searcher
> reloading plugin Color Highlighter.color_selection_listener
> reloading plugin Color Highlighter.colors
> reloading plugin Color Highlighter.content_listener
> reloading plugin Color Highlighter.convert_color_command
> reloading plugin Color Highlighter.css_colors
> reloading plugin Color Highlighter.dummy_event_listener
> reloading plugin Color Highlighter.gutter_icons_color_highlighter
> reloading plugin Color Highlighter.load_resource
> reloading plugin Color Highlighter.main
> reloading plugin Color Highlighter.path
> reloading plugin Color Highlighter.phantoms_color_highlighter
> reloading plugin Color Highlighter.pick_color_command
> reloading plugin Color Highlighter.regex_compiler
> reloading plugin Color Highlighter.regions
> reloading plugin Color Highlighter.set_setting_command
> reloading plugin Color Highlighter.settings
> reloading plugin Color Highlighter.st_helper
> reloading plugin Color Highlighter.topsort
> reloading plugin Package Control.1_reloader
> reloading plugin Package Control.2_bootstrap
> reloading plugin Package Control.Package Control
> reloading plugin Processing.Processing
> reloading plugin sublimelint.commands
> reloading plugin sublimelint.sublimelint
> reloading plugin Text Pastry.text_pastry
> reloading plugin Text Pastry.text_pastry_addons
> reloading plugin Text Pastry.text_pastry_clipboard
> reloading plugin Text Pastry.text_pastry_selection
> reloading plugin Deviot (Arduino IDE).Deviot
> reloading plugin SublimeLinter.commands
> reloading plugin SublimeLinter.sublimelinter
> reloading plugin SublimeREPL.__init__
> reloading plugin SublimeREPL.completions
> reloading plugin SublimeREPL.lang_integration
> reloading plugin SublimeREPL.run_existing_command
> reloading plugin SublimeREPL.sublimerepl
> reloading plugin SublimeREPL.sublimerepl_build_system_hack
> reloading plugin SublimeREPL.text_transfer
> reloading plugin TSQLEasy.sqlodbccon
> TSQLEasy: PyODBC was loaded successfully.
> reloading plugin TSQLEasy.tsqleasy
> plugins loaded
> SublimeLinter: debug mode: off 
> Traceback: error.
> 
> Output:
> 
> 
> Error:
> Invalid Parameter - TrueColorMatte
> 
> 
> Highlighting colors with gutter icons is not supported with current ImageMagick setup. Try configuring the "icon_factory setting"
> reloading settings Packages/User/deviot.sublime-settings
> Package Control: Installing 1 upgrades
> ignored packages updated to: ["SublimeLinter", "Vintage"]
> unloading plugin SublimeLinter.commands
> unloading plugin SublimeLinter.sublimelinter
> reloading settings Packages/SublimeLinter/SublimeLinter.sublime-settings
> reloading settings Packages/User/Package Control.sublime-settings
> reloading settings Packages/User/Preferences.sublime-settings
> Package Control: Upgraded SublimeLinter to 3.10.3
> ignored packages updated to: ["Vintage"]
> reloading plugin SublimeLinter.commands
> reloading plugin SublimeLinter.sublimelinter
> reloading settings Packages/SublimeLinter/SublimeLinter.sublime-settings
> SublimeLinter: daemon detected a reload 
> reloading settings Packages/User/Package Control.sublime-settings
> reloading settings Packages/User/Preferences.sublime-settings
> reloading settings Packages/User/Preferences.sublime-settings

gepd, 2017-10-29:
Try this; remove the Deviot folder from the Packages (`ST Menu > Preferences > Browse Packages`) and replace it with [this](https://github.com/gepd/Deviot/archive/hotfix/linux_serial.zip)

and let me know if that works

jostidetosti, 2017-11-01:
It works, I can now upload to the Arduino.

gepd, 2017-11-01:
Can you replace the files again, I've pushed some changes that should solve the problem

gepd, 2017-11-01:
Please keep reporting in: https://github.com/gepd/Deviot/issues/196

closed 2017-11-01 by gepd