gepd/Deviot #196: Serial Monitor doesn't show up

werofuentes, 2017-10-30:
Windows 7 Home x64

Arduino Mega 2560

I just install Deviot on Sublime Text, it works almost perfect, i only have one problem, i can`t open the serial monitor.

gepd, 2017-10-31:
please, paste here the ouput of the ST console (`View > Show console`) after start the Serial Monitor

werofuentes, 2017-10-31:
DPI scale: 1
startup, version: 3143 windows x64 channel: stable
executable: /C/Program Files/Sublime Text 3/sublime_text.exe
working dir: /C/Program Files/Sublime Text 3
packages path: /C/Users/Wero/AppData/Roaming/Sublime Text 3/Packages
state path: /C/Users/Wero/AppData/Roaming/Sublime Text 3/Local
zip path: /C/Program Files/Sublime Text 3/Packages
zip path: /C/Users/Wero/AppData/Roaming/Sublime Text 3/Installed Packages
ignored_packages: ["Vintage"]
pre session restore time: 0.893785
font face "Monaco" has different widths for bold characters, disabling to prevent text reflow
startup time: 0.931785
first paint time: 0.932785
reloading plugin Default.auto_indent_tag
reloading plugin Default.block
reloading plugin Default.comment
reloading plugin Default.convert_syntax
reloading plugin Default.copy_path
reloading plugin Default.delete_word
reloading plugin Default.detect_indentation
reloading plugin Default.duplicate_line
reloading plugin Default.echo
reloading plugin Default.exec
reloading plugin Default.fold
reloading plugin Default.font
reloading plugin Default.goto_line
reloading plugin Default.history_list
reloading plugin Default.indentation
reloading plugin Default.install_package_control
reloading plugin Default.kill_ring
reloading plugin Default.mark
reloading plugin Default.new_templates
reloading plugin Default.open_context_url
reloading plugin Default.open_in_browser
reloading plugin Default.pane
reloading plugin Default.paragraph
reloading plugin Default.paste_from_history
reloading plugin Default.profile
reloading plugin Default.quick_panel
reloading plugin Default.run_syntax_tests
reloading plugin Default.save_on_focus_lost
reloading plugin Default.scroll
reloading plugin Default.set_unsaved_view_name
reloading plugin Default.settings
reloading plugin Default.show_scope_name
reloading plugin Default.side_bar
reloading plugin Default.sort
reloading plugin Default.swap_line
reloading plugin Default.switch_file
reloading plugin Default.symbol
reloading plugin Default.transform
reloading plugin Default.transpose
reloading plugin Default.trim_trailing_white_space
reloading plugin Default.ui
reloading plugin CSS.css_completions
reloading plugin Diff.diff
reloading plugin HTML.encode_html_entities
reloading plugin HTML.html_completions
reloading plugin 0_package_control_loader.00-package_control
reloading plugin 0_package_control_loader.01-pygments
reloading plugin 0_package_control_loader.50-backrefs
reloading plugin 0_package_control_loader.50-markupsafe
reloading plugin 0_package_control_loader.50-pymdownx
reloading plugin 0_package_control_loader.50-python-markdown
reloading plugin 0_package_control_loader.50-pyyaml
reloading plugin 0_package_control_loader.51-python-jinja2
reloading plugin 0_package_control_loader.55-mdpopups
reloading plugin A File Icon.A File Icon
reloading plugin AdvancedNewFile.AdvancedNewFile
reloading plugin All Autocomplete.all_views_completions
[... 123 lines cut ...]
PermissionError: [WinError 32] El proceso no tiene acceso al archivo porque está siendo utilizado por otro proceso: 'C:\\Users\\Wero\\AppData\\Roaming\\Sublime Text 3\\Packages\\JavaScript Completions\\helper\\can_i_use\\can_i_use_data2.json'

Package Control: Skipping automatic upgrade, last run at 2017-10-31 17:25:45, next run at 2017-10-31 18:25:45 or after
reloading settings Packages/User/deviot.sublime-settings
Traceback (most recent call last):
  File "C:\Program Files\Sublime Text 3\sublime_plugin.py", line 797, in run_
    return self.run()
  File "C:\Users\Wero\AppData\Roaming\Sublime Text 3\Packages\Deviot-develop\commands\deviot_toggle_serial_monitor.py", line 6, in run
    serial.toggle_serial_monitor()
  File "C:\Users\Wero\AppData\Roaming\Sublime Text 3\Packages\Deviot-develop\libraries\serial.py", line 293, in toggle_serial_monitor
    serial_monitor = get_serial_monitor(port_id)
  File "C:\Users\Wero\AppData\Roaming\Sublime Text 3\Packages\Deviot-develop\libraries\serial.py", line 282, in get_serial_monitor
    serial_monitor = SerialMonitor(port_id)
  File "C:\Users\Wero\AppData\Roaming\Sublime Text 3\Packages\Deviot-develop\libraries\serial.py", line 52, in __init__
    self.serial = pyserial.Serial()
  File "C:\Users\Wero\AppData\Roaming\Sublime Text 3\Packages\Deviot-develop\libraries\pyserial\serialwin32.py", line 31, in __init__
    super(Serial, self).__init__(*args, **kwargs)
  File "C:\Users\Wero\AppData\Roaming\Sublime Text 3\Packages\Deviot-develop\libraries\pyserial\serialutil.py", line 229, in __init__
    self.exclusive = exclusive
  File "C:\Users\Wero\AppData\Roaming\Sublime Text 3\Packages\Deviot-develop\libraries\pyserial\serialwin32.py", line 475, in exclusive
    serial.SerialBase.exclusive.__set__(self, exclusive)
NameError: global name 'serial' is not defined

That's all.
Actually, i just noticed something weird, when i started sublime text, i clicked the button to start the serial monitora and it worked, but then, i compiled something just to check out and started again the serial monitor but this time didn't work.

gepd, 2017-11-01:
Please replace the current files of the plugin by [this](https://github.com/gepd/Deviot/archive/hotfix/linux_serial.zip) and let me know if you keep seeing the errors, I hope to merge that changes this week to solve this problem

werofuentes, 2017-11-01:
I tried what you say, but still the same. 
By the way, thank you.

gepd, 2017-11-01:
It's the exactly the same error? if that is the case, maybe the plugin wasn't update, it has a new code so the error should be different. If it's different please paste here to upload a new patch

You're welcome!

werofuentes, 2017-11-01:
Yeah, the same, i forgot to mention something, the link with the zip that you gave me, i had already replace it with the original, i had the problem about the serial port as well, but with that zip you fixed it.

gepd, 2017-11-01:
I've pushed a change today to fix other problems in that branch, replacing the files again should fix the problem, same as: https://github.com/gepd/Deviot/issues/195

werofuentes, 2017-11-02:
Wow, awesome, now it works, thanks a lot @gepd, you made a great work, keep it up.

jostidetosti, 2017-11-02:
My issues ( #195) are resolved. Everything works and no errors in console.

gepd, 2017-11-02:
Thanks both for the report, I'll let this open until I merge it into master branch

gepd, 2018-02-12:
fix merged in the developer branch.

Re-open it if you have the same issue

closed 2018-02-12 by gepd