gepd/Deviot #198: fatal error: twi.h: No such file or directory

jostidetosti, 2017-11-10:
Kubuntu 17.10
Arduino Uno

Code compiles fine in Arduino IDE, but not with Deviot. Error: 

> fatal error: twi.h: No such file or directory

No errors in ST Console.

Code:

```
#include <Wire.h>

void setup(){}

void loop(){}
```

gepd, 2017-11-10:
Can you try putting your sketch in a empty folder and try again.

Let me know if that works

jostidetosti, 2017-11-14:
Same result.

gepd, 2017-11-16:
Activate the option `Deviot > Options > Full Verbose Output` and compile again, it should display extra information in th deviot console, paste that info here. If you see the same info, restart ST and try again

jostidetosti, 2017-11-16:
`
[ Deviot 2.1.6 ] Starting...
[Thu Nov 16 10:04:13 2017] Processing uno (platform: atmelavr; lib_extra_dirs: /home/username/Programs/arduino-1.8.4/libraries; board: uno; framework: arduino)
--------------------------------------------------------------------------------
avr-g++ -o "/home/username/Code/Arduino/twitest/twitest.ino.cpp" -x c++ -fpreprocessed -dD -E "/tmp/tmpqiUdtH"
Collected 87 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <Wire> v1.0 (/home/username/.platformio/lib/Wire)
avr-g++ -o .pioenvs/uno/src/twitest.ino.o -c -fno-exceptions -fno-threadsafe-statics -fpermissive -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega328p -DF_CPU=16000000L -DPLATFORMIO=30401 -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10619 -I/home/username/.platformio/packages/framework-arduinoavr/cores/arduino -I/home/username/.platformio/packages/framework-arduinoavr/variants/standard -I/home/username/Code/Arduino/twitest -I/home/username/.platformio/lib/Wire /home/username/Code/Arduino/twitest/twitest.ino.cpp
avr-gcc-ar rc .pioenvs/uno/libFrameworkArduinoVariant.a
avr-gcc-ranlib .pioenvs/uno/libFrameworkArduinoVariant.a
avr-g++ -o .pioenvs/uno/FrameworkArduino/CDC.o -c -fno-exceptions -fno-threadsafe-statics -fpermissive -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega328p -DF_CPU=16000000L -DPLATFORMIO=30401 -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10619 -I/home/username/.platformio/packages/framework-arduinoavr/cores/arduino -I/home/username/.platformio/packages/framework-arduinoavr/variants/standard /home/username/.platformio/packages/framework-arduinoavr/cores/arduino/CDC.cpp
avr-g++ -o .pioenvs/uno/FrameworkArduino/HardwareSerial.o -c -fno-exceptions -fno-threadsafe-statics -fpermissive -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega328p -DF_CPU=16000000L -DPLATFORMIO=30401 -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10619 -I/home/username/.platformio/packages/framework-arduinoavr/cores/arduino -I/home/username/.platformio/packages/framework-arduinoavr/variants/standard /home/username/.platformio/packages/framework-arduinoavr/cores/arduino/HardwareSerial.cpp
avr-g++ -o .pioenvs/uno/FrameworkArduino/HardwareSerial0.o -c -fno-exceptions -fno-threadsafe-statics -fpermissive -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega328p -DF_CPU=16000000L -DPLATFORMIO=30401 -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10619 -I/home/username/.platformio/packages/framework-arduinoavr/cores/arduino -I/home/username/.platformio/packages/framework-arduinoavr/variants/standard /home/username/.platformio/packages/framework-arduinoavr/cores/arduino/HardwareSerial0.cpp
avr-g++ -o .pioenvs/uno/FrameworkArduino/HardwareSerial1.o -c -fno-exceptions -fno-threadsafe-statics -fpermissive -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega328p -DF_CPU=16000000L -DPLATFORMIO=30401 -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10619 -I/home/username/.platformio/packages/framework-arduinoavr/cores/arduino -I/home/username/.platformio/packages/framework-arduinoavr/variants/standard /home/username/.platformio/packages/framework-arduinoavr/cores/arduino/HardwareSerial1.cpp
avr-g++ -o .pioenvs/uno/FrameworkArduino/HardwareSerial2.o -c -fno-exceptions -fno-threadsafe-statics -fpermissive -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega328p -DF_CPU=16000000L -DPLATFORMIO=30401 -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10619 -I/home/username/.platformio/packages/framework-arduinoavr/cores/arduino -I/home/username/.platformio/packages/framework-arduinoavr/variants/standard /home/username/.platformio/packages/framework-arduinoavr/cores/arduino/HardwareSerial2.cpp
avr-g++ -o .pioenvs/uno/FrameworkArduino/HardwareSerial3.o -c -fno-exceptions -fno-threadsafe-statics -fpermissive -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega328p -DF_CPU=16000000L -DPLATFORMIO=30401 -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10619 -I/home/username/.platformio/packages/framework-arduinoavr/cores/arduino -I/home/username/.platformio/packages/framework-arduinoavr/variants/standard /home/username/.platformio/packages/framework-arduinoavr/cores/arduino/HardwareSerial3.cpp
avr-g++ -o .pioenvs/uno/FrameworkArduino/IPAddress.o -c -fno-exceptions -fno-threadsafe-statics -fpermissive -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega328p -DF_CPU=16000000L -DPLATFORMIO=30401 -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10619 -I/home/username/.platformio/packages/framework-arduinoavr/cores/arduino -I/home/username/.platformio/packages/framework-arduinoavr/variants/standard /home/username/.platformio/packages/framework-arduinoavr/cores/arduino/IPAddress.cpp
avr-g++ -o .pioenvs/uno/FrameworkArduino/PluggableUSB.o -c -fno-exceptions -fno-threadsafe-statics -fpermissive -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega328p -DF_CPU=16000000L -DPLATFORMIO=30401 -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10619 -I/home/username/.platformio/packages/framework-arduinoavr/cores/arduino -I/home/username/.platformio/packages/framework-arduinoavr/variants/standard /home/username/.platformio/packages/framework-arduinoavr/cores/arduino/PluggableUSB.cpp
[... cut ...] 9 -I/home/username/.platformio/packages/framework-arduinoavr/cores/arduino -I/home/username/.platformio/packages/framework-arduinoavr/variants/standard -I/home/username/.platformio/lib/Wire /home/username/.platformio/lib/Wire/Wire.cpp
/home/username/.platformio/lib/Wire/Wire.cpp:30:17: fatal error: twi.h: No such file or directory
#include "twi.h"
^
compilation terminated.
*** [.pioenvs/uno/lib/Wire/Wire.o] Error 1
========================== [ERROR] Took 0.77 seconds ==========================
`

gepd, 2017-11-30:
If you had this problem yet, please try this:

Go to `Deviot Menu > Show Terminal` and write `pio update` 

It should update all platforms and tools in your setup. 

`/home/username/.platformio/lib/Wire/Wire.cpp:30:17: fatal error: twi.h: No such file or directory
#include "twi.h"`

This line make me believe it's a problem with the framework

If the problem continues, remove the folder in `~/platformio` it will remove all the platforms, frameworks and tools installed by PIO, restart ST and try to campile again, it should install all neccesary files again

m-chichikalov, 2017-12-01:
If that said Gepd does not help, try to change  #include "twi.h"  for   #include <twi.h>

jostidetosti, 2017-12-17:
Removing all folders in ~/platformio did the trick. Thanks!

closed 2017-12-17 by jostidetosti