gepd/Deviot #200: Compile  Alt+D, Alt+F with an option "--disable-auto-clean"

m-chichikalov, 2017-11-26:
// Before report an issue, please make sure it hasn't been reported before. 
-> Look's like no one mention about that.

Hi there...
If use the short cuts in Deviot  (for instance, Compile  Alt+D, Alt+F) it always do it without --disable-auto-clean and compile all files and all libraries. That is really annoying... Can I fix it in some way?
Have a nice one.
Thanks in advance.

m-chichikalov, 2017-11-27:
I simply added --disable-auto-clean in a list "cmd" in  compile.py . By the way I don't like this. What do you think would be a better approach?
[image]

gepd, 2017-11-27:
I could add an option in `Deviot > Compile/Upload Options` I think it could be the best way, what do you think?

m-chichikalov, 2017-11-27:
Yes, it's Ok. However, maybe we should ask author of platformio to change "--disable-auto-clean" on to "--enable-auto-clean"? For me it looks more logical. I don't know...

gepd, 2017-11-27:
you mean, change the behavior and make this option disabled by default?

m-chichikalov, 2017-11-27:
Yes, but probably your decision better, it gives choice...

gepd, 2017-11-27:
Yes, I'll add this option atp, I hope soon

m-chichikalov, 2017-11-29:
If you don't have time for that I could take this upon myself. It works?

m-chichikalov, 2017-11-29:
I installed  the deviot developer branch, and it looks like in this branch (or maybe because it uses PlatformIO 3.5.0rc2 ) both, compilation and uploading  go with disable cleaning.  
I double checked, my corrected files (compile.py and upload.py) were overwritten.

gepd, 2017-11-30:
> If you don't have time for that I could take this upon myself. It works?

Yes, you can keep using what you showed in the image above.

> I installed the deviot developer branch, and it looks like in this branch (or maybe because it uses PlatformIO 3.5.0rc2 ) both, compilation and uploading go with disable cleaning.
> I double checked, my corrected files (compile.py and upload.py) were overwritten.

PlatformIO version is not depending on Deviot branch, you can select the PIO version in `Deviot Menu > Options > Use Development Pio Version`

I've not read the changelog of the PIO `3.5`, it may be a change in this new version