gepd/Deviot #204: "Deviot needs Python 2.7 to work" - but it is installed

mklengel, 2017-12-29:
### Operating system
Windows 10 Professional 64 bit

### Board used
N/A

### Description of problem
After the installation Deviot complains about the installation of Python2.

I have installed Python3 and Python2 in parallel:
c:\program files\python36
c:\program files\python27

According to https://github.com/gepd/Deviot/wiki/Setup I created a file
C:\Users\mklen\AppData\Roaming\Sublime Text 3\Packages\User\Deviot\python.txt

My first try was:
C:\Program Files\Python27
C:\Program Files\Python27\Scripts

My second try was:
C:\PROGRA~1\Python27
C:\PROGRA~1\Python27\Scripts

In both cases the message "Deviot needs Python 2.7 to work" occures after the restart of Sublime Text.
I know the blank in the path is not recommended but I though this is history. I had no problems since now. The 8.3 notation should solve this, if this is really the problem in this case.

My PATH variable is set to Python3 - should be not a problem according the the setup instructions.

### Error(s) in ST Console? (View > Show Console)

### Steps to Reproduce

### Actual Results

### Expected Results

### Additional info

*You can edit the text above according to your need*

gepd, 2017-12-29:
Please use the develop version and try again https://github.com/gepd/Deviot/wiki/Setup#developer

That feature is not in the master branch yet.

Let me know if it solves your problem

mklengel, 2017-12-29:
The problem is solved - thx for the quick help.

Necessary is the python.txt file with the following content:
C:\Program Files\Python27
C:\Program Files\Python27\Scripts

Restarting ST results in a new file deviot.ini with the following content (shortened):

[config]
env_path = C:\Users\mklen\AppData\Roaming\Sublime Text 3\Packages\User\Deviot\penv;C:\Users\mklen\AppData\Roaming\Sublime Text 3\Packages\User\Deviot\penv\Scripts;C:\Python27\;C:\Python27\Scripts;C:\Program Files\Python27;C:\Program Files\Python27\Scripts
installed = True

The path components C:\Python27 and C:\Python27\Scripts are clearly wrong. They are not in my PATH variable (System and User) so it results from the Deviot setup. I think this should be fixed.

gepd, 2017-12-29:
Thanks for the feedback, I'll add a fix in a future release!

fiveangle, 2018-02-05:
I also encountered this during an upgrade somewhere along the way in the dev branch.  I ultimately resolved by complete uninstall/reinstall.  That's great @gepd if you found a root cause for it to prevent in the future 👍

gepd, 2018-02-11:
I've released `dev13` with a fix for this.

Please re-open if you keep having this issue

closed 2018-02-11 by gepd