gepd/Deviot #206: Deviot > Serial Port(s): no ports are listed

ogre, 2018-02-01:
### Operating system
windows7, sublime text 3143
tried with both stable and development branch of Deviot

### Board used
arduino Uno and/or arduno M0

### Description of problem
menu: Deviot > Serial Port(s)
shows only 'COM1' and 'Add IP manually'.
No other ports are listed (arduino.cc IDE works ok)

### Error(s) in ST Console? (View > Show Console)
no errors. full ST at bottoms

-thanks for any tips !
-michal

sublime text log:
------------------------------
DPI scale: 1
startup, version: 3143 windows x64 channel: stable
executable: /C/Program Files/Sublime Text 3/sublime_text.exe
working dir: /C/Program Files/Sublime Text 3
packages path: /C/Users/mfratczak/AppData/Roaming/Sublime Text 3/Packages
state path: /C/Users/mfratczak/AppData/Roaming/Sublime Text 3/Local
zip path: /C/Program Files/Sublime Text 3/Packages
zip path: /C/Users/mfratczak/AppData/Roaming/Sublime Text 3/Installed Packages
ignored_packages: ["Jedi - Python autocompletion", "Vintage"]
pre session restore time: 0.0967136
startup time: 0.113714
first paint time: 0.114714
reloading plugin Default.auto_indent_tag
reloading plugin Default.block
reloading plugin Default.comment
reloading plugin Default.convert_syntax
reloading plugin Default.copy_path
reloading plugin Default.delete_word
reloading plugin Default.detect_indentation
reloading plugin Default.duplicate_line
reloading plugin Default.echo
reloading plugin Default.exec
reloading plugin Default.fold
reloading plugin Default.font
reloading plugin Default.goto_line
reloading plugin Default.history_list
reloading plugin Default.indentation
reloading plugin Default.install_package_control
reloading plugin Default.kill_ring
reloading plugin Default.mark
reloading plugin Default.new_templates
reloading plugin Default.open_context_url
reloading plugin Default.open_in_browser
reloading plugin Default.pane
reloading plugin Default.paragraph
reloading plugin Default.paste_from_history
reloading plugin Default.profile
reloading plugin Default.quick_panel
reloading plugin Default.run_syntax_tests
reloading plugin Default.save_on_focus_lost
reloading plugin Default.scroll
reloading plugin Default.set_unsaved_view_name
reloading plugin Default.settings
reloading plugin Default.show_scope_name
reloading plugin Default.side_bar
reloading plugin Default.sort
reloading plugin Default.swap_line
reloading plugin Default.switch_file
reloading plugin Default.symbol
reloading plugin Default.transform
reloading plugin Default.transpose
reloading plugin Default.trim_trailing_white_space
reloading plugin Default.ui
reloading plugin CSS.css_completions
reloading plugin Diff.diff
reloading plugin HTML.encode_html_entities
reloading plugin HTML.html_completions
reloading plugin 0_package_control_loader.00-package_control
reloading plugin 0_package_control_loader.01-pygments
reloading plugin 0_package_control_loader.02-bz2
reloading plugin 0_package_control_loader.50-backrefs
reloading plugin 0_package_control_loader.50-markupsafe
reloading plugin 0_package_control_loader.50-pymdownx
reloading plugin 0_package_control_loader.50-python-markdown
reloading plugin 0_package_control_loader.50-pyyaml
reloading plugin 0_package_control_loader.51-python-jinja2
reloading plugin 0_package_control_loader.55-mdpopups
reloading plugin Alignment.Alignment
reloading plugin Bitbucket.sublime_bucket
reloading plugin BracketHighlighter.bh_core
reloading plugin BracketHighlighter.bh_logging
reloading plugin BracketHighlighter.bh_plugin
reloading plugin BracketHighlighter.bh_popup
reloading plugin BracketHighlighter.bh_regions
reloading plugin BracketHighlighter.bh_remove
reloading plugin BracketHighlighter.bh_rules
reloading plugin BracketHighlighter.bh_search
reloading plugin BracketHighlighter.bh_swapping
reloading plugin BracketHighlighter.bh_wrapping
reloading plugin BracketHighlighter.support
reloading plugin Calculate.calculate
reloading plugin Function Name Display.FunctionNameStatus
reloading plugin MayaSublime.MayaSublime
reloading plugin Package Control.1_reloader
reloading plugin Package Control.2_bootstrap
reloading plugin Package Control.Package Control
reloading plugin Super Calculator.Super Calculator
reloading plugin SwapStrings.SwapStrings
reloading plugin Anaconda.anaconda
reloading plugin Anaconda.setup
reloading plugin Anaconda.version
reloading plugin ColorPicker.sublimecp
reloading plugin Deviot (Arduino IDE).Deviot
reloading plugin SFTP.SFTP
reloading plugin SublimeCodeIntel.ordereddict
reloading plugin SublimeCodeIntel.SublimeCodeIntel
reloading plugin Terminal.Terminal
plugins loaded
reloading settings Packages/User/deviot.sublime-settings
Package Control: No updated packages
reloading settings Packages/User/deviot.sublime-settings

gepd, 2018-02-02:
Please, can you check if there is extra info in the sublime console, after check the list of ports and paste it here?
How many boards Deviot isn't showing?

ogre, 2018-02-02:
There isn't anything else in ST console. I pasted everything.
I tried with arduino UNO and M0 as that's all I have.
Could this be installation issue? How can I try to debug it ?

thanks !

gepd, 2018-02-02:
Go to the line [130 in the preferences_bridge.py](https://github.com/gepd/Deviot/blob/develop/libraries/preferences_bridge.py#L130) file and add:

`print(ports_list)`

* Restart sublime text
* Go to `Deviot Menu > Devices` 

After list the devices close the float window and paste here the sublime text console output

ogre, 2018-02-05:
I've noticed smth else. I can't select any board, because of failed config file search.
Below is my sublime steps and following is a log.

Am I missing something in installation? Shoudl I install platformio independently ?
I don't have "platformio.ini" file anywhere.

thanks !
----------------

steps:
1. Deviot > Select Board ( error in libraries\readconfig\__init__.py )
2. Deviot > Serial Ports ( prints [['Port komunikacyjny (COM1)', 'COM1']] )

log:
DPI scale: 1
startup, version: 3143 windows x64 channel: stable
executable: /C/Program Files/Sublime Text 3/sublime_text.exe
working dir: /C/Program Files/Sublime Text 3
packages path: /C/Users/mfratczak/AppData/Roaming/Sublime Text 3/Packages
state path: /C/Users/mfratczak/AppData/Roaming/Sublime Text 3/Local
zip path: /C/Program Files/Sublime Text 3/Packages
zip path: /C/Users/mfratczak/AppData/Roaming/Sublime Text 3/Installed Packages
ignored_packages: ["Jedi - Python autocompletion", "Vintage"]
pre session restore time: 0.0972873
startup time: 0.114287
first paint time: 0.115287
reloading plugin Default.auto_indent_tag
reloading plugin Default.block
reloading plugin Default.comment
reloading plugin Default.convert_syntax
reloading plugin Default.copy_path
reloading plugin Default.delete_word
reloading plugin Default.detect_indentation
reloading plugin Default.duplicate_line
reloading plugin Default.echo
reloading plugin Default.exec
reloading plugin Default.fold
reloading plugin Default.font
reloading plugin Default.goto_line
reloading plugin Default.history_list
reloading plugin Default.indentation
reloading plugin Default.install_package_control
reloading plugin Default.kill_ring
reloading plugin Default.mark
reloading plugin Default.new_templates
reloading plugin Default.open_context_url
reloading plugin Default.open_in_browser
reloading plugin Default.pane
reloading plugin Default.paragraph
reloading plugin Default.paste_from_history
reloading plugin Default.profile
reloading plugin Default.quick_panel
reloading plugin Default.run_syntax_tests
reloading plugin Default.save_on_focus_lost
reloading plugin Default.scroll
reloading plugin Default.set_unsaved_view_name
reloading plugin Default.settings
reloading plugin Default.show_scope_name
reloading plugin Default.side_bar
reloading plugin Default.sort
reloading plugin Default.swap_line
reloading plugin Default.switch_file
reloading plugin Default.symbol
reloading plugin Default.transform
reloading plugin Default.transpose
reloading plugin Default.trim_trailing_white_space
reloading plugin Default.ui
reloading plugin CSS.css_completions
reloading plugin Diff.diff
reloading plugin HTML.encode_html_entities
reloading plugin HTML.html_completions
reloading plugin 0_package_control_loader.00-package_control
reloading plugin 0_package_control_loader.01-pygments
reloading plugin 0_package_control_loader.02-bz2
reloading plugin 0_package_control_loader.50-backrefs
reloading plugin 0_package_control_loader.50-markupsafe
reloading plugin 0_package_control_loader.50-pymdownx
reloading plugin 0_package_control_loader.50-python-markdown
reloading plugin 0_package_control_loader.50-pyyaml
reloading plugin 0_package_control_loader.51-python-jinja2
reloading plugin 0_package_control_loader.55-mdpopups
reloading plugin Alignment.Alignment
reloading plugin Bitbucket.sublime_bucket
reloading plugin BracketHighlighter.bh_core
reloading plugin BracketHighlighter.bh_logging
reloading plugin BracketHighlighter.bh_plugin
reloading plugin BracketHighlighter.bh_popup
reloading plugin BracketHighlighter.bh_regions
reloading plugin BracketHighlighter.bh_remove
reloading plugin BracketHighlighter.bh_rules
reloading plugin BracketHighlighter.bh_search
reloading plugin BracketHighlighter.bh_swapping
reloading plugin BracketHighlighter.bh_wrapping
reloading plugin BracketHighlighter.support
reloading plugin Calculate.calculate
reloading plugin Function Name Display.FunctionNameStatus
reloading plugin MayaSublime.MayaSublime
reloading plugin Package Control.1_reloader
reloading plugin Package Control.2_bootstrap
reloading plugin Package Control.Package Control
reloading plugin Super Calculator.Super Calculator
reloading plugin SwapStrings.SwapStrings
reloading plugin Anaconda.anaconda
reloading plugin Anaconda.setup
reloading plugin Anaconda.version
reloading plugin ColorPicker.sublimecp
reloading plugin Deviot (Arduino IDE).Deviot
reloading plugin SFTP.SFTP
reloading plugin SublimeCodeIntel.ordereddict
reloading plugin SublimeCodeIntel.SublimeCodeIntel
reloading plugin Terminal.Terminal
plugins loaded
reloading settings Packages/User/deviot.sublime-settings
Package Control: Skipping automatic upgrade, last run at 2018-02-05 21:18:31, next run at 2018-02-05 22:18:31 or after
Traceback (most recent call last):
  File "C:\Users\mfratczak\AppData\Roaming\Sublime Text 3\Packages\Deviot (Arduino IDE)\libraries\quick_menu.py", line 72, in callback_board
    self.save_selected_board(board_id)
  File "C:\Users\mfratczak\AppData\Roaming\Sublime Text 3\Packages\Deviot (Arduino IDE)\libraries\preferences_bridge.py", line 42, in save_selected_board
    self.remove_ini_environment(board_id)
  File "C:\Users\mfratczak\AppData\Roaming\Sublime Text 3\Packages\Deviot (Arduino IDE)\platformio\pio_bridge.py", line 90, in remove_ini_environment
    config.read(file_path)
  File "C:\Users\mfratczak\AppData\Roaming\Sublime Text 3\Packages\Deviot (Arduino IDE)\libraries\readconfig\__init__.py", line 99, in read
    if(not path.exists(filepath)):
  File "./python3.3/genericpath.py", line 18, in exists
TypeError: can't specify None for path argument
[['Port komunikacyjny (COM1)', 'COM1']]

gepd, 2018-02-06:
Try this:

Go to: `Deviot Menu > Options > Remove Preferences File`

After that restart ST, it should install PlatformIO again, if that doesn't solves the problem, follow [this](https://github.com/gepd/Deviot/wiki/Setup#developer) steps to install the develop branch. 

Let me know if any of that solutions solves the problem

ogre, 2018-02-06:
Unfortunatelly, neithter reinstalling platform.io nor upgrading to devel branch changes anything.
I even did a clean install by first removing "C:\Users\mfratczak\AppData\Roaming\Sublime Text 3"

For now I switched to ubuntu which works fine, however I'll gladly assist if you wish to get to the bottom of the problem.

-thanks

gepd, 2018-02-08:
I'll install Windows 7 in a virtual machine to try to replicate the problem

ogre, 2018-02-15:
Today Deviot autoupdated (in development branch) and all works OK again.
Not sure if this is code changes or my windows configuration changed (I did not touch anything), but all is good !

thanks!

P.S. donation links is broken ?

gepd, 2018-02-16:
Thanks for the notification, yes I did some changes in the serial code.

I've updated the donation link! 👍 

Please re-open this issue if you find any problem related with the serial ports

closed 2018-02-16 by gepd