gepd/Deviot #207: Can't find framework files

crowecawcaw, 2018-02-02:
Before report an issue, please make sure it hasn't been reported before.

If you need information about an option or feature make sure to first look into the [Deviot Wiki](https://github.com/gepd/Deviot/wiki)

### Operating system
Mac OSX High Sierra
### Board used
STM32L475 IoT Discovery
### Description of problem
Devoit fails to find the STM32Cube framework.
### Error(s) in ST Console? (View > Show Console)
```
...
Scanning dependencies...
Library Dependency Graph ( http://bit.ly/configure-pio-ldf )
|-- <board>
|-- <tinyprintf>
|-- <wifi>
Compiling .pioenvs/disco_l152rb/src/main.o
Compiling .pioenvs/disco_l152rb/FrameworkCMSISVariant/startup_stm32l152xb.o
In file included from src/main.h:1:0,
from src/main.c:1:
lib/board/stm32l475e_iot01.h:57:27: fatal error: stm32l4xx_hal.h: No such file or directory
**********************************************************************
* Looking for stm32l4xx_hal.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:stm32l4xx_hal.h"
* Web  > http://platformio.org/lib/search?query=header:stm32l4xx_hal.h
*
**********************************************************************
#include "stm32l4xx_hal.h"
^
compilation terminated.
...
```
### Expected Results
The project combiles correctly with Atom's PlatformIO plugin. Devoit also correctly finds the framework paths and includes the files in the .clang_complete fil:
```
-I/Users/stephen/.platformio/packages/framework-stm32cube/l4/Drivers/STM32L4xx_HAL_Driver/Inc
```
I checked that the file actually exists there too.

Is there something that's required to get frameworks working with Devoit?

gepd, 2018-02-03:
It shoudn't be showing that error, Deviot doesn't require any extra step to work ith frameworks.

Can you try to upgrade PlatformIO from:

`Deviot Menu > Options > Upgrade PlatformIO`

if there is an update restart ST to the changes be applied

if the problem continues, please give an example code to replicate the problem

crowecawcaw, 2018-02-07:
It turned out to be a configuration problem with PIO. The Atom IDE had its own version of PIO that was slightly newer which allowed my project to run with the faulty configuration. I've gotten the project to compile under Devoit now too. Thanks for your help.

closed 2018-02-07 by crowecawcaw