gepd/Deviot #208: Using USBasp with Deviot

m4rcu5, 2018-02-12:
I'd like to use a USBasp to program my Atmega chip, for which I am using the following configuration in my `platformio.ini` file:

```
[env:prod]
platform = atmelavr
board = atmega328pb
board_f_cpu = 8000000L
framework = arduino
upload_protocol = usbasp
upload_flags = -Pusb
```

This way of uploading does not need a serial port, as USBasp shows up as a device in `/dev/bus/usb/...`. However the Deviot interface still prompts for a serial port on upload, blocking the process.

gepd, 2018-02-12:
Please check the changes in the last release `-dev14`

It shouldn't ask you to select a port, but you should select the protocol in `Deviot Menu > Programmer > Usbasp` 

If you don't do this last step, `upload_protocol` and `upload_flags` will be removed from `platformio.ini` I will fix this in the next release.

m4rcu5, 2018-02-12:
👍 Like a charm!

gepd, 2018-02-12:
fixed in `dev.15`, now it will automatically detect the programmer in `platformio.ini`

hackermagnet, 2018-11-11:
I still cannot use usbasp to upload my sketch. I select usbasp from the programmer menu but I get the "select port" window when I hit upload. What am I missing?

gepd, 2018-11-14:
Are you using the `dev` version?

hackermagnet, 2018-11-14:
I am.
I added
```upload_protocol = usbasp
upload_flags = -Pusb```
in platformio.ini and now I am able to upload. I thought it would be have done automatically when I chose USBasp as my programmer. Sorry to bother

gepd, 2018-11-16:
It should be done automatically, I'll check to commit a fix

hackermagnet, 2018-11-28:
Any news on this? Have you tried to replicate it?

jon-daemon, 2024-10-05:
Just an update after isntalling it again on a newer system (MacOS 13, ST4, Deviot dev 2.2.2, avrdude 8).
The file platformio.ini is still not updating itself after selecting the programmer.
I added this in `[env:uno]` to be able to flash using a usbasp.
```
upload_flags =
    -C
    ${platformio.packages_dir}/tool-avrdude/avrdude.conf
    -p
    $BOARD_MCU
    -c
    usbasp
upload_command = avrdude $UPLOAD_FLAGS -U flash:w:$SOURCE:i
```