gepd/Deviot #209: NEW DEV BUGS

tablatronix, 2018-02-12:
The new commits issues.

Ill just post them here before creating a bunch on new dev issues.

## terminal select all
Windows
The deviot terminal get selected all when it is running. It is very odd. as it updates all the text highlights and stays selected.

tablatronix, 2018-02-12:
### Serial pane
Serial monitor forces new layout pane, kind of annoying, I already use serial in pane, but I don't always want it, it also adds new panes when I already have one open, sometimes I have multiple terminals open also. Interesting feature, might need some polishing or options. 

If i already have custom layout, I don't need new panes created, so it conflicts with my layouts and messes stuff up

tablatronix, 2018-02-12:
Windows
Compile linter context errors not working anymore ( I know they were wrong line numbers and annoying but they were very useful, I stated getting used to them )
(FYI aside, arduino precompiler says it adds #linenum directives to source code, wonder if that would help)

gepd, 2018-02-12:
> Every time I close and restart serial, it creates an additional new pane, also.

What is you O.S?
I'm on windows and that is not the behavior I get when I restart it

> Also if I do shortcut for serial monitor , it does not seem to work always, but seems to only work when > that pane is focused, this was not the case in the past.

I've not change any code related with the shortcut(s), can you check if there is an error in the console when you run the shortcut?

> also please fix #171

it was fixed in `dev14`

tablatronix, 2018-02-12:
Sorry it has nothing to do with shortcuts, the menu does it also, open serial monitor, new pane is created, close serial tab, open serial monitor , new pane is created again ad infinitum

tablatronix, 2018-02-12:
I think the shortcut not working was something to do with the focus and text selection constantly changing. I will see if it happens agian.

tablatronix, 2018-02-12:
ahh with regards to the shortut issue, I did not realize that the shortcut toggled the serial monitor, I was toggling it on and off accidentally, while trying to restart it , It is hard to know when its on or off.

gepd, 2018-02-12:
> could the tab title be changed to reflect it ?

In `dev14` I've introduced a new feature that makes the status bar change the color to green when the serial monitor is connected, and red for 3 seconds when it's disconnected. 

However there are some issues I'm solving now related to this new feature (the color keeps green when you close the window)

The updates will come in a few hours

tablatronix, 2018-02-12:
Will this continue to work with multiple serial windows running ?

tablatronix, 2018-02-12:
[image]

not sure if gifs work but this is a gif of several issues

gepd, 2018-02-12:
I'll add a modification to show the color in the tab instead of the status bar, that way shouldn't be a problem

tablatronix, 2018-02-12:
This is how I use serial alot, sometimes even 4 uarts at a time ( debug uarts etc )
[image]

gepd, 2018-02-12:
> console doesn't work until its created

I'll solved this now, the rest of the issues are already solved. I'll it push it all together

tablatronix, 2018-02-12:
Sounds good, let me know if you need anything to be tested, I will follow up also on OSX later

gepd, 2018-02-12:
#### Released `dev.15`
> The deviot terminal and serial monitor text gets selected all(highlighted) when it is running. It is very odd. as it updates all the text highlights and stays selected.

Fixed

> Serial monitor forces new layout pane, kind of annoying, I already use serial in pane, but I don't always want it, it also adds new panes when I already have one open, sometimes I have multiple terminals open also (I change com port and start, so i can debug multiple uarts and devices at once). Interesting feature, might need some polishing or options.
>
> If i already have custom layout, I don't need new panes created, so it conflicts with my layouts and messes stuff up.

I've added the menu to edit the deviot preferences, go to `Preferences > Package Settings > Deviot > Settings`

Changes the option `terminal_direction` and `monitor_direction` to `self`. It will avoid to create a new panel

> Compile linter context errors not working anymore ( I know they were wrong line numbers and annoying but they were very useful, I stated getting used to them )
> (FYI aside, arduino precompiler says it adds #linenum directives to source code, wonder if that would help)

I hope to bring back this soon

> console doesn't work until its created

Fixed too

tablatronix, 2018-02-13:
osx first run update, it said could not import color_ something, some color file and deviot didnt start, I restarted and it works. Ill see if i logged it somewhere.

tablatronix, 2018-02-13:
osx serial not working

```
reloading settings Packages/User/deviot.sublime-settings
Exception in thread Thread-519:
Traceback (most recent call last):
  File "./python3.3/threading.py", line 901, in _bootstrap_inner
  File "./python3.3/threading.py", line 858, in run
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/serial.py", line 94, in start
    if is_available(self.port):
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/serial.py", line 182, in is_available
    serial.open()
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/pyserial/serialposix.py", line 271, in open
    self._reconfigure_port(force_update=True)
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/pyserial/serialposix.py", line 373, in _reconfigure_port
    if self._stopbits == serial.STOPBITS_ONE:
NameError: global name 'serial' is not defined
```

tablatronix, 2018-02-13:
upload error

```

top level value must be an array
Exception in thread Thread-563:
Traceback (most recent call last):
  File "./python3.3/threading.py", line 901, in _bootstrap_inner
  File "./python3.3/threading.py", line 858, in run
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/platformio/upload.py", line 40, in start_upload
    self.check_port_selected()
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/project_check.py", line 229, in check_port_selected
    config.read(ini_path)
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/readconfig/__init__.py", line 99, in read
    if(not path.exists(filepath)):
  File "./python3.3/genericpath.py", line 18, in exists
TypeError: can't specify None for path argument
```

gepd, 2018-02-13:
I've pushed a commit to solve the serial problems, checks and let me know if you keep getting that errors.

Upload compiles too, I suspect it's related with the serial issue

tablatronix, 2018-02-13:
here is that upgrade error, osx and windows.

```
reloading plugin Deviot (Arduino IDE).Deviot
Traceback (most recent call last):
  File "C:\Users\alverson\Documents\projects\bin\sublimetext\sublime_plugin.py", line 109, in reload_plugin
    m = importlib.import_module(modulename)
  File "./python3.3/importlib/__init__.py", line 90, in import_module
  File "<frozen importlib._bootstrap>", line 1584, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1565, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1532, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 584, in _check_name_wrapper
  File "<frozen importlib._bootstrap>", line 1022, in load_module
  File "<frozen importlib._bootstrap>", line 1003, in load_module
  File "<frozen importlib._bootstrap>", line 560, in module_for_loader_wrapper
  File "<frozen importlib._bootstrap>", line 868, in _load_module
  File "<frozen importlib._bootstrap>", line 313, in _call_with_frames_removed
  File "C:\Users\alverson\Documents\projects\bin\sublimetext\Data\Packages\Deviot (Arduino IDE)\Deviot.py", line 20, in <module>
    from .libraries.paths import getDeviotUserPath, getPluginName, status_color_folder
ImportError: cannot import name status_color_folder
```

deviot does not start, after restarting ST it works.

tablatronix, 2018-02-13:
terminal_direction not self still makes additional panes, not sure if you meant that was fixed or not.
`self` works as before 👍

tablatronix, 2018-02-13:
deviot console does not scroll if serial auto scroll off.

gepd, 2018-02-13:
No new panel is created when I change the setting in my machine:

[image]

tablatronix, 2018-02-13:
When NOT self

tablatronix, 2018-02-13:
windows
starting serial monitor opens deviot console.

gepd, 2018-02-13:
> When NOT self

That is the intention when is not self, I don't understand what do you expect

gepd, 2018-02-13:
> windows
> starting serial monitor opens deviot console.

In the gif I opened the serial monitor two times and it's not happening. I tried the shortcut and from menu, I can't replicate that

tablatronix, 2018-02-13:
hmm, if i set it to `right`, it still opens multiple panes over and over, that is what I meant.

tablatronix, 2018-02-13:
windows
clean monitor view does not appear to work, I assume it is meant to empty the buffer completely, or is this constrained to 20k lines like auto ?

gepd, 2018-02-13:
> hmm, if i set it to right, it still opens multiple panes over and over, that is what I meant.

yep, to avoid that you have to change the value to `self`

gepd, 2018-02-13:
> clean monitor view does not appear to work, I assume it is meant to empty the buffer completely, or is this constrained to 20k lines like auto ?
it should work regardless of the 20 lines, I'll check that

tablatronix, 2018-02-13:
did you see this one too, i slipped it in there above

windows
starting serial monitor opens deviot console.

gepd, 2018-02-13:
I tried to replicate it, but I can't. Look the gif I [posted](https://github.com/gepd/Deviot/issues/209#issuecomment-365310351), it doesn't open the deviot console

tablatronix, 2018-02-13:
[image]

gepd, 2018-02-13:
Is that windows?
which version?

tablatronix, 2018-02-13:
windows, should be latest of both

gepd, 2018-02-13:
> Also where can I find my .elf file ?

If you are using the PIO structure it will be located in the `.pioenvs` folder (same folder of your sketch) If you're not using it, it should be in `%temp%\Deviot\your_sketch_name\.pioenvs`

tablatronix, 2018-02-13:
Thanks I was trying to automate this, and maybe add it somehow and grab the environments from pio.

```
java -jar "%userprofile%\Documents\projects\electronics\EspStackTraceDecoder.jar" "%userprofile%\.platformio\packages\toolchain-xtensa\bin\xtensa-lx106-elf-addr2line" "%userprofile%\AppData\Local\Temp\Deviot\OnDemandConfigPortal\.pioenvs\nodemcuv2_dev\firmware.elf" "%userprofile%\Documents\projects\electronics\exception.txt"
```

gepd, 2018-02-13:
> I noticed in my deviot console the filenames are being clipped for the build assets.
eg. nodemcuv2_dev

Fixed with: https://github.com/gepd/Deviot/commit/db9396d09142179f37e2e2c1aa093c3c5485a0e3

gepd, 2018-02-13:
You can use:
`%temp%\Deviot\OnDemandConfigPortal\.pioenvs\nodemcuv2_dev\firmware.elf`

Are you using python?

tablatronix, 2018-02-13:
Gonna try so i can use it cross platform and add to ST

gepd, 2018-02-13:
It need to be an standalone plugin?

tablatronix, 2018-02-13:
Nope, ST has tools, build tools, and instant plugins, and macros etc.

tablatronix, 2018-02-14:
osx, probably windows also
I noticed now on compile error, with `"monitor_direction": "self",`, the monitor restarts in self pane (current focused pane), which is a bit annoying, so it stops the monitor running, opens a new one in the current pane for example, whereas it used to just resume as it was.

gepd, 2018-02-14:
That is because you are using the `upload` option instead of `compile`. 

`Upload` needs the serial port so it's closed

[image]

tablatronix, 2018-02-14:
Of course
Ill make a gif, my serial restarts in the focused pane, the left pane. On osx

tablatronix, 2018-02-14:
Windows
console still opens when starting serial monitor.

tablatronix, 2018-02-14:
so far very stable, been using it all day. Will try to test advanced stuff like ota etc.

gepd, 2018-02-15:
I've pushed a possible fix for the console issue (https://github.com/gepd/Deviot/commit/9b77d5cbab1bcab06ff88f12627639252d010a61)

I'll add two more features before release the next version
1. Linter context errors
2. An option to cancel the compile/upload process

tablatronix, 2018-02-16:
using osx today, also looking pretty good and stable.

Can you add a timestamp to the end of the deviot console when done with stuff, I know its at the top, but having an end stamp would be useful

tablatronix, 2018-02-16:
Does serial monitor log to a temp file, or it is just a buffer?

gepd, 2018-02-16:
> Can you add a timestamp to the end of the deviot console when done with stuff
Yes I can, I did that in the past for the same reason. When I rewrite the code I forgot to added again.

> Does serial monitor log to a temp file, or it is just a buffer?
It's just a buffer, but wouldn't it be a problem to be stored in a file. Do you want this for the trace decoder?

tablatronix, 2018-02-16:
Yes I figured I can run external stuff on it in real time if so, without hooking into ST to do it.

I have been using log highlighter for coloring the terminal,  serial ports and live log files, its pretty powerful.

gepd, 2018-02-16:
Excelent. Maybe I can log only the backtraces and exceptions, that way would be easy

tablatronix, 2018-02-16:
That might be too specific for other uses, unless you were adding specfic debug save to file feature maybe with regex i guess.

gepd, 2018-02-16:
>That might be too specific for other uses, unless you were adding specfic debug save to file feature maybe with regex i guess.

Yes, it's a good idea, I don't know how many people will need that feature, but maybe will be good to have it

I've added the

tablatronix, 2018-02-16:
only thing i noticed is that serial terminal takes a long time now to start back up, even adding a 5000 delay, i miss the boot sometimes.

gepd, 2018-02-17:
You mean, when you are using the serial monitor, uploads your firmware and the monitor starts again?

tablatronix, 2018-02-17:
Yup

tablatronix, 2018-02-17:
I think auto clean broke, on osx i have 77k lines

gepd, 2018-02-17:
good catch, fixed

[11 more replies]