gepd/Deviot #210: Features/ Improvements for v2.3

gepd, 2018-02-17:
- [ ]  Integrate `env_default` when a project is opened/loaded in Deviot
- [ ] Ask for a board when a new sketch is created #165 
- [ ] Make Use PlatformIO Structure a per-file setting. #165 
- [ ] Save monitor serial output in a file with the option to filter what will be stored #209 
- [ ] Improve Deviot output strings #209

Booli, 2018-03-26:
Cool! I would like to add a feature of possible. Can we add, for ESP8226 (and higher?) an option to upload the data folder to SPIFFS. Data folder needs to be in SRC (or adjusted in platformio.ini)

It can be done with PIO like this `pio run --target uploadfs`. I can already use the PIO terminal in DEVIOT to do the same thing.

gepd, 2018-04-01:
Sure I can @Booli I'll try it that this week, it seems to be very easy to implement

gepd, 2018-04-26:
@Booli I've readed in the documentation that you have `buildfs` and `uploadfs`, do you need both commands?

Booli, 2018-05-08:
I think having access to both is nice, building the FS can take some time, and you don't want to do that always if you flash multiple NodeMCU for example.