gepd/Deviot #212: is this normal compile ?

tablatronix, 2018-02-20:
dev version

I have a warning in a .h file, and noticed it outputs twice does the precompiler run twice on .h files?

```
Compiling .pioenvs\esp32doit-devkit-v1\src\OnDemandConfigPortal.ino.o
In file included from C:/Users/user/documents/projects/arduino/sketches/libraries/WiFiManager/examples/OnDemandConfigPortal/OnDemandConfigPortal.ino:1:0:
C:/Users/user/documents/projects/arduino/sketches/libraries/WiFiManager/WiFiManager.h:38:10: warning: #warning "WEBSERVER not implemented in espressif/esp32, see readme notes" [-Wcpp]
#warning "WEBSERVER not implemented in espressif/esp32, see readme notes"
^
Compiling .pioenvs\esp32doit-devkit-v1\libe17\WiFiManager\WiFiManager.o
In file included from C:\Users\user\documents\projects\arduino\sketches\libraries\WiFiManager\WiFiManager.cpp:14:0:
C:\Users\user\documents\projects\arduino\sketches\libraries\WiFiManager\WiFiManager.h:38:10: warning: #warning "WEBSERVER not implemented in espressif/esp32, see readme notes" [-Wcpp]
#warning "WEBSERVER not implemented in espressif/esp32, see readme notes"
^
Archiving .pioenvs\esp32doit-devkit-v1\libe17\libWiFiManager.a
Linking .pioenvs\esp32doit-devkit-v1\firmware.elf
```

gepd, 2018-02-20:
I noted the first time I used the WiFiManager, I have not had time to verify it but, seems like the warning will be displayed for each `#include` to the `.h` file

You can see the first call is made in the `OnDemandConfigPoral.ino`

`C:/Users/user/documents/projects/arduino/sketches/libraries/WiFiManager/examples/OnDemandConfigPortal/OnDemandConfigPortal.ino:1:0:`

And the second one in the `WiFiManager.cpp`

`C:\Users\user\documents\projects\arduino\sketches\libraries\WiFiManager\WiFiManager.cpp:14:0:`

Seems normal to me, does Arduino IDE display the same?

tablatronix, 2018-02-20:
Ill check arduino
I added a ifdef to .cpp and it still did it