gepd/Deviot #213: osx platformio.ini getting removed

tablatronix, 2018-02-22:
It looks like all my platformio.ini keep getting deleted.
says they are here

/private/tmp/Deviot/folder/platformio.ini

Is there somewhere else you can default build folder to , so this doesn't happen to people
I just changed my build folder, hopefully this fixes it.

gepd, 2018-02-23:
Getting removed how?

You can change the path from `Deviot > Option > Change build folder`

If you care about the `platformio.ini` file may be better to use the option `Deviot > Option > Use PlatformIO Structure` that way your file will be in the same folder as your sketch

tablatronix, 2018-02-23:
I have no idea, I guess osx wipes these folder now and then or on upgrades?

Doesn't platformio structure also require that deps and lib be in the same folder?

gepd, 2018-02-23:
It doesn't, you should be able to work same as you have been doing it until now. The no-platformio structure was thinked to people who doesn't know platformio, and would like to keep their files as Arduino IDE does, that is why it doesn't matter if it are removed or not.

PlatformIO will always look for global and local dependencies, so you should not be afraid to use that option

tablatronix, 2018-02-23:
I will check it out.

tablatronix, 2018-02-23:
Looks like pio structure needs a lib folder and stuff, I am a new file, paste and compile kind of person, I will have to get used to using "new sketch" I guess. well see. lol

gepd, 2018-02-23:
the lib folder is needed in both cases, the only change is that in no-pio structure that files are stored in the temp folder and in the other, at the same location of the sketch.

If you already have an sketch and you select `Use PlatformIO Structure`, Deviot will stop to search in the `temp` folder and will search in the same location of your sketch. If the `platformio.ini` file isn't found, it will be created, but you don't need to do anything, it will be automatically created.

Once the project is initialized, it will compile the file, if you have set an extra lib folder, it will be added to the `platformio.ini` file, you can check this looking for the flag `lib_extra_dirs`. If you let the `lib` folder empty none error should be displayed because your libraries are located in `lib_extra_dirs`

If that is not what are you experiencing it may be a bug

tablatronix, 2018-07-01:
I can confirm that OSX wipes the temp folders now and then ( I think on reboots or updates maybe )
I forgot to change my build location and lost all my platformio files.

Kind of annoying that the default location does this, maybe a different default location would be better to avoid confusion for new users. 

Either way changing build folder fixes it, not sure if you want to consider a bug or not.

gepd, 2018-07-05:
Yes, It may be consider a bug. I'm working in the installing code as soon as I finish it I'll change the default folder, I guess a better place would be `/Users/x/.platformio` which is where platformio stores all information