gepd/Deviot #214: osx serial ports do not show port id

tablatronix, 2018-03-09:
I cannot tell which device is which, I love seeing the names, but when there are dups, its a problem

[image]

gepd, 2018-04-01:
I'll try to do this tomorrow I don't have a OSX or Linux machine to change this

tablatronix, 2018-04-01:
Let me know where to start. I need to learn python at some point

gepd, 2018-04-02:
It was easier than I thought, It should be resolved with https://github.com/gepd/Deviot/commit/3f92c9d0dc88d262f98027896bc4d07ed6730c5a, let me know if that is what you expect

tablatronix, 2018-04-05:
This seems to be a breaking change, tested on windows, it says select port over and over in a loop.

gepd, 2018-04-05:
Fixed

tablatronix, 2018-06-14:
This seems to work and help, although its still a bit of a mystery lol
Can probably close.

[image]

gepd, 2018-06-18:
I see, I think it would be always confusing when you have multiples devices with the same spec.. 
A solution would be to add an alias, I could look for a kind of ID that will make possible to always assign an alias that can replace or be merged with the current name

tablatronix, 2018-06-18:
Yeah this is on my list to look into, I will look into having some kind of tty alias or have a naming convention, Not sure where OSX is getting the 15,16,17 suffixes but it must be caching them somewhere.

It is mostly an edge case.

gepd, 2019-02-18:
What about having an option that when it's active, it asks for an alias everytime you connect the device, I don't see other way to do it

tablatronix, 2019-02-19:
How would that work? How about indicating the last used port, colored highlighted? Might help a little bit. 

I have another issue where some usb shows 2 ports, ill check if its a pio issue or not

gepd, 2019-02-19:
Yes it may be tedious

Idicanting the last used port where?

tablatronix, 2019-02-20:
In the devices list, can it appear highlighted?

gepd, 2019-02-20:
Oh now I understand, ST does not provide anyway to highlight it, but what I can do is to add a character at the beginning of the active ports.

```js
* CP2102 USB to UART Bridge Controller...
CP2102 USB to UART Bridge Controller...
* CP2102 USB to UART Bridge Controller...
```

In the previous case the first one and the last one are active.

What do you think?

tablatronix, 2019-02-20:
yeah that might be cool, my main issue is that I have over 5 uarts, some are connected to test equipment like lab power supplies and multimeter, and some arduino etc, and of course they are all just slab_uart, or ftdi, etc, and it is hard to keep track of them lol

Even better would be remember the last device that was used and indicate it , probably requires saving stuff somewhere so might not be a good idea and prohibitive

gepd, 2019-02-20:
To remember what was the last device used isn't a problem, but where should it be displayed?

And what about the cases where you have devices with the same name, until now I didn't find any unique parameter to identify each device