gepd/Deviot #217: LLVM / Clang a thing?

RKouchoo, 2018-03-28:
Tried the sublime Clang plugins with clang installed. Tried on both Ubuntu and windows.

Is it a thing that has been/can be implemented easily?

Thanks

gepd, 2018-04-01:
I've tried https://github.com/niosus/EasyClangComplete the autocompletion seems to work fine after add this to the settings file:

```
"auto_complete_triggers":
  [
    {
      "selector": "source.c++ - string - comment - constant.numeric",
      "characters": ".>:",
    },
    {
      "selector": "source.c - string - comment - constant.numeric",
      "characters": ".>:",
    },
    {
      "selector": "source.objc++ - string - comment - constant.numeric",
      "characters": ".>: ",
    },
    {
      "selector": "source.objc - string - comment - constant.numeric",
      "characters": ".>: ",
    },
    {
      "selector": "source.cuda-c++ - string - comment - constant.numeric",
      "characters": ".>: ",
    },
    {
	"selector": "source.iot - string - comment - constant.numeric",
	"characters": ".>:"
    }
  ],
```

Let me know if that is what are you looking

manojbaishya, 2020-04-28:
## System info:
- Sublime Text version: v3.2.2 Build 3211
- Operating system: Xubuntu 16.04 LTS
- Clang version: 10.0.0-++20200412072846+50d7e5d5e7d-1\~exp1\~20200412053445.132
- Target: x86_64-pc-linux-gnu

## Issue
Hey @gepd, Deviot v2.2.1 clashes with EasyClangComplete v6.4.2 in my installation when I open C files belonging to a non-arduino project. I have added "deviot" to the list of accepted syntax schemes for the C language in the ECC settings. When both Deviot and ECC are enabled, ECC fails to load with the following error messages:

<details>
<summary>Console output and image:</summary>

```
---------- more code above ---------------

removed old modules: ['Debugger.__init__']
reloading plugin Deviot (Arduino IDE).Deviot
reloading plugin EasyClangComplete.EasyClangComplete
reloading plugin Gutter Color.file

---------- more code below ---------------

Traceback (most recent call last):
  File "/opt/sublime_text/sublime_plugin.py", line 298, in on_api_ready
    plc()
  File "/home/manoj/.config/sublime-text-3/Packages/EasyClangComplete/EasyClangComplete.py", line 85, in plugin_loaded
    module_reloader.ModuleReloader.reload_all(ignore_string='singleton')
  File "/home/manoj/.config/sublime-text-3/Packages/EasyClangComplete/plugin/utils/module_reloader.py", line 25, in reload_all
    ModuleReloader.reload_once(prefix, ignore_string)
  File "/home/manoj/.config/sublime-text-3/Packages/EasyClangComplete/plugin/utils/module_reloader.py", line 35, in reload_once
    for name, module in sys.modules.items():
RuntimeError: dictionary changed size during iteration

---------- more code below ---------------

Traceback (most recent call last):
  File "/opt/sublime_text/sublime_plugin.py", line 249, in synthesize_on_activated_async
    el.on_activated_async(v)
  File "/home/manoj/.config/sublime-text-3/Packages/EasyClangComplete/EasyClangComplete.py", line 303, in on_activated_async
    settings = EasyClangComplete.settings_manager.settings_for_view(view)
AttributeError: 'NoneType' object has no attribute 'settings_for_view'
```

[image]

[image]

</details>

However, when I disable Deviot, ECC loads fine.

<details>
<summary>Console output and image</summary>

```
[ECC:INFO   ]: Getting version from command: `clang++-10 -v`
[ECC:INFO   ]: init completer based on libclang
[ECC:INFO   ]: Found libclang library file: '/usr/lib/llvm-10/lib/libclang.so.1'
[ECC:INFO   ]: init completer based on libclang
```

[image]

</details>

## Info
There have been some work before on resolving this conflict in the ECC repo:
https://github.com/niosus/EasyClangComplete/issues/464
https://github.com/niosus/EasyClangComplete/issues/432
https://github.com/niosus/EasyClangComplete/commit/ced5542d4566e042edd1be2f991aff11044a71ba
https://github.com/niosus/EasyClangComplete/commit/d6cbd070c2ccd6150ee5d974f9c863d9fd08d83d

Entries titled Version 5.0.6 and Version 5.2.0 reference 'deviot' in the [release notes](https://github.com/niosus/EasyClangComplete/blob/dfaa1117121b7e6b39c22575f8c471c9896a8b00/messages/6.0.0.rst).

Also, why does Deviot automatically set my C file syntax as 'Deviot'? As per this [article](http://arduino.land/FAQ/content/1/37/en/what-are-the-different-file-extensions-for.html), I understand that .c and .h are valid Arduino syntaxes but unless there is a setting by the user to parse a C file as Deviot on a per project basis, it should not automatically hijack the file and alter the default syntax. I am not sure whether this is the cause of the conflict as I have already set the relevant settings. I write a lot of numerical solvers in C and ECC is a big productivity booster, however I would like to use Deviot for the occasional board project. Thanks for the awesome plugin!