gepd/Deviot #218: multiple plateformio cores after brew install

chienMouille, 2018-04-03:
Hello,

after installing deviot, plateformio wouldn't install by itself, promping that the command pip wasn't found. i thus installed plateformio through homebrew.

the plugin works great now! however it shows this message everytime i build a sketch: 
"Obsolete PIO Core v3.5.2 is used (previous was 3.5.3a8)
Please remove multiple PIO Cores from a system:
http://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system"

i can't find any other version of PIO in my system.
(and when hitting upgrade i get told that i have the last version)

any thought?

thanks a lot

gepd, 2018-04-09:
Hi @chienMouille can you try removing the folder `Packages\User\Deviot\penv` you cango to the package folder from `Sublime Menu > Preferences > Browse Packages`

chienMouille, 2018-04-09:
hey, 
i can't find any penv folder. in Packages\User\Deviot there is only 'deviot.ini' and 'pio'. and in Packages\Deviot (Arduino IDE) there are lots of things but no 'penv'

thank you!

gepd, 2018-04-10:
So your system has not multiples version, have you trying updating platformio?
Run this commands from the terminal: `platformio update` and `platformio upgrade`

Let me know if you keep seeing that warning after update

chienMouille, 2018-04-10:
in terminal,
when I run: `brew updgrade platformio`
I get: `Error: platformio 3.5.2 already installed`

when I run `platformio upgrade`
I get: `Error: Can not retrieve the latest PlatformIO version`

and when I run `platformio update`
I get: `Obsolete PIO Core v3.5.2 is used (previous was 3.5.3a8)
Please remove multiple PIO Cores from a system:
http://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system`

in ST3, I still get the same warning upon compilation.

chienMouille, 2018-06-11:
update on the problem:
due to some problems with loading a sketch on an esp32, I read in another issue I needed to update platformio to the latest version. trying to do so through deviot, I would read the following error:
' Error: Can not retrieve the latest PlatformIO version '.
as described in the previous post, homebrew wouldn't let me update either.

after various trials and errors, here is what fixed it:
- uninstalling platformio with homebrew
- removing deviot
- installing deviot (which would install platformio)
(at this point deviot would not actually work (which is why I installed platformio through homebrew initially) and trying to update would return 'no module named platformio found')
- removing deviot without removing platformio and installing it again

I can now update platoformio, load a sketch to my esp32, and the error message indicating several CORES disappeared.

I'm definitely not sure why, but I thought I'd share.

thanks for the help!