gepd/Deviot #219: KeyError: auth_upload

msanaullahsahar, 2018-04-11:
### Operating system
Windows 10

### Board
Espressif ESP32 Dev Module

### Environment
Espressif ESP32 Dev Module

### Description of problem

### Error(s) in ST Console? (View > Show Console)
DPI mode: per-monitor v2
startup, version: 3161 windows x64 channel: dev
executable: /C/Program Files/Sublime Text 3/sublime_text.exe
working dir: /C/Users/Sahar/Dropbox/nodeMCU/esp32
packages path: /C/Users/Sahar/AppData/Roaming/Sublime Text 3/Packages
state path: /C/Users/Sahar/AppData/Roaming/Sublime Text 3/Local
zip path: /C/Program Files/Sublime Text 3/Packages
zip path: /C/Users/Sahar/AppData/Roaming/Sublime Text 3/Installed Packages
ignored_packages: ["Vintage"]
pre session restore time: 0.107431
startup time: 0.153431
first paint time: 0.153431
reloading plugin Default.arithmetic
reloading plugin Default.auto_indent_tag
reloading plugin Default.block
reloading plugin Default.colors
reloading plugin Default.comment
reloading plugin Default.convert_color_scheme
reloading plugin Default.convert_syntax
reloading plugin Default.copy_path
reloading plugin Default.delete_word
reloading plugin Default.detect_indentation
reloading plugin Default.duplicate_line
reloading plugin Default.echo
reloading plugin Default.exec
reloading plugin Default.fold
reloading plugin Default.font
reloading plugin Default.goto_line
reloading plugin Default.history_list
reloading plugin Default.indentation
reloading plugin Default.install_package_control
reloading plugin Default.kill_ring
reloading plugin Default.mark
reloading plugin Default.new_templates
reloading plugin Default.open_context_url
reloading plugin Default.open_in_browser
reloading plugin Default.pane
reloading plugin Default.paragraph
reloading plugin Default.paste_from_history
reloading plugin Default.profile
reloading plugin Default.quick_panel
reloading plugin Default.rename
reloading plugin Default.run_syntax_tests
reloading plugin Default.save_on_focus_lost
reloading plugin Default.scroll
reloading plugin Default.set_unsaved_view_name
reloading plugin Default.settings
reloading plugin Default.show_scope_name
reloading plugin Default.side_bar
reloading plugin Default.sort
reloading plugin Default.swap_line
reloading plugin Default.switch_file
reloading plugin Default.symbol
reloading plugin Default.transform
reloading plugin Default.transpose
reloading plugin Default.trim_trailing_white_space
reloading plugin Default.ui
reloading plugin CSS.css_completions
reloading plugin Diff.diff
reloading plugin HTML.encode_html_entities
reloading plugin HTML.html_completions
reloading plugin ShellScript.ShellScript
reloading plugin 0_package_control_loader.00-package_control
reloading plugin Package Control.1_reloader
reloading plugin Package Control.2_bootstrap
reloading plugin Package Control.Package Control
reloading plugin Deviot (Arduino IDE).Deviot
plugins loaded
reloading settings Packages/User/deviot.sublime-settings
Package Control: Skipping automatic upgrade, last run at 2018-04-12 08:46:32, next run at 2018-04-12 09:46:32 or after
Exception in thread Thread-5:
Traceback (most recent call last):
  File "./python3.3/threading.py", line 901, in _bootstrap_inner
  File "./python3.3/threading.py", line 858, in run
  File "C:\Users\Sahar\AppData\Roaming\Sublime Text 3\Packages\Deviot (Arduino IDE)\platformio\upload.py", line 40, in start_upload
    self.check_port_selected()
  File "C:\Users\Sahar\AppData\Roaming\Sublime Text 3\Packages\Deviot (Arduino IDE)\libraries\project_check.py", line 224, in check_port_selected
    ports_list = self.get_ports_list()
  File "C:\Users\Sahar\AppData\Roaming\Sublime Text 3\Packages\Deviot (Arduino IDE)\libraries\preferences_bridge.py", line 132, in get_ports_list
    services = self.get_mdns_services()
  File "C:\Users\Sahar\AppData\Roaming\Sublime Text 3\Packages\Deviot (Arduino IDE)\libraries\preferences_bridge.py", line 344, in get_mdns_services
    return MDNS.formated_list()
  File "C:\Users\Sahar\AppData\Roaming\Sublime Text 3\Packages\Deviot (Arduino IDE)\libraries\mdns\mdns.py", line 111, in formated_list
    auth = value['auth_upload']
KeyError: 'auth_upload'

### Compiling
Compiling is fine with no error appear. Below is deviot console output.

[04/12/18 08:56:10] Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)

Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Espressif 32 > Espressif ESP32 Dev Module
SYSTEM: ESP32 240MHz 288KB RAM (1.25MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(light)
Collected 20 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Wire> v1.0
|-- <Adafruit SSD1306> v1.1.2
|   |-- <Wire> v1.0
|   |-- <SPI> v1.0
|   |-- <Adafruit GFX Library> v1.2.3
|   |   |-- <SPI> v1.0
|-- <Adafruit GFX Library> v1.2.3
|   |-- <SPI> v1.0
Calculating size .pioenvs\esp32dev\firmware.elf
text	   data	    bss	    dec	    hex	filename
136461	  38456	   2808	 177725	  2b63d	.pioenvs\esp32dev\firmware.elf
 [SUCCESS] Took 1.75 seconds 

[SUMMARY] 
Environment leonardo	[SKIP]
Environment esp32dev	[SUCCESS]
[SUCCESS] Took 1.75 seconds 
[04/12/18 08:56:12]

### Steps to Reproduce
On windows 10, I installed latest sublime text 3, then i installed python 2.7.14, then using package control in sublime text 3, i installed deviot plugin. deviot installed platform io itself. My code is in .cpp file.

### Actual Results
On clicking UPLOAD, Just this appears in deviot console "[ Deviot 2.2.1 ] Starting..." but in actual nothing happens. No upload.

### Expected Results
I am using it first time, so i am expecting to send this sketch to esp32.

### Additional info
On left clicking "Devices Alt+P, Alt+D" in Deviot menu, nothing appears or nothing happens.
No option to select serial port.

thewilloftheshadow, 2018-04-13:
also #221

gepd, 2018-04-26:
Can you install the [develop](https://github.com/gepd/Deviot/wiki/Setup#developer) branch and try again?

closed 2018-09-07 by msanaullahsahar