gepd/Deviot #220: upgrade platformio

thewilloftheshadow, 2018-04-13:
When i try to upgrade platformio, i get this. i have reinstalled deviot and nothing happened...
[image]

gepd, 2018-04-13:
Did Deviot show you any error during the setup?

thewilloftheshadow, 2018-04-14:
no i have no errors on setup. i run windows 10 home and sublime text 3

gepd, 2018-04-14:
Please go to: `Packages\User\Deviot` and copy here the content of the `deviot.ini` file

thewilloftheshadow, 2018-04-14:
i do not see a deviot.ini file under C:\Users\MYUSERNAME\AppData\Roaming\Sublime Text 3\Packages\Deviot (Arduino IDE)

thewilloftheshadow, 2018-04-14:
[config]
last_action = 2
env_path = C:\Users\willi\AppData\Roaming\Sublime Text 3\Packages\User\Deviot\penv;C:\Users\willi\AppData\Roaming\Sublime Text 3\Packages\User\Deviot\penv\Scripts;C:\Python27\;C:\Python27\Scripts;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Common Files\Roxio Shared\DLLShared\;C:\Program Files (x86)\Common Files\Roxio Shared\VHStoDVD\DLLShared\;C:\Users\willi\AppData\Local\Microsoft\WindowsApps;
installed = True
last_check_update = 2018-04-17 16:04:26.859297

gepd, 2018-04-14:
Can you check if you have any file here: `C:\Users\willi\AppData\Roaming\Sublime Text 3\Packages\User\Deviot\penv\Scripts` If not can you open a terminal window and run `platformio --version` and tell me if you get any output

thewilloftheshadow, 2018-04-16:
i have these files:
[image]

thewilloftheshadow, 2018-06-11:
[image]

thewilloftheshadow, 2018-06-11:
won't install either the main or dev versions of platformio and i get this message a lot

thewilloftheshadow, 2018-09-28:
any update on this?

closed 2022-08-16 by thewilloftheshadow