gepd/Deviot #221: select board

thewilloftheshadow, 2018-04-13:
When i try to select a board, nothing happens... i reinstalled deviot and tried to upgrade platformio (#220 )
can you help?

thewilloftheshadow, 2018-04-13:
the console for sublime shows this:
Traceback (most recent call last):
  File "C:\Program Files\Sublime Text 3\sublime_plugin.py", line 797, in run_
    return self.run()
  File "C:\Users\willi\AppData\Roaming\Sublime Text 3\Packages\Deviot (Arduino IDE)\commands\deviot_select_boards.py", line 7, in run
    items = Quick.boards_list()
  File "C:\Users\willi\AppData\Roaming\Sublime Text 3\Packages\Deviot (Arduino IDE)\libraries\quick_menu.py", line 89, in boards_list
    boards = boards_file.read_json()
  File "C:\Users\willi\AppData\Roaming\Sublime Text 3\Packages\Deviot (Arduino IDE)\libraries\file.py", line 57, in read_json
    return json.loads(file)
  File "./python3.3/json/__init__.py", line 316, in loads
  File "./python3.3/json/decoder.py", line 351, in decode
  File "./python3.3/json/decoder.py", line 369, in raw_decode
ValueError: No JSON object could be decoded

joaomartins33376, 2018-04-15:
These are the steps for fixing it:

First:
Deviot > options > Use Development Pio Version

Second:
Deviot > Rebuild Boards File

After this, you get the boards list.

gepd, 2018-04-26:
@thewilloftheshadow do you keep having this issue?

efduarte, 2018-04-27:
@gepd I'm not @thewilloftheshadow, but I'm having exactly the same issue on a fresh install of Ubuntu 18.04.

When I try to use the development PIO version, as suggested by @joaomartins33376, I get the following error in the console:

Traceback (most recent call last):
  File "/opt/sublime_text/sublime_plugin.py", line 744, in is_checked_
    raise ValueError("is_checked must return a bool", self)
ValueError: ('is_checked must return a bool', <Deviot (Arduino IDE).commands.deviot_developer_pio.DeviotDeveloperPio object at 0x7f355fbfdbd0>)
Traceback (most recent call last):
  File "/opt/sublime_text/sublime_plugin.py", line 744, in is_checked_
    raise ValueError("is_checked must return a bool", self)
ValueError: ('is_checked must return a bool', <Deviot (Arduino IDE).commands.deviot_developer_pio.DeviotDeveloperPio object at 0x7f355fbfdbd0>)
Traceback (most recent call last):
  File "/opt/sublime_text/sublime_plugin.py", line 744, in is_checked_
    raise ValueError("is_checked must return a bool", self)
ValueError: ('is_checked must return a bool', <Deviot (Arduino IDE).commands.deviot_developer_pio.DeviotDeveloperPio object at 0x7f355fbfdbd0>)

I'm under the impression that Deviot is failing to install PIO correctly: ".platformio" in the home directory is empty, and the boards.json has a not found error inside. In the meantime, removing Deviot, installing platformio with "sudo pip install -U platformio" and then reinstalling Deviot seems to solve the issue.

I'm not sure if all this information was useful, but I'm available to do some tests and provide further details if needed.

thewilloftheshadow, 2018-04-27:
i can't get platformio to install. will try @efduarte suggestion soon

gepd, 2018-04-28:
Please install the [develop](https://github.com/gepd/Deviot/wiki/Setup#developer) version, this problem should have been solved with https://github.com/gepd/Deviot/commit/4bb7dc4248021f9a3ef262ebc301e77becea1590

thewilloftheshadow, 2018-06-11:
@efduarte @gepd still can't get platformio to install. see #220

closed 2022-08-16 by thewilloftheshadow