gepd/Deviot #225: new scroll method?

tablatronix, 2018-04-18:
windows

Did you say you were testing a new scroll method, I don't remember is this was the same as before.

But my focus gets stolen when things are running like compile etc. focus switches to console while I am typing and upload finished etc.

tablatronix, 2018-04-19:
Serial monitor also seems to not go all the way to the bottom, it seems to be off by several lines maybe 5 or so ?

tablatronix, 2018-04-19:
hmm auto clean not working again

tablatronix, 2018-04-19:
hmm,  maybe I am not running dev again, not sure how that could happen

gepd, 2018-04-20:
It was that, you wasn't running the dev?

tablatronix, 2018-04-20:
I think so, i noticed both my machines say deviot 
1.2.1 or something but saw you commited a new version a while ago. Ill double check manually later

gepd, 2018-04-20:
yep, you are not in the dev branch: https://github.com/gepd/Deviot/commit/a379d55e31fd36c6081be701d8433a99cc6785b5

tablatronix, 2018-04-20:
Ill have to see why its not updating automatically

tablatronix, 2018-04-21:
Its a bug in pc

https://github.com/renerocksai/sublime_zk/issues/50#issuecomment-377724551

tablatronix, 2018-04-25:
I am still having issues with this, If I try typing when uploading I have issues, sometimes when I compile it tries to compile the serial window etc.
Focus seems to move around for auto scrolling...