gepd/Deviot #226: Can't upload to Teensy through Deviot menu "Upload"

hijomojo, 2018-04-23:
When trying to upload to Teensy by selecting the upload menu option, Deviot asks for a device to upload to (Deviot console shows "Select a Port From The List", and the same selection list appears as if clicking the "Device(s)" menu item), however there is no way to select anything relevant to Teensy, given they flash via the Teensy tool, not via serial. This means there doesn't appear to be a method to upload to Teensy via the Deviot menu or shortcuts.

There is no option to select "Not used" as per resolution for #29 , assuming that "Serial ports" became "Device(s)" since the builds referenced in that issue.

In the mean time I've been using "pio run -t upload" in the PIO terminal, after running "Build" from Deviot, and this works fine.

OS - Windows 10

Board used - Any Teensy

Error(s) in ST Console? - None

Steps to Reproduce - Open Arduino sketch (.ino), Deviot Menu > Select Board > Teensy XX, Deviot menu > Upload

Actual Results - Deviot Console says "Select a Port From The List" and device selection list appears

Expected Results - Arduino sketch is built > Teensy GUI tool is updated to newly built .hex file > Teensy is flashed by Teensy tool.

gepd, 2018-04-26:
I've pused a commit adding that option again. 

Make sure you are using the `developer` branch https://github.com/gepd/Deviot/wiki/Setup#developer

Let me know if it works as spected

gepd, 2018-04-29:
I think the solution may be to always ask what port do you want to work with after press the `Start/Stop Monitor` I've been working with multiples devices this week and I've feel the same.

closed 2019-08-08 by hijomojo

dchaplinsky, 2021-01-01:
That it still an issue for me using latest released version from package control

hijomojo, 2021-01-03:
> That it still an issue for me using latest released version from package control

When you chose the port via "Devices", do you have the "Not Used" option? This needs to be picked for upload to Teensy. Just in case for additional reference, the Teensy utility will only open when you "Upload". The "Compile" will not open it if it's not already open.

If you don't have that option, I guess you still need to switch to the dev version using the instructions at gepd's link above. 
What Deviot version are you on, if you "List Packages" in package control? For comparison, I'm on v2020.07.13.18.02.23 and have no issues uploading to Teensy.

dchaplinsky, 2021-01-06:
I don'h have that option indeed. But for some reason it started to work at some stage (there are no "Not Used" option, but I can upload/monitor serial). It seems, that Teensy little app should be launched manually to do so.

My Deviot version is v. 2.2.2.

hijomojo, 2021-01-06:
Okay, I guess if you want the option to be there (e.g. if it simplifies the process you're currently using) then I think you should switch to the dev channel as gepd's commit to fix this issue was added to to 2.3.0.dev2. The commit log seems a little tangled and I can't find where 2.2.2 came in, or 2.2.3 (current release version), in relation to 2.3.0.dev2, but the numbering would indicate the change isn't yet in the release version, despite it being 2 years newer.
For comparison I'm on 2.3.0.dev13

dchaplinsky, 2021-01-06:
Thanks, I'll try it!