gepd/Deviot #227: bug with settings file ?

tablatronix, 2018-04-25:
my console log is full of this

```
reloading settings Packages/User/deviot.sublime-settings
reloading settings Packages/User/deviot.sublime-settings
reloading settings Packages/User/deviot.sublime-settings
reloading settings Packages/User/deviot.sublime-settings
reloading settings Packages/User/deviot.sublime-settings
reloading settings Packages/User/deviot.sublime-settings
top level value must be an array
reloading settings Packages/User/deviot.sublime-settings
reloading settings Packages/User/deviot.sublime-settings
reloading settings Packages/User/deviot.sublime-settings
reloading settings Packages/User/deviot.sublime-settings
reloading settings Packages/User/deviot.sublime-settings
reloading settings Packages/User/deviot.sublime-settings
reloading settings Packages/User/deviot.sublime-settings
reloading settings Packages/User/deviot.sublime-settings
reloading settings Packages/User/deviot.sublime-settings
reloading settings Packages/User/deviot.sublime-settings
reloading settings Packages/User/deviot.sublime-settings
top level value must be an array
```

Seems odd, and totally not normal

gepd, 2018-04-25:
It isn't a bug, Deviot relies on the preference settings to store some data, like the boards used, current serial port selected, and other options, some of them are automatically updated when you compile or upload a sketch, ST just reloads the file each time it's modified.

gepd, 2018-04-26:
Fixed with https://github.com/gepd/Deviot/commit/6aa43651716f3653018aff162b93e04639cbd18b

closed 2018-06-26 by gepd