gepd/Deviot #230: [Decode error - output not utf-8]

avrosmalen, 2018-04-30:
When trying to compile any random piece of code for Arduino I'm getting lots of 

[Decode error - output not utf-8]

gepd, 2018-04-30:
Please next time use the issue template. To help you I need more information;

What is your O.S?
What is the Deviot version used?
What are the steps to reproduce this problem?

avrosmalen, 2018-05-05:
Sorry for the novice question, but how do I remove the "needs more info" label as the request for more info has been answered?

avrosmalen, 2018-05-11:
chcp gives Active code page 850
I have no clue what that means, but Google tells me it has something to do with this problem

Compiling
void setup()
{
	//
}

void loop()
{
	//
}

gives

[ Deviot 2.2.1 ] Starting...
[Decode error - output not utf-8]

PLATFORM: Atmel AVR > Arduino Mega or Mega 2560 ATmega2560 (Mega 2560)
SYSTEM: ATMEGA2560 16MHz 8KB RAM (248KB Flash)
avr-g++ -o "C:\Users\Anton\AppData\Local\Temp\Deviot\1525998082\src\1525998082.ino.cpp" -x c++ -fpreprocessed -dD -E "c:\users\anton\appdata\local\temp\tmpov_tdg"
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(light)
Collected 61 compatible libraries
Scanning dependencies...
No dependencies
avr-g++ -o .pioenvs\megaatmega2560\src\1525998082.ino.cpp.o -c -fno-exceptions -fno-threadsafe-statics -fpermissive -std=gnu++11 -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega2560 -DF_CPU=16000000L -DPLATFORMIO=30502 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -DARDUINO=10805 -IC:\Users\Anton\.platformio\packages\framework-arduinoavr\cores\arduino -IC:\Users\Anton\.platformio\packages\framework-arduinoavr\variants\mega -Isrc src\1525998082.ino.cpp
avr-g++ -o .pioenvs\megaatmega2560\FrameworkArduino\WMath.cpp.o -c -fno-exceptions -fno-threadsafe-statics -fpermissive -std=gnu++11 -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega2560 -DF_CPU=16000000L -DPLATFORMIO=30502 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -DARDUINO=10805 -IC:\Users\Anton\.platformio\packages\framework-arduinoavr\cores\arduino -IC:\Users\Anton\.platformio\packages\framework-arduinoavr\variants\mega C:\Users\Anton\.platformio\packages\framework-arduinoavr\cores\arduino\WMath.cpp
avr-g++ -o .pioenvs\megaatmega2560\FrameworkArduino\WString.cpp.o -c -fno-exceptions -fno-threadsafe-statics -fpermissive -std=gnu++11 -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega2560 -DF_CPU=16000000L -DPLATFORMIO=30502 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -DARDUINO=10805 -IC:\Users\Anton\.platformio\packages\framework-arduinoavr\cores\arduino -IC:\Users\Anton\.platformio\packages\framework-arduinoavr\variants\mega C:\Users\Anton\.platformio\packages\framework-arduinoavr\cores\arduino\WString.cpp
avr-gcc -x assembler-with-cpp -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega2560 -DF_CPU=16000000L -DPLATFORMIO=30502 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -DARDUINO=10805 -IC:\Users\Anton\.platformio\packages\framework-arduinoavr\cores\arduino -IC:\Users\Anton\.platformio\packages\framework-arduinoavr\variants\mega -c -o .pioenvs\megaatmega2560\FrameworkArduino\_wiring_pulse.S.o C:\Users\Anton\.platformio\packages\framework-arduinoavr\cores\arduino\_wiring_pulse.S
avr-g++ -o .pioenvs\megaatmega2560\FrameworkArduino\abi.cpp.o -c -fno-exceptions -fno-threadsafe-statics -fpermissive -std=gnu++11 -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega2560 -DF_CPU=16000000L -DPLATFORMIO=30502 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -DARDUINO=10805 -IC:\Users\Anton\.platformio\packages\framework-arduinoavr\cores\arduino -IC:\Users\Anton\.platformio\packages\framework-arduinoavr\variants\mega C:\Users\Anton\.platformio\packages\framework-arduinoavr\cores\arduino\abi.cpp
avr-gcc -o .pioenvs\megaatmega2560\FrameworkArduino\hooks.c.o -c -std=gnu11 -fno-fat-lto-objects -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega2560 -DF_CPU=16000000L -DPLATFORMIO=30502 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -DARDUINO=10805 -IC:\Users\Anton\.platformio\packages\framework-arduinoavr\cores\arduino -IC:\Users\Anton\.platformio\packages\framework-arduinoavr\variants\mega C:\Users\Anton\.platformio\packages\framework-arduinoavr\cores\arduino\hooks.c
avr-g++ -o .pioenvs\megaatmega2560\FrameworkArduino\main.cpp.o -c -fno-exceptions -fno-threadsafe-statics -fpermissive -std=gnu++11 -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega2560 -DF_CPU=16000000L -DPLATFORMIO=30502 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -DARDUINO=10805 -IC:\Users\Anton\.platformio\packages\framework-arduinoavr\cores\arduino -IC:\Users\Anton\.platformio\packages\framework-arduinoavr\variants\mega C:\Users\Anton\.platformio\packages\framework-arduinoavr\cores\arduino\main.cpp
avr-g++ -o .pioenvs\megaatmega2560\FrameworkArduino\new.cpp.o -c -fno-exceptions -fno-threadsafe-statics -fpermissive -std=gnu++11 -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega2560 -DF_CPU=16000000L -DPLATFORMIO=30502 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -DARDUINO=10805 -IC:\Users\Anton\.platformio\packages\framework-arduinoavr\cores\arduino -IC:\Users\Anton\.platformio\packages\framework-arduinoavr\variants\mega C:\Users\Anton\.platformio\packages\framework-arduinoavr\cores\arduino\new.cpp
avr-gcc -o .pioenvs\megaatmega2560\FrameworkArduino\wiring.c.o -c -std=gnu11 -fno-fat-lto-objects -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega2560 -DF_CPU=16000000L -DPLATFORMIO=30502 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -DARDUINO=10805 -IC:\Users\Anton\.platformio\packages\framework-arduinoavr\cores\arduino -IC:\Users\Anton\.platformio\packages\framework-arduinoavr\variants\mega C:\Users\Anton\.platformio\packages\framework-arduinoavr\cores\arduino\wiring.c
avr-gcc -o .pioenvs\megaatmega2560\FrameworkArduino\wiring_analog.c.o -c -std=gnu11 -fno-fat-lto-objects -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega2560 -DF_CPU=16000000L -DPLATFORMIO=30502 -DARDUINO_AVR_MEGA2560 -DARD [... cut ...] libFrameworkArduinoVariant.a .pioenvs\megaatmega2560\libFrameworkArduino.a -lm -Wl,--end-group
avr-objcopy -O ihex -R .eeprom .pioenvs\megaatmega2560\firmware.elf .pioenvs\megaatmega2560\firmware.hex
avr-size --mcu=atmega2560 -C -d .pioenvs\megaatmega2560\firmware.elf
AVR Memory Usage
----------------
Device: atmega2560

Program:     662 bytes (0.3% Full)
(.text + .data + .bootloader)

Data:          9 bytes (0.1% Full)
(.data + .bss + .noinit)

 [SUCCESS] Took 2.02 seconds 

[05/11/18 02:24:03]