gepd/Deviot #232: I can't use Deviot on Ubuntu

zhaojiansong, 2018-05-07:
### Operating system
ubuntu 18.04/16.04
### Description of problem
In Ubutntu, i can't download files and update
### Error(s) in ST Console? (View > Show Console)
[image]
[image]

gepd, 2018-05-08:
can you please go to:

1. `ST Menu > Deviot > Options > Remove Preferences Files` and restart sublime text

2. Deviot will start to install PlatformIO again, if it fails copy the output of the ST console (`ST Menu > View > Show Console`) and paste it here.

You can manually install PlatformIO from the terminal with `pip install -U platformio` after that repeat the step 1 and Deviot should work

zhaojiansong, 2018-05-09:
DPI scale: 1
startup, version: 3170 linux x64 channel: stable
executable: /snap/sublime-text/9/sublime_text
working dir: /
packages path: /home/ghost-girl/.config/sublime-text-3/Packages
state path: /home/ghost-girl/.config/sublime-text-3/Local
zip path: /snap/sublime-text/9/Packages
zip path: /home/ghost-girl/.config/sublime-text-3/Installed Packages
ignored_packages: ["Vintage"]
pre session restore time: 0.0750039
startup time: 0.201085
first paint time: 0.296201
reloading plugin Default.arithmetic
reloading plugin Default.auto_indent_tag
reloading plugin Default.block
reloading plugin Default.colors
reloading plugin Default.comment
reloading plugin Default.convert_color_scheme
reloading plugin Default.convert_syntax
reloading plugin Default.copy_path
reloading plugin Default.delete_word
reloading plugin Default.detect_indentation
reloading plugin Default.duplicate_line
reloading plugin Default.echo
reloading plugin Default.exec
reloading plugin Default.fold
reloading plugin Default.font
reloading plugin Default.goto_line
reloading plugin Default.history_list
reloading plugin Default.indentation
reloading plugin Default.install_package_control
reloading plugin Default.kill_ring
reloading plugin Default.mark
reloading plugin Default.new_templates
reloading plugin Default.open_context_url
reloading plugin Default.open_in_browser
reloading plugin Default.pane
reloading plugin Default.paragraph
reloading plugin Default.paste_from_history
reloading plugin Default.profile
reloading plugin Default.quick_panel
reloading plugin Default.rename
reloading plugin Default.run_syntax_tests
reloading plugin Default.save_on_focus_lost
reloading plugin Default.scroll
reloading plugin Default.set_unsaved_view_name
reloading plugin Default.settings
reloading plugin Default.show_scope_name
reloading plugin Default.side_bar
reloading plugin Default.sort
reloading plugin Default.swap_line
reloading plugin Default.switch_file
reloading plugin Default.symbol
reloading plugin Default.transform
reloading plugin Default.transpose
reloading plugin Default.trim_trailing_white_space
reloading plugin Default.ui
reloading plugin CSS.css_completions
reloading plugin Diff.diff
reloading plugin HTML.encode_html_entities
reloading plugin HTML.html_completions
reloading plugin ShellScript.ShellScript
reloading plugin 0_package_control_loader.00-package_control
reloading plugin 0_package_control_loader.01-pygments
reloading plugin 0_package_control_loader.50-backrefs
reloading plugin 0_package_control_loader.50-markupsafe
reloading plugin 0_package_control_loader.50-pymdownx
reloading plugin 0_package_control_loader.50-python-markdown
reloading plugin 0_package_control_loader.50-pyyaml
reloading plugin 0_package_control_loader.51-python-jinja2
reloading plugin 0_package_control_loader.55-jsonschema
reloading plugin 0_package_control_loader.55-mdpopups
reloading plugin Alignment.Alignment
reloading plugin All Autocomplete.all_views_completions
reloading plugin BracketHighlighter.bh_core
reloading plugin BracketHighlighter.bh_logging
reloading plugin BracketHighlighter.bh_plugin
reloading plugin BracketHighlighter.bh_popup
reloading plugin BracketHighlighter.bh_regions
reloading plugin BracketHighlighter.bh_remove
reloading plugin BracketHighlighter.bh_rules
reloading plugin BracketHighlighter.bh_search
reloading plugin BracketHighlighter.bh_swapping
reloading plugin BracketHighlighter.bh_wrapping
reloading plugin BracketHighlighter.support
reloading plugin Colorsublime.colorsublime-plugin
reloading plugin DocBlockr.jsdocs
reloading plugin GBK Support.sublime_gbk
reloading plugin GitGutter.plugin
reloading plugin Package Control.1_reloader
reloading plugin Package Control.2_bootstrap
reloading plugin Package Control.Package Control
reloading plugin SublimeLinter-cppcheck.linter
reloading plugin SublimeLinter.__init__
reloading plugin SublimeLinter.active_linters_view
reloading plugin SublimeLinter.busy_indicator_view
reloading plugin SublimeLinter.goto_commands
reloading plugin SublimeLinter.highlight_view
reloading plugin SublimeLinter.log_handler
reloading plugin SublimeLinter.message_view
reloading plugin SublimeLinter.panel_view
reloading plugin SublimeLinter.status_bar_view
reloading plugin SublimeLinter.sublime_linter
reloading plugin WordCount.WordCount
reloading plugin Codecs33.Codecs33
reloading plugin ConvertToUTF8.ConvertToUTF8
reloading plugin CoolFormat.Coolformat
reloading plugin Deviot (Arduino IDE).Deviot
reloading plugin SublimeCodeIntel.ordereddict
reloading plugin SublimeCodeIntel.SublimeCodeIntel
plugins loaded
reloading settings Packages/User/deviot.sublime-settings
Package Control: Skipping automatic upgrade, last run at 2018-05-09 20:36:15, next run at 2018-05-09 21:36:15 or after
Traceback (most recent call last):
  File "/snap/sublime-text/9/sublime_plugin.py", line 959, in is_checked_
    raise ValueError("is_checked must return a bool", self)
ValueError: ('is_checked must return a bool', <Deviot (Arduino IDE).commands.deviot_developer_pio.DeviotDeveloperPio object at 0x7fa2853a9090>)
Traceback (most recent call last):
  File "/snap/sublime-text/9/sublime_plugin.py", line 959, in is_checked_
    raise ValueError("is_checked must return a bool", self)
ValueError: ('is_checked must return a bool', <Deviot (Arduino IDE).commands.deviot_developer_pio.DeviotDeveloperPio object at 0x7fa2853a9090>)
Traceback (most recent call last):
  File "/snap/sublime-text/9/sublime_plugin.py", line 959, in is_checked_
    raise ValueError("is_checked must return a bool", self)
ValueError: ('is_checked must return a bool', <Deviot (Arduino IDE).commands.deviot_developer_pio.DeviotDeveloperPio object at 0x7fa2853a9090>)
reloading settings Packages/User/deviot.sublime-settings

zhaojiansong, 2018-05-22:
yes,i am tired again and again

itechnocrat, 2018-07-03:
I have the same problem.
Going crazy!

1. The plug-in for the MS VS Code installs the PlatformIO into ~/.platformio
2. The Deviot installs the PlatformIO into ~/.config/sublime-text-3/Packages/User/Deviot/penv/bin
Deviot reports that:

```
PlatformIO is not installed
Preparing to install it. Please be patient it; may take a while.
Downloading necessary files...
Extracting files...
Installing PlatformIO...
There was an error
Collecting platformio
  Using cached https://files.pythonhosted.org/packages/8c/f9/1da2df1b4b66e5ec777df79613ea3e8db3228f6121c7ae15803d79bc5157/platformio-3.5.4-py27-none-any.whl
Requirement not upgraded as not directly required: lockfile<0.13,>=0.9.1 in /home/technocrat/.local/lib/python2.7/site-packages (from platformio) (0.12.2)
Requirement not upgraded as not directly required: colorama in /home/technocrat/.local/lib/python2.7/site-packages (from platformio) (0.3.9)
Requirement not upgraded as not directly required: semantic-version<3,>=2.5.0 in /home/technocrat/.local/lib/python2.7/site-packages (from platformio) (2.6.0)
Requirement not upgraded as not directly required: bottle<0.13 in /home/technocrat/.local/lib/python2.7/site-packages (from platformio) (0.12.13)
Requirement not upgraded as not directly required: requests<3,>=2.4.0 in /home/technocrat/.local/lib/python2.7/site-packages (from platformio) (2.19.1)
Requirement not upgraded as not directly required: click<6,>=5 in /home/technocrat/.local/lib/python2.7/site-packages (from platformio) (5.1)
Requirement not upgraded as not directly required: pyserial!=3.3,<4,>=3 in /home/technocrat/.local/lib/python2.7/site-packages (from platformio) (3.4)
Requirement not upgraded as not directly required: certifi>=2017.4.17 in /home/technocrat/.local/lib/python2.7/site-packages (from requests<3,>=2.4.0->platformio) (2018.4.16)
Requirement not upgraded as not directly required: chardet<3.1.0,>=3.0.2 in /home/technocrat/.local/lib/python2.7/site-packages (from requests<3,>=2.4.0->platformio) (3.0.4)
Requirement not upgraded as not directly required: urllib3<1.24,>=1.21.1 in /home/technocrat/.local/lib/python2.7/site-packages (from requests<3,>=2.4.0->platformio) (1.23)
Requirement not upgraded as not directly required: idna<2.8,>=2.5 in /home/technocrat/.local/lib/python2.7/site-packages (from requests<3,>=2.4.0->platformio) (2.7)
Installing collected packages: platformio
Could not install packages due to an EnvironmentError: [Errno 13] Permission denied: '/usr/local/lib/python2.7/dist-packages/platformio-3.5.4.dist-info'
Consider using the `--user` option or check the permissions.

Installation finished, now you can enjoy Deviot!
```  

Nothing helps.  
The development repository does not help.  
It does not help to delete the preferences file.  
Does not help Deviot - Options - Upgrade the PlatformIO.  
Does not help Deviot - Options - Use Development Pio version.  

In one of the last two options, the ~/.config/sublime-text-3/Packages/User/Deviot/penv/bin directory does not contain a PlatformIO.

3. Python installs the platform into ~/.local/lib/python2.7/site-packages/platformio and into ~/.local/lib/python2.7/site-packages/platformio-3.5.4.dist-info  

Are you all crazy?

itechnocrat, 2018-07-03:
I found how to get around this crazy.
1. Install MS VS Code
2. Install the extension PlatformIO for MS VS Code
3. add to ~/.profile `export PATH=$PATH:~/.platformio/penv/bin`
4. Install the Deviot in the Sublime Text 3 through the package manager
Deviot will see the PlatformIO and everything will work.

gepd, 2018-07-03:
I'm working in setup process to avoid this kind of fixes