gepd/Deviot #234: platformio.ini is not an IoT File

Bob-the-Kuhn, 2018-05-21:
I'm having startup problems with Deviot on macOS.  Probably a macOS newbie issue.

I don't think Deviot is pointed at my PlatformIO sketch.  The Deviot menu has "Select Environment" greyed out and when I click on compile I get "[ Deviot 2.2.1 ] Starting...   platformio.ini is not an IoT File " in the window on the bottom.

Here's what I've done so far:
1. Installed Deviot into Sublime 3.1.1 via the package manager.
2. Opened an existing PlatformIO folder via Sublime's File ...  Open ... dialog.
3. Changed the build folder to the folder that contains the platformio.ini
4. Selected "Use Platformio Structure"
5. Platformio.ini Untouch is selected
6. Opened & closed Sublime
7. Eventually clicked "Use Development PIO Version" (didn't make a difference)
8. Clicked "Upgrade PlatformIO" and got "You're up-to-date!   PlatformIO 3.5.3b5 is currently the newest version available."
9. Under Project clicked "Add Folder to Project", saved the project and re-opened it.
10. Went to Atom and used PlatformIO to sucessfully compile.
11. Closed Atom and came back to Deviot

gepd, 2018-05-21:
Hi @Bob-the-Kuhn you should follow this steps to compile a project in Deviot

1. Go to your PIO project and open any file inside the `src` folder (if there's more than one)
2. Make sure the Sublime Text tab with the file recently opened is in focus and go to `Deviot > Select Environment` and select the environment you want to Compile/Upload
3. Compile or upload the sketch, if you choose the last option you will be asked to select a serial port.

You can read more information about the menu options in:  https://github.com/gepd/Deviot/wiki/Deviot-Menu 

Let me know if you need more help or information

Bob-the-Kuhn, 2018-05-21:
I've downloaded Marlin 2.0.  Now I want to make some changes to the configuration.h file using Sublime and then use Deviot to compile & upload it.

If I open the Configuration.h file in Deviot the select environment option becomes available.

I've been through the WIKI & I'm wondering if Deviot can handle projects where the source files are in multiple subdirectories.

Marlin 1.x is structured exactly as shown in the WIKI.  Marlin 2.0 splits the source files out into multiple subdirectories, each of which usually has at lease one subdirectory.

gepd, 2018-05-21:
Can you please paste here the output of the Sublime Text console (`ST Menu > View > Show Console`) after select the Deviot menu?

Always make sure that the tab with the file is in focus, Deviot check the project based in that file, the `Open folder` options doesn't have influence.

Have you tried to compile any other file in Deviot? (Linux and macOS) 
just to make sure PlatformIO is being recognized

Bob-the-Kuhn, 2018-05-21:
After selecting some text in Configuration.h:
```
[ Deviot 2.2.1 ]
\nYou will see more information after build your project or upload your firmware.\n\n- Check the wiki to know more about Deviot https://goo.gl/PKju55\n- If this plugin has been helpful for you, consider to make a donation https://goo.gl/7K4BXh
```

No change after clicking on the Deviot menu and the Select Environment or the Select Board.  When I then click on Compile I get:
```
 [ Deviot 2.2.1 ] Starting...
Select a Board From The List
```

Going back and clicking on Select Environment and Select Board produce no changes in the Deviot Console window.

Ran Platformio from the command line and it compiled correctly.

Went back & repeated the above steps with the same results.

Bob-the-Kuhn, 2018-05-21:
Sorry - wrong console.  Here's the Sublime console's contents aft

```
DPI scale (detected): 1
startup, version: 3176 linux x64 channel: stable
executable: /opt/sublime_text/sublime_text
working dir: /
packages path: /home/bob/.config/sublime-text-3/Packages
state path: /home/bob/.config/sublime-text-3/Local
zip path: /opt/sublime_text/Packages
zip path: /home/bob/.config/sublime-text-3/Installed Packages
ignored_packages: ["Vintage"]
scan: /home/bob/.config/sublime-text-3/Packages/User/Deviot/penv/local/bin has been seen before, skipping (using inode) previous path: /home/bob/.config/sublime-text-3/Packages/User/Deviot/penv/bin inode: 1186149
scan: /home/bob/.config/sublime-text-3/Packages/User/Deviot/penv/local/include has been seen before, skipping (using inode) previous path: /home/bob/.config/sublime-text-3/Packages/User/Deviot/penv/include inode: 1186147
scan: /home/bob/.config/sublime-text-3/Packages/User/Deviot/penv/local/lib has been seen before, skipping (using inode) previous path: /home/bob/.config/sublime-text-3/Packages/User/Deviot/penv/lib inode: 1184765
scan: /home/bob/.config/sublime-text-3/Packages/User/Deviot/penv/local/virtualenv has been seen before, skipping (using inode) previous path: /home/bob/.config/sublime-text-3/Packages/User/Deviot/penv/virtualenv inode: 1051494
pre session restore time: 0.0562953
startup time: 0.0991484
first paint time: 0.106717
reloading plugin Default.arithmetic
reloading plugin Default.auto_indent_tag
reloading plugin Default.block
reloading plugin Default.colors
reloading plugin Default.comment
reloading plugin Default.convert_color_scheme
reloading plugin Default.convert_syntax
reloading plugin Default.copy_path
reloading plugin Default.delete_word
reloading plugin Default.detect_indentation
reloading plugin Default.duplicate_line
reloading plugin Default.echo
reloading plugin Default.exec
reloading plugin Default.fold
reloading plugin Default.font
reloading plugin Default.goto_line
reloading plugin Default.history_list
reloading plugin Default.indentation
reloading plugin Default.install_package_control
reloading plugin Default.kill_ring
reloading plugin Default.mark
reloading plugin Default.new_templates
reloading plugin Default.open_context_url
reloading plugin Default.open_in_browser
reloading plugin Default.pane
reloading plugin Default.paragraph
reloading plugin Default.paste_from_history
reloading plugin Default.profile
reloading plugin Default.quick_panel
reloading plugin Default.rename
reloading plugin Default.run_syntax_tests
reloading plugin Default.save_on_focus_lost
reloading plugin Default.scroll
reloading plugin Default.set_unsaved_view_name
reloading plugin Default.settings
reloading plugin Default.show_scope_name
reloading plugin Default.side_bar
reloading plugin Default.sort
reloading plugin Default.swap_line
reloading plugin Default.switch_file
reloading plugin Default.symbol
reloading plugin Default.transform
reloading plugin Default.transpose
reloading plugin Default.trim_trailing_white_space
reloading plugin Default.ui
reloading plugin CSS.css_completions
reloading plugin Diff.diff
reloading plugin HTML.encode_html_entities
reloading plugin HTML.html_completions
reloading plugin ShellScript.ShellScript
reloading plugin 0_package_control_loader.00-package_control
reloading plugin Package Control.1_reloader
reloading plugin Package Control.2_bootstrap
reloading plugin Package Control.Package Control
reloading plugin WebDevShell.WebDevShell
reloading plugin Deviot (Arduino IDE).Deviot
plugins loaded
reloading settings Packages/User/deviot.sublime-settings
Package Control: Skipping automatic upgrade, last run at 2018-05-21 15:25:29, next run at 2018-05-21 16:25:29 or after
```
No change after clicking on the Deviot menu.

After clicking on Select Environment the following is added:
```
Traceback (most recent call last):
  File "./python3.3/json/decoder.py", line 367, in raw_decode
StopIteration

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/sublime_text/sublime_plugin.py", line 1036, in run_
    return self.run()
  File "/home/bob/.config/sublime-text-3/Packages/Deviot (Arduino IDE)/commands/deviot_select_environment.py", line 8, in run
    items = Quick.environment_list()
  File "/home/bob/.config/sublime-text-3/Packages/Deviot (Arduino IDE)/libraries/quick_menu.py", line 139, in environment_list
    boards = self.boards_list()
  File "/home/bob/.config/sublime-text-3/Packages/Deviot (Arduino IDE)/libraries/quick_menu.py", line 89, in boards_list
    boards = boards_file.read_json()
  File "/home/bob/.config/sublime-text-3/Packages/Deviot (Arduino IDE)/libraries/file.py", line 57, in read_json
    return json.loads(file)
  File "./python3.3/json/__init__.py", line 316, in loads
  File "./python3.3/json/decoder.py", line 351, in decode
  File "./python3.3/json/decoder.py", line 369, in raw_decode
ValueError: No JSON object could be decoded
```

Looks like it isn't finding the platformio.ini file.

Bob-the-Kuhn, 2018-05-21:
I added a couple of print statements to quickmenu:
```
        selected_boards = self.get_selected_boards()
        **print('selected_boards: ', selected_boards)**
        boards_path = getBoardsFileDataPath()
        **print('boards_path: ', boards_path)**
        boards_file = File(boards_path)
        **print('boards_file: ', boards_file)**
        boards = boards_file.read_json()
```

And got the following (prior to the error message):
```
selected_boards:  ['at90USB1286_DFU', 'melzi_optiboot', 'teensy20', 'melzi', 'rambo', 'sanguino_atmega1284p', 'sanguino_atmega644p', 'megaatmega1280', 'at90USB1286_CDC', 'megaatmega2560']

boards_path:  /home/bob/.config/sublime-text-3/Packages/User/Deviot/pio/boards.json

boards_file:  <Deviot (Arduino IDE).libraries.file.File object at 0x7f9b411faf90>
```
The contents of the boards.json file is:
`/bin/sh: 1: /home/bob/.config/sublime-text-3/Packages/User/Deviot/penv/bin/platformio: not found`

Bob-the-Kuhn, 2018-05-21:
In **save_board_list** the run_command is returning:
`/bin/sh: 1: /home/bob/.config/sublime-text-3/Packages/User/Deviot/penv/bin/platformio: not found`

gepd, 2018-06-07:
Please install the develop branch and let me know what errors you keep having https://github.com/gepd/Deviot/wiki/Setup#developer