gepd/Deviot #236: bug

zjbdiy, 2018-06-06:
首先感谢作者的付出,我非常喜欢这个插件,经我测试不能在win10,win7 上运行 ,linux未进行测试

If you can't speak Chinese, I can repeat it in English  .  thanks again.

我的计算机环境  sublime text 3 3176 /win10 x64 17134 / deviot 2.2.1

安装时报错  'gbk' codec can't encode character '\xc8' in position 908: illegal multibyte sequence

我尝试修改 Deviot\libraries\tools.py", line 173  为
with codecs.open(sys_path, 'w','utf8') as configfile:  以utf8编码写入文件

安装成功

之后编译正常  下载正常    重复下载 3次   报错

[ Deviot 2.2.1 ] 启动中...
[06/06/18 15:24:50] Processing uno (platform: atmelavr; board: uno; framework: arduino)

Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Atmel AVR > Arduino Uno
SYSTEM: ATMEGA328P 16MHz 2KB RAM (31.50KB Flash)
Converting fuck.ino
[Decode error - output not utf-8]
[Decode error - output not utf-8]
*** Error 1
AssertionError: :
File "C:\Users\ADMINI~1.WIN\AppData\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\platformio\builder\main.py", line 165:
env.SConscript("$BUILD_SCRIPT")
File "C:\Users\Administrator.WIN-2JL8VFQVGMB\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
return _SConscript(self.fs, *files, **subst_kw)
File "C:\Users\Administrator.WIN-2JL8VFQVGMB\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
exec _file_ in call_stack[-1].globals
File "C:\Users\Administrator.WIN-2JL8VFQVGMB\.platformio\platforms\atmelavr\builder\main.py", line 232:
target_elf = env.BuildProgram()
File "C:\Users\Administrator.WIN-2JL8VFQVGMB\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\Users\ADMINI~1.WIN\AppData\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 69:
env.BuildFrameworks(env.get("PIOFRAMEWORK"))
File "C:\Users\Administrator.WIN-2JL8VFQVGMB\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\Users\ADMINI~1.WIN\AppData\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 275:
env.ConvertInoToCpp()
File "C:\Users\Administrator.WIN-2JL8VFQVGMB\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\Users\ADMINI~1.WIN\AppData\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\platformio\builder\tools\piomisc.py", line 189:
out_file = c.convert(ino_nodes)
File "C:\Users\ADMINI~1.WIN\AppData\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\platformio\builder\tools\piomisc.py", line 54:
return self.process(contents)
File "C:\Users\ADMINI~1.WIN\AppData\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\platformio\builder\tools\piomisc.py", line 77:
assert self._gcc_preprocess(contents, out_file)
 [ERROR] Took 0.82 seconds 

重启sublime  执行编译   报错

DPI mode: per-monitor v2
startup, version: 3176 windows x64 channel: stable
executable: /C/Program Files/Sublime Text 3/sublime_text.exe
working dir: /C/Users/Administrator.WIN-2JL8VFQVGMB/Desktop
packages path: /C/Users/Administrator.WIN-2JL8VFQVGMB/AppData/Roaming/Sublime Text 3/Packages
state path: /C/Users/Administrator.WIN-2JL8VFQVGMB/AppData/Roaming/Sublime Text 3/Local
zip path: /C/Program Files/Sublime Text 3/Packages
zip path: /C/Users/Administrator.WIN-2JL8VFQVGMB/AppData/Roaming/Sublime Text 3/Installed Packages
ignored_packages: ["Vintage"]
pre session restore time: 0.180554
startup time: 0.245554
first paint time: 0.247554
reloading plugin Default.arithmetic
reloading plugin Default.auto_indent_tag
reloading plugin Default.block
reloading plugin Default.colors
reloading plugin Default.comment
reloading plugin Default.convert_color_scheme
reloading plugin Default.convert_syntax
reloading plugin Default.copy_path
reloading plugin Default.delete_word
reloading plugin Default.detect_indentation
reloading plugin Default.duplicate_line
reloading plugin Default.echo
reloading plugin Default.exec
reloading plugin Default.fold
reloading plugin Default.font
reloading plugin Default.goto_line
reloading plugin Default.history_list
reloading plugin Default.indentation
reloading plugin Default.install_package_control
reloading plugin Default.kill_ring
reloading plugin Default.mark
reloading plugin Default.new_templates
reloading plugin Default.open_context_url
reloading plugin Default.open_in_browser
reloading plugin Default.pane
reloading plugin Default.paragraph
reloading plugin Default.paste_from_history
reloading plugin Default.profile
reloading plugin Default.quick_panel
reloading plugin Default.rename
reloading plugin Default.run_syntax_tests
reloading plugin Default.save_on_focus_lost
reloading plugin Default.scroll
reloading plugin Default.set_unsaved_view_name
reloading plugin Default.settings
reloading plugin Default.show_scope_name
reloading plugin Default.side_bar
reloading plugin Default.sort
reloading plugin Default.swap_line
reloading plugin Default.switch_file
reloading plugin Default.symbol
reloading plugin Default.transform
reloading plugin Default.transpose
reloading plugin Default.trim_trailing_white_space
reloading plugin Default.ui
reloading plugin CSS.css_completions
reloading plugin Diff.diff
reloading plugin HTML.encode_html_entities
reloading plugin HTML.html_completions
reloading plugin ShellScript.ShellScript
reloading plugin Deviot.Deviot
plugins loaded
Traceback (most recent call last):
  File "C:\Program Files\Sublime Text 3\sublime_plugin.py", line 283, in on_api_ready
    m.plugin_loaded()
  File "C:\Users\Administrator.WIN-2JL8VFQVGMB\AppData\Roaming\Sublime Text 3\Packages\Deviot\Deviot.py", line 53, in plugin_loaded
    from package_control import events
ImportError: No module named 'package_control'
Exception in thread Thread-4:
Traceback (most recent call last):
  File "./python3.3/threading.py", line 901, in _bootstrap_inner
  File "./python3.3/threading.py", line 858, in run
  File "C:\Users\Administrator.WIN-2JL8VFQVGMB\AppData\Roaming\Sublime Text 3\Packages\Deviot\platformio\compile.py", line 43, in start_compilation
    self.run_command(cmd)
  File "C:\Users\Administrator.WIN-2JL8VFQVGMB\AppData\Roaming\Sublime Text 3\Packages\Deviot\platformio\command.py", line 153, in run_command
    os.environ['PATH'] = env_path
  File "./python3.3/os.py", line 662, in __setitem__
ValueError: the environment variable is longer than 32767 characters

gepd, 2018-06-07:
Hi @zjbdiy can you tell me what is the language configuration are you using in your sistem? I'd like to create a virtual machine to try fix this.

The second list of error are related with package control. It seems that you doesn't have installed. Installing it should solve that problems.

PS: It's better to use english, it may be useful to other people with similar problems

zjbdiy, 2018-06-09:
在cmd中执行chcp 显示如下

活动代码页:936

gepd, 2018-06-09:
Please use English

zjbdiy, 2018-06-13:
OK,this is bad news for me,thanks again.

gepd, 2019-02-15:
Please re open this issue if you still need help

closed 2019-02-15 by gepd