gepd/Deviot #237: Dev problem with example

tablatronix, 2018-06-11:
OSX 
installed a new lib, went to use examples, menu closes

```
Traceback (most recent call last):
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/quick_menu.py", line 406, in callback_library
    library_path = os.path.join(library_path, '*')
UnboundLocalError: local variable 'library_path' referenced before assignment
Traceback (most recent call last):
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/quick_menu.py", line 406, in callback_library
    library_path = os.path.join(library_path, '*')
UnboundLocalError: local variable 'library_path' referenced before assignment
```

tablatronix, 2018-06-11:
uninstalled, reinstalled
error no command PIP

platformio not exist.
hmm

tablatronix, 2018-06-11:
and

```
Traceback (most recent call last):
  File "./python3.3/json/decoder.py", line 367, in raw_decode
StopIteration

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Applications/Sublime Text.app/Contents/MacOS/sublime_plugin.py", line 1036, in run_
    return self.run()
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/commands/deviot_libraries_examples.py", line 6, in run
    QuickMenu().quick_libraries()
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/quick_menu.py", line 357, in quick_libraries
    platform = self.get_platform()
  File "/Users/alverson/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/preferences_bridge.py", line 115, in get_platform
    boards = boards_file.read_json()
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/file.py", line 57, in read_json
    return json.loads(file)
  File "./python3.3/json/__init__.py", line 316, in loads
  File "./python3.3/json/decoder.py", line 351, in decode
  File "./python3.3/json/decoder.py", line 369, in raw_decode
ValueError: No JSON object could be decoded
```

tablatronix, 2018-06-11:
and

on install

```
PlatformIO is not installed
Preparing to install it. Please be patient it; may take a while.
Downloading necessary files...
Extracting files...
Installing PlatformIO...
There was an error
/bin/sh: pip: command not found
Installation finished, now you can enjoy Deviot!
```

tablatronix, 2018-06-12:
where is this python 3.3 coming from ?

```

Traceback (most recent call last):
  File "./python3.3/json/decoder.py", line 367, in raw_decode
StopIteration

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Applications/Sublime Text.app/Contents/MacOS/sublime_plugin.py", line 1036, in run_
    return self.run()
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/commands/deviot_libraries_examples.py", line 6, in run
    QuickMenu().quick_libraries()
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/quick_menu.py", line 357, in quick_libraries
    platform = self.get_platform()
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/preferences_bridge.py", line 115, in get_platform
    boards = boards_file.read_json()
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/file.py", line 57, in read_json
    return json.loads(file)
  File "./python3.3/json/__init__.py", line 316, in loads
  File "./python3.3/json/decoder.py", line 351, in decode
  File "./python3.3/json/decoder.py", line 369, in raw_decode
ValueError: No JSON object could be decoded
using gpu buffer for window

```

tablatronix, 2018-06-12:
```
Exception in thread Thread-70:
Traceback (most recent call last):
  File "./python3.3/threading.py", line 901, in _bootstrap_inner
  File "./python3.3/threading.py", line 858, in run
  File "/Users/alverson/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/platformio/update.py", line 103, in developer_pio
    set_sysetting('pio_developer', True)
NameError: global name 'set_sysetting' is not defined

```

I cannot get this to work at all, bizarre.

tablatronix, 2018-06-12:
I am guess st includes python 3.3 somewhere, I am guessing this is a pathing issue, and deviot is not using its local python or something... That might explain the myriad of errors and nothing working..

tablatronix, 2018-06-12:
suggestion welcome, Will try anything

gepd, 2018-06-13:
Python 3.3 comes from Sublime, but I think it's not the source of the problem

Can you go to `Deviot Menu > Library Options > Rebuild Library List`

If there is any error in the ST console after that paste it here.

You can try this too: `Deviot Menu > Options > Remove Preferences File` restart ST and let me know what is in the ST console

tablatronix, 2018-06-13:
Yeah I tried all that already. I removed user folder entirely reinstalled etc

tablatronix, 2018-06-13:
Ill paste any specific code from the library rebuild though

tablatronix, 2018-06-14:
```

reloading settings Packages/User/deviot.sublime-settings
Errors parsing theme:
icon_file_type is missing layer0.opacity, setting to 1.0 for backwards compatibility
icon_folder is missing layer0.opacity, setting to 1.0 for backwards compatibility
Package Control: No updated packages
Traceback (most recent call last):
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/platformio/command.py", line 235, in _finish
    self._txt.print("\n[{0}]", end_time)
AttributeError: 'NoneType' object has no attribute 'print'
Exception in thread Thread-20:
Traceback (most recent call last):
  File "./python3.3/json/decoder.py", line 367, in raw_decode
StopIteration

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./python3.3/threading.py", line 901, in _bootstrap_inner
  File "./python3.3/threading.py", line 858, in run
  File "/Users/.../Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/libraries.py", line 306, in save_installed_list
    out = loads(out)
  File "./python3.3/json/__init__.py", line 316, in loads
  File "./python3.3/json/decoder.py", line 351, in decode
  File "./python3.3/json/decoder.py", line 369, in raw_decode
ValueError: No JSON object could be decoded

```

tablatronix, 2018-06-14:
ok debugging 

file.py

``` py
read_json FILE ----
/Users/alverson/Library/Application Support/Sublime Text 3/Packages/User/Deviot/penv/bin/python: No module named platformio

----
Traceback (most recent call last):
  File "./python3.3/json/decoder.py", line 367, in raw_decode
StopIteration
````

tablatronix, 2018-06-14:
boards.json

```
/Users/.../Library/Application Support/Sublime Text 3/Packages/User/Deviot/pio/boards.json
```

tablatronix, 2018-06-14:
/bin/sh: /Users/alverson/Library/Application Support/Sublime Text 3/Packages/User/Deviot/penv/bin/python: No such file or directory

tablatronix, 2018-06-14:
Debugging reinstallation

```
 PlatformIO is not installed
Preparing to install it. Please be patient it; may take a while.
Downloading necessary files...
Extracting files...
Installing PlatformIO...
**There was an error**
Collecting platformio
  Using cached https://files.pythonhosted.org/packages/53/1a/05eb8ff68c4f8193a5b9110edc79f55c77a2659a931c9a29958fb54ccbf3/platformio-3.5.3-py27-none-any.whl
Collecting click<6,>=5 (from platformio)
  Using cached https://files.pythonhosted.org/packages/8f/98/14966b6d772fd5fba1eb3bb34a62a7f736d609572493397cdc5715c14514/click-5.1-py2.py3-none-any.whl
Collecting colorama (from platformio)
  Using cached https://files.pythonhosted.org/packages/db/c8/7dcf9dbcb22429512708fe3a547f8b6101c0d02137acbd892505aee57adf/colorama-0.3.9-py2.py3-none-any.whl
Collecting bottle<0.13 (from platformio)
Collecting pyserial!=3.3,<4,>=3 (from platformio)
  Using cached https://files.pythonhosted.org/packages/0d/e4/2a744dd9e3be04a0c0907414e2a01a7c88bb3915cbe3c8cc06e209f59c30/pyserial-3.4-py2.py3-none-any.whl
Collecting requests<3,>=2.4.0 (from platformio)
  Downloading https://files.pythonhosted.org/packages/65/47/7e02164a2a3db50ed6d8a6ab1d6d60b69c4c3fdf57a284257925dfc12bda/requests-2.19.1-py2.py3-none-any.whl (91kB)
Collecting lockfile<0.13,>=0.9.1 (from platformio)
  Using cached https://files.pythonhosted.org/packages/c8/22/9460e311f340cb62d26a38c419b1381b8593b0bb6b5d1f056938b086d362/lockfile-0.12.2-py2.py3-none-any.whl
Collecting semantic-version<3,>=2.5.0 (from platformio)
Collecting certifi>=2017.4.17 (from requests<3,>=2.4.0->platformio)
  Using cached https://files.pythonhosted.org/packages/7c/e6/92ad559b7192d846975fc916b65f667c7b8c3a32bea7372340bfe9a15fa5/certifi-2018.4.16-py2.py3-none-any.whl
Collecting chardet<3.1.0,>=3.0.2 (from requests<3,>=2.4.0->platformio)
  Using cached https://files.pythonhosted.org/packages/bc/a9/01ffebfb562e4274b6487b4bb1ddec7ca55ec7510b22e4c51f14098443b8/chardet-3.0.4-py2.py3-none-any.whl
Collecting urllib3<1.24,>=1.21.1 (from requests<3,>=2.4.0->platformio)
  Using cached https://files.pythonhosted.org/packages/bd/c9/6fdd990019071a4a32a5e7cb78a1d92c53851ef4f56f62a3486e6a7d8ffb/urllib3-1.23-py2.py3-none-any.whl
Collecting idna<2.8,>=2.5 (from requests<3,>=2.4.0->platformio)
  Using cached https://files.pythonhosted.org/packages/4b/2a/0276479a4b3caeb8a8c1af2f8e4355746a97fab05a372e4a2c6a6b876165/idna-2.7-py2.py3-none-any.whl
matplotlib 1.3.1 requires nose, which is not installed.
matplotlib 1.3.1 requires tornado, which is not installed.
Installing collected packages: click, colorama, bottle, pyserial, certifi, chardet, urllib3, idna, requests, lockfile, semantic-version, platformio
Could not install packages due to an EnvironmentError: [Errno 13] Permission denied: '/Library/Python/2.7/site-packages/click'
Consider using the `--user` option or check the permissions.

Installation finished, now you can enjoy Deviot!
```

tablatronix, 2018-06-14:
I think I might have it figured out, somehow I have 2 versions of st installed...
Going to try to fix real quick.

tablatronix, 2018-06-14:
From what I can tell it is this that is failing

```
        cmd = create_command(['pip', 'install', '-U', 'platformio'])
        out = run_command(cmd, "setup_error_2", cwd=self.V_ENV_BIN_PATH)
```

tablatronix, 2018-06-14:
added --user to pip install error goes away still doesn't work though....

tablatronix, 2018-06-14:
what does this mean

``` /Users/.../Library/Python/2.7/lib/python/site-packages (3.5.3)```

Where is that? I have nothing like that anywhere

tablatronix, 2018-06-14:
ok well this makes sense now

 /Users/.../Library/Python/2.7/lib/python/site-packages (3.5.3) 

why is it using user ?

tablatronix, 2018-06-14:
Yeah something is wrong with the python path ( i removed --user), I added permissions to global /Library/python

and it installs there now successfully , but nothing works in deviot.. so install seems to have env wrong ??

tablatronix, 2018-06-14:
ok it works now, but seems to just be using global, that seems wrong

tablatronix, 2018-06-14:
It looks ok

        print(self.V_ENV_PATH)
        print(self.V_ENV_BIN_PATH)

/Users/.../Library/Application Support/Sublime Text 3/Packages/User/Deviot/penv
/Users/.../Library/Application Support/Sublime Text 3/Packages/User/Deviot/penv/bin

tablatronix, 2018-06-14:
and save_env_paths looks ok

tablatronix, 2018-06-14:
heh still doesn't work right, no idea whats going on.

```
Traceback (most recent call last):
  File "/Users/alverson/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/quick_menu.py", line 406, in callback_library
    library_path = os.path.join(library_path, '*')
UnboundLocalError: local variable 'library_path' referenced before assignment
```

gepd, 2018-06-18:
I think I know what where may be the problem, there is any file in `/Users/.../Library/Application Support/Sublime Text 3/Packages/User/Deviot/penv/bin` ?

I think platformio is being installed globally but deviot is trying to run it locally

tablatronix, 2018-06-18:
Yeah I checked my other mac, and its installing locally , this one is not for some reason.

I don't even have a `user/deviot/penv`

I tried debugging up til save_env paths, but didn't see anything wrong that jumped out

tablatronix, 2018-06-21:
Any idea why?
Can I roll back to an older version or remove python global from path or something?

tablatronix, 2018-06-21:
I wiped a bunch of stuff and reinstalled it, and now I have files in user penv but still doesnt work.
[image]

tablatronix, 2018-06-21:
The bin folder contents keep disappearing also, restart st and its empty.

gepd, 2018-06-21:
I'm trying to reproduce the problem in OSX to push a fix

tablatronix, 2018-06-21:
Good luck, I have one machine that still works, and a new macbook that never did. Cant think of anything different. Today i tried stuffing penv path into python.txt but it didnt help

tablatronix, 2018-06-21:
I should be available all say tomorrow CST
feel free to have me try stuff

tablatronix, 2018-06-27:
got a new one today testing again

```
PlatformIO is not installed
Preparing to install it. Please be patient it; may take a while.
Downloading necessary files...
Extracting files...
Installing PlatformIO...
There was an error
Traceback (most recent call last):
  File "/usr/local/bin/pip", line 5, in <module>
    from pkg_resources import load_entry_point
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources/__init__.py", line 3095, in <module>
    @_call_aside
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources/__init__.py", line 3081, in _call_aside
    f(*args, **kwargs)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources/__init__.py", line 3108, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources/__init__.py", line 658, in _build_master
    ws.require(__requires__)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources/__init__.py", line 959, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources/__init__.py", line 846, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'pip==10.0.1' distribution was not found and is required by the application
Installation finished, now you can enjoy Deviot!
```

tablatronix, 2018-06-27:
```
$ echo $PATH
/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin
```

tablatronix, 2018-06-27:
I may have gotten it to work..

I reinstalled xcode with command line, accepted agreements.,
Fixed pip with easy_install

and deleted user/user and restarted ST

Seems to be working now for the most part.

closed 2018-07-18 by gepd

hertell, 2020-08-24:
> and deleted user/deviot and restarted ST

I comment on this in case someone stumbles with similar errors where ST can't find platformio. Deleting User/Deviot fixed this for me